Koenkk/zigbee2mqtt #1635: Error while starting zigbee-shepherd! (Error: request timeout) in combination with an ESP8266 and socat

nightcat91, 2019-06-14:
# Bug Report
## What happened
Error while starting zigbee2mqtt with this command from the docs "npm start"

## What did you expect to happen
not an error message....

## How to reproduce it (minimal and precise)
install zigbee2mqtt inside a LCX container (proxmox hypervisor), set up and ESP8266 as described in the docs, and then try to run the whole setup

## Debug Info
zigbee2mqtt version: installed today following the docs (https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html), so should be the latest, "Starting zigbee2mqtt version 1.4.0 (commit #927c4db)"
CC2530 firmware version: flashed the CC2530 with this file:
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/CC2530_20190425.zip

Hi,

I have already set up zigbee2mqtt sucessfully with a rapsberry pi zero and the CC2531 USB Stick, now I wanted to move zigbee2mqtt inside a linux container on my proxmox hypervisor and use the CC2530 connected to an ESP8266 as described here:
https://www.zigbee2mqtt.io/how_tos/how_to_esp8266_with_cc2530.html

I set up zigbee2mqtt inside the linux container as described in the docs, normal installation, all went well, i didnt start it yet.
Then I flashed the CC2530 with the coordinator firmware and connected it to the EP8266 (which was flashed with ESPeasy and set up following the guide). 
I installed socat inside the linux container and tested the connection, looked good to me.

I changed the serial port in the config file of zigbee2mqtt and tried to run it after making sure that socat was running, but I get the following error message:

```
root@zigbee2mqttcontainer:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 6/14/2019, 11:25:07 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-14.23-25-07'
  zigbee2mqtt:info 6/14/2019, 11:25:07 PM Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
  zigbee2mqtt:info 6/14/2019, 11:25:07 PM Starting zigbee-shepherd
  zigbee2mqtt:info 6/14/2019, 11:25:14 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 6/14/2019, 11:26:14 PM Starting zigbee-shepherd
  zigbee2mqtt:error 6/14/2019, 11:26:21 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 6/14/2019, 11:26:21 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 6/14/2019, 11:26:21 PM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:269:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:248:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 6/14/2019, 11:26:21 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-06-14T23_26_21_347Z-debug.log
root@zigbee2mqttcontainer:/opt/zigbee2mqtt#
```

This is the log file from /root/.npm/_logs/2019-06-14T23_26_21_347Z-debug.log :

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.4.0~prestart: zigbee2mqtt@1.4.0
6 info lifecycle zigbee2mqtt@1.4.0~start: zigbee2mqtt@1.4.0
7 verbose lifecycle zigbee2mqtt@1.4.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.4.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/$
9 verbose lifecycle zigbee2mqtt@1.4.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.4.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.4.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.4.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.4.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55$
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.4.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.15.18-9-pve
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.0
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.4.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.4.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

And this is all that was written in the log inside the data directory from zigbee2mqtt:

```
6/14/2019, 11:43:42 PM - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-14.23-43-42'
6/14/2019, 11:43:42 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221$
6/14/2019, 11:43:42 PM - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
6/14/2019, 11:43:42 PM - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
6/14/2019, 11:43:42 PM - info: Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
6/14/2019, 11:43:42 PM - info: Starting zigbee-shepherd
6/14/2019, 11:43:49 PM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: r$
6/14/2019, 11:44:49 PM - info: Starting zigbee-shepherd
6/14/2019, 11:44:56 PM - error: Error while starting zigbee-shepherd! (Error: request timeout)
```

Any help is welcome, let me know if more info is needed!

And one more thing:
In the docs to set up the ESP8266 and socat there is a small mistake:
https://github.com/Koenkk/zigbee2mqtt.io/blame/master/how_tos/how_to_esp8266_with_cc2530.md#L89
this line has a double "t" at the end, this has to be removed for the service to work correctly, otherwise the "sudo systemctl enable socat-vusb.service" command will fail

JLFN, 2019-06-15:
can you share your config file from zigbee2mqtt. What user is running service? Have you checked rights?

nightcat91, 2019-06-15:
this is my config:

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.11.11.1'
  user: ***********
  password: *********
  include_device_information: true
serial:
  port: /opt/zigbee2mqtt/vusb/zigbee_cc2530
advanced:
  channel: 18
  log_level: debug
  rtscts: false
  network_key:
    - *
    - *
    - *
    - *
    - **
    - **
    - *
    - *
    - *
    - *
    - **
    - *
    - *
    - *
    - *
    - **
```

the socat service is run as root:
```
root@zigbee2mqttcontainer:/etc/systemd/system# ls -l
total 28
drwxr-xr-x 2 root root   3 Jan 24 06:35 getty.target.wants
drwxr-xr-x 2 root root  10 Jun 14 23:00 multi-user.target.wants
drwxr-xr-x 2 root root   3 Jan 24 06:35 network-online.target.wants
-rw-r--r-- 1 root root 264 Jun 14 23:00 socat-vusb.service
lrwxrwxrwx 1 root root  31 Jan 24 06:35 sshd.service -> /lib/systemd/system/ssh.service
drwxr-xr-x 2 root root   3 Jan 24 06:35 sysinit.target.wants
lrwxrwxrwx 1 root root  35 Jan 24 06:35 syslog.service -> /lib/systemd/system/rsyslog.service
drwxr-xr-x 2 root root   4 Jan 24 06:35 timers.target.wants
```

and looks like this:
```
[Unit]
Description=socat-vusb
After=network-online.target

[Service]
User=root
ExecStart=/usr/bin/socat -d -d pty,raw,echo=0,link=/opt/zigbee2mqtt/vusb/zigbee_cc2530 tcp:10.11.12.10:1775,reuseaddr
Restart=always
RestartSec=10

[Install]
WantedBy=multi-user.target
```

the same is true for the zigbee2mqtt service

the zigbee2mqtt folders and files have root permissions as well as the mount point of the "serial port":
```
root@zigbee2mqttcontainer:/opt# ls -l
total 17
drwxr-xr-x 14 root root 26 Jun 14 21:00 zigbee2mqtt
root@zigbee2mqttcontainer:/opt# cd zigbee2mqtt/
root@zigbee2mqttcontainer:/opt/zigbee2mqtt# ls -l
total 280
-rw-r--r--   1 root root  35141 Jun 14 14:51 LICENSE
-rw-r--r--   1 root root   3791 Jun 14 14:51 README.md
drwxr-xr-x   4 root root     10 Jun 15 09:19 data
drwxr-xr-x   2 root root     10 Jun 14 14:51 docker
drwxr-xr-x   2 root root      4 Jun 14 14:51 docs
drwxr-xr-x   2 root root      4 Jun 14 14:51 images
-rw-r--r--   1 root root    546 Jun 14 14:51 index.js
drwxr-xr-x   5 root root      9 Jun 14 14:51 lib
drwxr-xr-x 510 root root    510 Jun 14 14:53 node_modules
-rw-r--r--   1 root root 259762 Jun 14 14:51 npm-shrinkwrap.json
-rw-r--r--   1 root root   1478 Jun 14 14:51 package.json
drwxr-xr-x   2 root root      4 Jun 14 14:51 scripts
drwxr-xr-x   2 root root     14 Jun 14 14:51 test
-rwxr-xr-x   1 root root    439 Jun 14 14:51 update.sh
drwxr-xr-x   2 root root      3 Jun 15 09:16 vusb
root@zigbee2mqttcontainer:/opt/zigbee2mqtt# 
```

```
root@zigbee2mqttcontainer:/opt/zigbee2mqtt# cd vusb/
root@zigbee2mqttcontainer:/opt/zigbee2mqtt/vusb# ls -l
total 1
lrwxrwxrwx 1 root root 10 Jun 15 09:16 zigbee_cc2530 -> /dev/pts/2
root@zigbee2mqttcontainer:/opt/zigbee2mqtt/vusb# 
```

and this is my output of the test if the socat works:
```
root@zigbee2mqttcontainer:/# socat -d -d pty,raw,echo=0,link=/opt/zigbee2mqtt/vusb/zigbee_cc2530 tcp-connect:10.11.12.10:1775
2019/06/15 09:25:09 socat[394] N PTY is /dev/pts/4
2019/06/15 09:25:09 socat[394] N opening connection to AF=2 10.11.12.10:1775
2019/06/15 09:25:09 socat[394] N successfully connected from local address AF=2 10.11.10.109:49858
2019/06/15 09:25:09 socat[394] N starting data transfer loop with FDs [5,5] and [7,7]
```

I did not know there was an install script, I will give that one a try...

JLFN, 2019-06-15:
There is manual buildin. what happens if you remove network key part?

nightcat91, 2019-06-15:
The network key is only the encryption for the zigbee network, i used the same as my currently running network so I dont have to repair my devices, but i will check if removing the key from the config file changes anything

nightcat91, 2019-06-15:
removing the network key doesnt change anything

nightcat91, 2019-06-15:
I did one more test, fresh debian container, new installation of socat and zigbee2mqtt following both documentation and using a user called pi this time, so I used exactly the things written down in the documentation...

Still I get this error message:

```
pi@zigbee2mqttcontainer:/usr/bin$ sudo systemctl restart zigbee2mqtt.service && sudo journalctl -f -u zigbee2mqtt.service
-- Logs begin at Sat 2019-06-15 19:48:32 UTC. --
Jun 15 21:26:12 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jun 15 21:26:12 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jun 15 21:26:13 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Jun 15 21:26:13 zigbee2mqttcontainer systemd[1]: Stopped zigbee2mqtt.
Jun 15 21:26:13 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Jun 15 21:26:13 zigbee2mqttcontainer systemd[1]: Failed to start zigbee2mqtt.
Jun 15 21:26:13 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jun 15 21:26:13 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jun 15 21:27:43 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Failed to reset devices.list: Operation not permitted
Jun 15 21:27:43 zigbee2mqttcontainer systemd[1]: Started zigbee2mqtt.
Jun 15 21:27:44 zigbee2mqttcontainer npm[6862]: > zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
Jun 15 21:27:44 zigbee2mqttcontainer npm[6862]: > node index.js
Jun 15 21:27:44 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:info 6/15/2019, 9:27:44 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-15.21-27-44'
Jun 15 21:27:44 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:info 6/15/2019, 9:27:44 PM Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
Jun 15 21:27:44 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:info 6/15/2019, 9:27:44 PM Starting zigbee-shepherd
Jun 15 21:27:51 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:info 6/15/2019, 9:27:51 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
Jun 15 21:28:51 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:info 6/15/2019, 9:28:51 PM Starting zigbee-shepherd
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:error 6/15/2019, 9:28:58 PM Error while starting zigbee-shepherd! (Error: request timeout)
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:error 6/15/2019, 9:28:58 PM Press the reset button on the stick (the one closest to the USB) and start again
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:error 6/15/2019, 9:28:58 PM Failed to start
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]:         {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:269:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:248:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]:   zigbee2mqtt:error 6/15/2019, 9:28:58 PM Exiting...
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR! code ELIFECYCLE
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR! errno 1
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR! Exit status 1
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR!
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR! A complete log of this run can be found in:
Jun 15 21:28:58 zigbee2mqttcontainer npm[6862]: npm ERR!     /home/pi/.npm/_logs/2019-06-15T21_28_58_524Z-debug.log
Jun 15 21:28:58 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jun 15 21:28:58 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jun 15 21:28:58 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jun 15 21:28:58 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Jun 15 21:28:58 zigbee2mqttcontainer systemd[1]: Stopped zigbee2mqtt.
Jun 15 21:28:58 zigbee2mqttcontainer systemd[1]: zigbee2mqtt.service: Failed to reset devices.list: Operation not permitted
Jun 15 21:28:58 zigbee2mqttcontainer systemd[1]: Started zigbee2mqtt.
Jun 15 21:28:59 zigbee2mqttcontainer npm[6892]: > zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
Jun 15 21:28:59 zigbee2mqttcontainer npm[6892]: > node index.js
Jun 15 21:28:59 zigbee2mqttcontainer npm[6892]:   zigbee2mqtt:info 6/15/2019, 9:28:59 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-15.21-28-59'
Jun 15 21:28:59 zigbee2mqttcontainer npm[6892]:   zigbee2mqtt:info 6/15/2019, 9:28:59 PM Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
Jun 15 21:28:59 zigbee2mqttcontainer npm[6892]:   zigbee2mqtt:info 6/15/2019, 9:28:59 PM Starting zigbee-shepherd
Jun 15 21:29:06 zigbee2mqttcontainer npm[6892]:   zigbee2mqtt:info 6/15/2019, 9:29:06 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
^C
pi@zigbee2mqttcontainer:/usr/bin$ 

```

JLFN, 2019-06-16:
Can you try your cc2530 with an usb ttl 3.3v so comfirm the device is working.

squishykid, 2019-06-19:
I am having the same issues, same ESP & cc2530+cc2591 setup.

Seems similar to #1492 and #1370.

I found that power cycling the CC2530 lets zigbee2mqttt connect again. restarting the socat link or rebooting the ESP doesn't help.

KiLLeRRaT, 2019-06-25:
> I am having the same issues, same ESP & cc2530+cc2591 setup.
> 
> Seems similar to #1492 and #1370.
> 
> I found that power cycling the CC2530 lets zigbee2mqttt connect again. restarting the socat link or rebooting the ESP doesn't help.
> 
> As recommended in #1370 I have already tried the max stability firmware. I have also tried another CC2530+CC2591 device. When I switch back to a CC2530 stick the issue is not present.

I have found the same thing.  Only a power cycle of the ESP8266+CC2530 works.

How _should_ we deal with reliability?

I haven't, but want to make sure the following scenarios recover fully:

1. Power cycle ESP8266 only socat and zigbee2mqtt running
2. Restart socat leaving ESP8266 and zigbee2mqtt running
3.  Restart zigbee2mqtt, leaving ESP8266 and socat running
4.  Virtual Machine host reboot

I feel like it won't recover in all 4 those scenarios.

Should we create a script that would:

- Reboot ESP8266
- Restart socat
- Restart zigbee2mqtt

as soon as we detect that the sensors haven't been updated in a certain time?

KiLLeRRaT, 2019-07-14:
@squishykid fancy seeing you here (I'm the guy with the Solax Inverter in the HASS.IO forum discussion).

Did you ever get to the bottom of this problem?

I am on the dev channel and hasn't really improved for me, works for a while then starts acting up.

Cheers,

squishykid, 2019-07-26:
@KiLLeRRaT I haven't. I am seriously wondering if it's something related to the Serial-via-TCP bridge. I will test this and get back to you.

closed 2019-10-01 by stale[bot]

squishykid, 2019-10-08:
This appears to have been related to my setup with a CC2530 and ESP with serial tunnel. I switched to a GBAN GB2530 connected directly to the server via USB and my range and stability is fantastic now.