Koenkk/zigbee2mqtt #1637: FATAL ERROR: Allocation failed

olomarco, 2019-06-15:
https://pastebin.com/rq4eQr7V

I flashed the firmware on the cc2531 stick with a cc dongle and everything was fine.  I installed mqtt broker on home assistant, and that one looks fine too, the only problem I' having is this message I get with zigbee2mqtt. takes around 15 minutes to do the sheperd and after I get the message it start back again.
Can you guys please give me a hand?

Thank you for your time

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: cc2531

Koenkk, 2019-06-15:
- On what system/os are you running zigbee2mqtt?
- Can you provide the log of `DEBUG=* npm start`.

olomarco, 2019-06-18:
Sorry koenkk but I'm a noob, where can I get that?

Koenkk, 2019-06-19:
Assuming you are running with http://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#5-optional-running-as-a-daemon-with-systemctl, change `ExecStart=/usr/bin/npm start` to `ExecStart=DEBUG=* /usr/bin/npm start`

closed 2019-08-24 by stale[bot]