Koenkk/zigbee2mqtt #1640: Missing "long" from Aqara single switch (WXKG03LM), but has "action": "hold"

mikeage, 2019-06-15:
# Bug Report
## What happened

When using my Aqara single wireless switch (WXKG03LM, CMIIT ID 2016DP1334), I do not get notifications of a long press. I do, however, get a "hold" message, but this is not parsed by the HA switch created from the autodiscovery. 
 
## What did you expect to happen
MQTT message containing `"click":"long"` would be sent, similar to what I get from double switches
Instead, I get an MQTT message containing `"action":"hold"`

## How to reproduce it (minimal and precise)
Press and hold. See log below (with `log_level: debug` set in configuration.yaml
```
  zigbee2mqtt:debug 6/15/2019, 11:21:51 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":0}}' of device 'lumi.remote.b186acn01' (0x00158d0002b899dd) of endpoint 1
  zigbee2mqtt:info 6/15/2019, 11:21:51 PM MQTT publish: topic 'zigbee2mqtt/singleswitch1', payload '{"linkquality":63,"battery":100,"voltage":3125,"action":"hold"}'
  zigbee2mqtt:info 6/15/2019, 11:21:51 PM MQTT publish: topic 'zigbee2mqtt/singleswitch1', payload '{"linkquality":63,"battery":100,"voltage":3125,"action":""}'
```

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: CC2531ZNP-Prod.hex

mikeage, 2019-06-15:
(This might be a duplicate of #817 )

Koenkk, 2019-06-16:
As it would be a breaking change to refactor this, I've added discovery for the action sensor in the dev branch.

In zigbee2mqtt version 2 we can refactor this.

mikeage, 2019-06-16:
Fair enough :-)
I'm not running from dev, but I'll switch over (or just cherry-pick this one change). Thanks for the quick response!

Koenkk, 2019-06-16:
👍

closed 2019-06-16 by Koenkk