Koenkk/zigbee2mqtt #1646: zigbee-herdsman 0.1.9 crashes

hanzoh, 2019-06-19:
# Bug Report
## What happened
Updated to latest Docker dev image tonight.
Right after the start, zigbee-herdsman crashed:
```
(node:16) UnhandledPromiseRejectionWarning: ReferenceError: result is not defined
    at Parser.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:95:33)
    at Parser.emit (events.js:187:15)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/dist/unpi/parser.js:37:26)
    at Parser._transform (/app/node_modules/zigbee-herdsman/dist/unpi/parser.js:23:14)
    at Parser.Transform._read (_stream_transform.js:190:10)
    at Parser.Transform._write (_stream_transform.js:178:12)
    at doWrite (_stream_writable.js:410:12)
    at writeOrBuffer (_stream_writable.js:394:5)
    at Parser.Writable.write (_stream_writable.js:294:11)
    at SerialPort.ondata (_stream_readable.js:666:20)
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:16) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
  zigbee2mqtt:error 6/19/2019, 2:15:21 AM Failed to mount the cieApp
```

## What did you expect to happen
No crash :)

## How to reproduce it (minimal and precise)
Use latest dev commit

## Debug Info
zigbee2mqtt version: latest dev commit
CC253X firmware version: latest 3.0 firmware for CC2530+2591

Koenkk, 2019-06-19:
This was due to a little typo: https://github.com/Koenkk/zigbee-herdsman/commit/882fbaa506c294fe22ec0369630f340ccb67ab89#diff-97d1c29e2cad8ee0162e5e6e9c4557cdR131.

However I'm wondering why an error is thrown in your case, the latest dev branch should fix your crash, could you provide the log of `DEBUG=* npm start`?

hanzoh, 2019-06-19:
I have pulled the latest dev:
```
Using '/app/data' as data directory

> zigbee2mqtt@1.4.0 start /app
> node index.js

  zigbee2mqtt:info 6/19/2019, 6:32:04 PM Logging to directory: '/app/data/log/2019-06-19.18-32-04'
  zigbee2mqtt:debug 6/19/2019, 6:32:04 PM Removing old log directory '/app/data/log/2019-06-19.09-37-52'
  zigbee2mqtt:debug 6/19/2019, 6:32:04 PM Loaded state from file /app/data/state.json
  zigbee2mqtt:debug 6/19/2019, 6:32:04 PM Saving state to file /app/data/state.json
  zigbee2mqtt:info 6/19/2019, 6:32:04 PM Starting zigbee2mqtt version 1.4.0 (commit #unknown)
  zigbee2mqtt:info 6/19/2019, 6:32:04 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 6/19/2019, 6:32:04 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[25],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":false}}'
  zigbee2mqtt:info 6/19/2019, 6:32:09 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 6/19/2019, 6:33:09 PM Starting zigbee-shepherd
  zigbee2mqtt:error 6/19/2019, 6:33:16 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 6/19/2019, 6:33:16 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 6/19/2019, 6:33:16 PM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:209:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:193:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 6/19/2019, 6:33:16 PM Exiting...
```

Here is the npm debug log:
```
~/.npm/_logs # cat 2019-06-19T16_38_41_580Z-debug.log 
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.14.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.4.0~prestart: zigbee2mqtt@1.4.0
6 info lifecycle zigbee2mqtt@1.4.0~start: zigbee2mqtt@1.4.0
7 verbose lifecycle zigbee2mqtt@1.4.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.4.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/app/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.4.0~start: CWD: /app
10 silly lifecycle zigbee2mqtt@1.4.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.4.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.4.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.4.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:182:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:182:13)
13 verbose stack     at maybeClose (internal/child_process.js:962:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
14 verbose pkgid zigbee2mqtt@1.4.0
15 verbose cwd /app
16 verbose Linux 4.19.0-5-amd64
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.14.2
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.4.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.4.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

I have tried to unplug and replug the CC2530+2591, but still the same issue.
Maybe I have to reflash the firmware again like I once had in November with CC2531?

Koenkk, 2019-06-19:
Can you provide the `DEBUG=* npm start` log?

hanzoh, 2019-06-19:
The run above was with DEBUG set to "* npm start", is there any other log file written then the ones I have posted (which include debug messages)?

Koenkk, 2019-06-19:
It shows in the console, you need to set the env variable `DEBUG` to `*`.

hanzoh, 2019-06-19:
Ah, okay, now I got it:

```
Using '/app/data' as data directory

> zigbee2mqtt@1.4.0 start /app
> node index.js

2019-06-19T18:16:45.073Z serialport/bindings loading LinuxBinding
  zigbee2mqtt:info 6/19/2019, 8:16:45 PM Logging to directory: '/app/data/log/2019-06-19.20-16-44'
  zigbee2mqtt:debug 6/19/2019, 8:16:45 PM Removing old log directory '/app/data/log/2019-06-19.20-05-07'
  zigbee2mqtt:debug 6/19/2019, 8:16:45 PM Loaded state from file /app/data/state.json
  zigbee2mqtt:debug 6/19/2019, 8:16:45 PM Saving state to file /app/data/state.json
  zigbee2mqtt:info 6/19/2019, 8:16:45 PM Starting zigbee2mqtt version 1.4.0 (commit #unknown)
  zigbee2mqtt:info 6/19/2019, 8:16:45 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 6/19/2019, 8:16:45 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[25],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":false}}'
2019-06-19T18:16:45.205Z zigbee-shepherd:init zigbee-shepherd booting...
2019-06-19T18:16:45.208Z cc-znp Initialize, opening serial port
2019-06-19T18:16:45.209Z serialport/stream opening path: /dev/zigbee
2019-06-19T18:16:45.209Z serialport/binding-abstract open
2019-06-19T18:16:45.210Z serialport/stream _read queueing _read for after open
2019-06-19T18:16:45.216Z serialport/bindings/poller Creating poller
2019-06-19T18:16:45.216Z serialport/stream opened path: /dev/zigbee
2019-06-19T18:16:45.216Z serialport/stream _write 1 bytes of data
2019-06-19T18:16:45.216Z serialport/binding-abstract write 1 bytes
2019-06-19T18:16:45.217Z serialport/stream _read reading
2019-06-19T18:16:45.217Z serialport/binding-abstract read
2019-06-19T18:16:45.217Z serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1
2019-06-19T18:16:45.217Z serialport/bindings/unixRead Starting read
2019-06-19T18:16:45.219Z serialport/bindings/unixWrite write returned null 1
2019-06-19T18:16:45.219Z serialport/bindings/unixWrite wrote 1 bytes
2019-06-19T18:16:45.219Z serialport/bindings/unixWrite Finished writing 1 bytes
2019-06-19T18:16:45.219Z serialport/stream binding.write write finished
2019-06-19T18:16:45.220Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2019-06-19T18:16:45.220Z serialport/bindings/poller Polling for "readable"
2019-06-19T18:16:46.221Z cc-znp The serialport /dev/zigbee is opened.
2019-06-19T18:16:46.222Z zigbee-shepherd:request REQ --> SYS:version
2019-06-19T18:16:46.227Z cc-znp:SREQ --> SYS:version, {}
2019-06-19T18:16:46.230Z unpi:writer --> frame [254,0,33,2,35]
2019-06-19T18:16:46.230Z serialport/stream _write 5 bytes of data
2019-06-19T18:16:46.230Z serialport/binding-abstract write 5 bytes
2019-06-19T18:16:46.231Z serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5
2019-06-19T18:16:46.231Z serialport/bindings/unixWrite write returned null 5
2019-06-19T18:16:46.231Z serialport/bindings/unixWrite wrote 5 bytes
2019-06-19T18:16:46.231Z serialport/bindings/unixWrite Finished writing 5 bytes
2019-06-19T18:16:46.231Z serialport/stream binding.write write finished
2019-06-19T18:16:46.237Z serialport/bindings/poller received "readable"
2019-06-19T18:16:46.237Z serialport/binding-abstract read
2019-06-19T18:16:46.237Z serialport/bindings/unixRead Starting read
2019-06-19T18:16:46.237Z serialport/bindings/unixRead Finished read 11 bytes
2019-06-19T18:16:46.238Z serialport/stream binding.read finished
2019-06-19T18:16:46.239Z unpi:parser <-- [254,14,97,2,2,2,2,7,2,59,21]
2019-06-19T18:16:46.239Z unpi:parser --- parseNext [254,14,97,2,2,2,2,7,2,59,21]
2019-06-19T18:16:46.240Z serialport/stream _read reading
2019-06-19T18:16:46.240Z serialport/binding-abstract read
2019-06-19T18:16:46.240Z serialport/bindings/unixRead Starting read
2019-06-19T18:16:46.241Z serialport/bindings/unixRead Finished read 8 bytes
2019-06-19T18:16:46.241Z serialport/stream binding.read finished
2019-06-19T18:16:46.241Z unpi:parser <-- [52,1,2,0,0,0,0,115]
2019-06-19T18:16:46.241Z unpi:parser --- parseNext [254,14,97,2,2,2,2,7,2,59,21,52,1,2,0,0,0,0,115]
2019-06-19T18:16:46.242Z unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,2,2,7,2,59,21,52,1,2,0,0,0,0] - 115
2019-06-19T18:16:46.255Z cc-znp Frame {
  type: 'SRSP',
  subsystem: 1,
  commandID: 2,
  data: [ 2, 2, 2, 7, 2, 59, 21, 52, 1, 2, 0, 0, 0, 0 ],
  length: 14,
  fcs: 115,
  csum: 115,
  sof: 254,
[... 562 lines cut ...]
2019-06-19T18:17:56.561Z zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:209:22)
    at Object.onceWrapper (events.js:273:13)
    at CcZnp.emit (events.js:182:13)
    at Timeout.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:193:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
  zigbee2mqtt:error 6/19/2019, 8:17:56 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 6/19/2019, 8:17:56 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 6/19/2019, 8:17:56 PM Failed to start
[... cut ...]   at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 6/19/2019, 8:17:56 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-06-19T18_17_56_652Z-debug.log
```

hanzoh, 2019-06-20:
Latest commit fixed it for me, thanks!

closed 2019-06-20 by hanzoh

Koenkk, 2019-06-20:
I didn't intentionally fix something regarding this, good that it works now but let's keep an eye on future errors.