Koenkk/zigbee2mqtt #1647: How Define Self-Defined Cluster

msadeghz, 2019-06-19:
Hi friends,
I want to add a self-defined cluster with a specific Manufacture Code. how can I do that?

msadeghz, 2019-07-12:
I have a new device which is HA v1.2 and joined successfully on zigbee2mqtt. I want to send it a  command to a self-defined cluster. how can I do that? 
@Koenkk , don't you know how I can find the procedure?

Koenkk, 2019-07-12:
You can define the `cid` as hex like done with `attrId` here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js#L906

msadeghz, 2019-07-12:
> You can define the `cid` as hex like done with `attrId` here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js#L906

thank you, I will try and let others know.

Koenkk, 2019-07-12:
Good! I will close this as it's not an issue, but feel free to ask any other questions.

closed 2019-07-12 by Koenkk

Arminhosseini, 2019-09-02:
Hi,
I'm using a new device that has a self_defined cluster with a specific Manufacture Code and joined successfully to zigbee2mqtt.
I'm trying to send a command to a self-defined cluster and I got an error.
Here are my code and error
[image]
[image]