Koenkk/zigbee2mqtt #1648: TRADFRI signal repeater gives no linkquality

rolfberkenbosch, 2019-06-19:
# Bug Report
I seen no  linkquality for the TRADFRI signal repeater

## What happened
I see the folowing in home assistant:
0x000d6ffffebe3887_linkquality    Onbekend
(onbekend (dutch language means uknown)

## What did you expect to happen
I was expecting to see an integer

## How to reproduce it (minimal and precise)
Just pair the  TRADFRI signal repeater to zigbee2mqtt following the howto guides.

## Debug Info
zigbee2mqtt version: latest master release
CC253X firmware version: latest master release

Koenkk, 2019-06-20:
Linkquality will only be present when zigbee messages of a device are translated to MQTT which is currently not the case for this device.

As I don't have this device, you could try changing the following:

- Change https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L716 to `fromZigbee: [fz.E1746_basic_change, fz.E1746_diagnostic_change],`
- Add to https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L136 

```js
    E1746_basic_change: {
        cid: 'genBasic',
        type: 'devChange',
        convert: (model, msg, publish, options) => {return {};},
    },
    E1746_diagnostic_change: {
        cid: 'haDiagnostic',
        type: 'devChange',
        convert: (model, msg, publish, options) => {return {};},
    },
```

It could take some time before the linkquality popups (depending when the repeater triggering one of the above converters).

To edit these files you can follow: http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

=====================================

If after a day still nothing is reported it would suggest changing:
- https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L710 to 

```js
    {
        zigbeeModel: ['TRADFRI signal repeater'],
        model: 'E1746',
        description: 'TRADFRI signal repeater',
        supports: '',
        vendor: 'IKEA',
        fromZigbee: [fz.ignore_basic_change, fz.ignore_diagnostic_change, fz.E1746_genBasic_report],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('genBasic', 'modelId', 0, 3600, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

- Add to https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L136 

```js
    E1746_genBasic_report: {
        cid: 'genBasic',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            return {};
        },
    },
```

This would setup the device to send it's `modelId` every `3600` seconds, triggering the `E1746_genBasic_report` converter.

closed 2019-08-19 by Koenkk

BearTi, 2020-09-21:
What is the status here? Is the fix only availabvle in Dev branche?