Koenkk/zigbee2mqtt #1649: CC2530+CC2591 router documentation unclear

andreasbrett, 2019-06-19:
I'm trying to build a CC2530+CC2591 router like documented here: https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

Only issue I have is that I'm unsure about the pin layout for the WeeBee CC2530. Should I use the 6 pins that are marked red: 2x GND, 2x VCC, 1x P0_2\U_RX, 1x P0_3\U_TX? If so, why is RST not marked red and why and how use P0_2 and P0_3 instead of P2_1 and P2_2? Or should I just ignore the red color and connect by name? Also it shows the board from underneath. So every PIN is inverted left-to-right...

I think it would be better to use a different graphic that is perfectly clear and leaves no questions. I think (!) it should be like this: http://www.imagebam.com/image/7483031255249044

Koenkk, 2019-06-20:
The red color can be ignored, your picture indeed makes it much more clear. 

Could you make a pr to change this picture? (https://github.com/Koenkk/zigbee2mqtt.io/blob/master/images/webee_cc2530_cc2591_pinlayout.png)

andreasbrett, 2019-06-20:
> Could you make a pr to change this picture? (https://github.com/Koenkk/zigbee2mqtt.io/blob/master/images/webee_cc2530_cc2591_pinlayout.png)

Done

closed 2019-06-20 by andreasbrett

andreasbrett, 2019-06-20:
Although I already closed I can confirm that the pin layout works like that. I just flashed my webee cc2530+cc2591. Flashing worked without an issue and it has already joined my network.