Koenkk/zigbee2mqtt #1651: last_seen no longer working reliably

marc-gist, 2019-06-19:
Latest update (did a few days ago), seems to be causing issues with "last_seen" timestamp in payloads for some devices.

> 6/19/2019, 7:53:14 PM - info: Zigbee publish to device 'sengled_rgb_bulb', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null

> 6/19/2019, 7:53:14 PM - info: MQTT publish: topic 'zigbee/device/sengled_rgb_bulb', payload '{"state":"ON","linkquality":110,"last_seen":"2019-06-18T03:59:01.488Z","color_temp":488,"color":{"x":0.7006,"y":0.2993},"color_mode":1,"brightness":255}'

> 6/19/2019, 7:53:15 PM - info: Zigbee publish to device 'sengled_rgb_bulb', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null

> 6/19/2019, 7:53:15 PM - info: MQTT publish: topic 'zigbee/device/sengled_rgb_bulb', payload '{"state":"OFF","linkquality":110,"last_seen":"2019-06-18T03:59:01.488Z","color_temp":488,"color":{"x":0.7006,"y":0.2993},"color_mode":1,"brightness":255}'

As you can see the log time stamp is for 6-19-2019, and the payload is 6-18-2019, a day before.
Yet the item is working.  I've restarted the item (removed power and back on) and restarted Z2M

lolorc, 2019-06-20:
might have been b0rken by the latest changes for network map. (just a wild guess)

clockbrain, 2019-06-21:
Might be that but since network map last_seen is tracked separately to mqtt last_seen I suspect something else. Perhaps the device has forgotten its 64 bit address? I've had that happen and yet the device still works. What does state.json show for that device?

marc-gist, 2019-06-22:
Appears to have only happened on powered devices, battery ones are fine.
state.json has the date that keeps being sent in the payload, so something changed in the code that is not updating that, despite Z2M clearly "seeing" it to control it :(

Koenkk, 2019-06-24:
I think the problem is that the device doesn't send any zigbee messages on state change (the ones you see are artificially generated by zigbee2mqtt because the publish succeeds). 

When the publish succeeds, the last seen will also update know (latest dev branch).

marc-gist, 2019-06-24:
It worked fine before upgrading ☹️

Koenkk, 2019-06-24:
Can you post the debug log to confirm that zigbee messages received, also from what version did you upgrade?

closed 2019-08-30 by stale[bot]