Koenkk/zigbee2mqtt #1652: Device with modelID 'undefined' is not supported.

mf01, 2019-06-20:
Hi.

Using debug I've this message...
`Device with modelID 'undefined' is not supported.`

I've tried anything.
- remove Zigbee device from "network" 
- deleting relevant data from database.db
- restarting zigbee2mqtt
- resetting end devices (here contact sensors)
- updating zigbee2mqtt to 1.4.0.

Now I have no idea what is needed that all sensors will will correctly and send status information through MQTT.

Using graphviz I can see this....

```
"0x00158d000312ddd7" [style="rounded, dashed", label="{0x00158d000312ddd7|EndDevice|No model information available|online}"];
"0x00158d000312da6d" [style="rounded, dashed", label="{0x00158d000312da6d|EndDevice|No model information available|online}"];
"0x00158d000312e09f" [style="rounded, dashed", label="{0x00158d000312e09f|EndDevice|No model information available|online}"];
"0x00158d0003121f99" [style="rounded, dashed", label="{0x00158d0003121f99|EndDevice|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online}"];
```

All devices have the same model(id) but only one send information.

```
{"ieeeAddr":"0x00158d000312ddd7","type":"EndDevice","friendly_name":"0x00158d000312ddd7","nwkAddr":57727,"manufId":4151},
{"ieeeAddr":"0x00158d000312da6d","type":"EndDevice","friendly_name":"0x00158d000312da6d","nwkAddr":52325,"manufId":4151},
{"ieeeAddr":"0x00158d000312e09f","type":"EndDevice","friendly_name":"0x00158d000312e09f","nwkAddr":41746,"manufId":4151},
{"ieeeAddr":"0x00158d0003121f99","type":"EndDevice","model":"MCCGQ11LM","friendly_name":"0x00158d0003121f99","nwkAddr":26423,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet.aq2","hwVersion":2,"swBuildId":"3000-0001","dateCode":"20161128"}
```

What is really need to fix this issue?

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: 20190223

Koenkk, 2019-06-21:
Assuming this is a Xiaomi device, you can fix it by pressing the reset button once.

mf01, 2019-06-21:
That what I have tried last days, too..... I've tried everything. Now two of three are now connected with data. I'll make the 3rd if the room is free again.

mf01, 2019-06-21:
What is the normal flow?

- permit_join: true
- connect device.
- wait.
- normally all should be fine.

If now ....
- just leave permit_join:true and press the button? This wasn't working. 

Not. I've tried to press the button nothing happens. set permit_join:true (devices was already connected) and press the device button I've got my data.

Koenkk, 2019-06-21:
The pairing procedure for each device can be found on the device page, e.g. http://www.zigbee2mqtt.io/devices/WSDCGQ01LM

mf01, 2019-06-21:
Yes. That's what I had done... without success.
http://www.zigbee2mqtt.io/devices/MCCGQ11LM

However. Now it works. Thx.

closed 2019-06-21 by Koenkk

rsalunga29, 2019-11-04:
> Assuming this is a Xiaomi device, you can fix it by pressing the reset button once.

This worked for me on my Aqara Motion Sensor