Koenkk/zigbee2mqtt #1655: SJCGQ11LM pairing instructions (Aqara water leak sensor)

ovitters, 2019-06-21:
# Bug Report
## What happened
https://www.zigbee2mqtt.io/devices/SJCGQ11LM mentions as pairing instructions:

> Press and hold reset button on the device for +- 5 seconds until the blue light blinks three times, release the reset button (the blue light will blink once more) and wait.

This is incorrect. There's no reset button, nor is there any reset hole or anything. The reset is also not e.g. in the battery section. Instead, the device has a button hidden under the water logo if you just press on the plastic (5mm or so). There's also a LED light near that water logo which is visible through the plastic.

## What did you expect to happen

The pairing instructions should be something like:

> Press and hold water logo on the device for +- 10 seconds until the blue light blinks three times, release the water logo (the blue light will blink once more) and wait.

So 10 seconds instead of 5 plus a mention of the water logo.

## How to reproduce it (minimal and precise)

1. Go to https://www.zigbee2mqtt.io/devices/SJCGQ11LM
2. Read wrong pairing instructions

Shred99, 2019-06-22:
The button is the top of the water leak sensor.  Just push down in the middle.  The LED is under the white top of the sensor - as long as you're not in bright sunlight, you will see the blue LED shine through the white plastic top of the sensor.

Please bear in mind that the people developing this software and the documentation are volunteers, doing it in their own time.  Nobody gets paid for this.

Koenkk, 2019-06-24:
Thanks, updated!

closed 2019-06-24 by Koenkk

trullock, 2022-07-25:
Worth noting that its a physical button, not a touch sensor,. You need to press quite hard to bend the case at the top to press the button inside

ltsalvatore, 2024-02-07:
> Worth noting that its a physical button, not a touch sensor,. You need to press quite hard to bend the case at the top to press the button inside

such a valuable piece of information. thanks a lot mate👌