Koenkk/zigbee2mqtt #1657: Zigbee2MQTT can not reach devices after restart

regnerus, 2019-06-21:
# Bug Report
## What happened
I am in the process of moving my home automation related services to an Intel NUC. When I copied the old configuration (including database.db) from my Raspberry Pi I got errors that devices could not be reached. I decided that repairing might be the best option. Repairing of devices works, however connection is lost again after restarting Zigbee2MQTT.

## How to reproduce it (minimal and precise)
1. Powered down all Zigbee Routers (including Tradfri bulbs) 
2. I start Zigbee2MQTT (currently using the Hassio Addon)
3. Paired one Tradfri bulb
4. I can control the bulb via MQTT/Home Assistant
5. I restart Zigbee2MQTT again
6. Connection to the bulb seems to be lost, not able to control it again.

Here is a pastebin of what happens after restarting the Zigbee2MQTT add-on: https://pastebin.com/bTxrKYCc

## Debug Info
I tried both the normal and edge version of the Hassio Zigbee2MQTT add-on aswell as the official Zigbee2MQTT docker container. Furthermore I tried different Z-Stack firmware for the CC2531 such as versions: 20190425, 20190619 (both Zigbee 1.2 HA) and 20190425 (Zigbee 3.0). Also tried different CC2531 sticks to check if it might be a hardware issue.

zigbee2mqtt version: 1.4
CC253X firmware version: 20190425

I also have quite a few Aqara sensors/buttons which I did not take down yet, however since these are all end devices they should not keep the previous network up right?

Koenkk, 2019-06-23:
Do you use an usb extension cable? Plugging it directly might cause interference.

regnerus, 2019-06-24:
Yes, I've read the page about improving network range and I tried both a short (30cm) and longer (2m) extension cable, also I've switched to Zigbee channel 25 which should interfere less with my 2.4 GHz wifi. Furthermore the NUC is in the same position where the Raspberry Pi was. 

I also have two CC2530 routers but those also lose the connection after restarting Zigbee2MQTT.

Koenkk, 2019-06-24:
Can you try without using the reporting feature?

regnerus, 2019-06-25:
Unfortunately the same issue when I disable the reporting feature.

Here is a Pastebin of the initial pairing and controlling of the light: https://pastebin.com/1s6M15vZ

And here is a Pastebin of what happens after restarting the add-on, just only about a minute later and without changing anything else: https://pastebin.com/4qRhcsgg

_I am aware that the logs also contain some messages received by MQTT for lights currently not on the network initiated by some automations._

regnerus, 2019-06-26:
I just tried setting up Zigbee2MQTT on my Raspberry Pi again. Replicated the steps of pairing the one bulb and then restarting Zigbee2MQTT. I left the CC2531 stick connected in the same position connected to the same extension cable and kept the same `configuration.yaml` file...

On the Pi it behaves exactly as you would expect, after the restart the light is reconnected.

What I haven't tried so far is installing Zigbee2MQTT directly on the NUC instead of running it in a Docker container. That is something I will try tonight...

Koenkk, 2019-06-26:
My whole home automation setup runs on a NUC6CAYH. 

You might try and USB extension cable, as the housing of the NUC can give interference.

Please keep me up-to-date as information and a possible solution is highly valuable for me and other users.

regnerus, 2019-06-26:
Interesting... 

I already tried USB extension cables of different lengths and types... I'm really not sure if interference is the problem because the coordinator is far away from the NUC.

Right now I switched to a Pi which only runs Zigbee2MQTT which works... If I have some more time this weekend I might try to reflash one of my CC2530 routers as a coordinator and see if that does anything. Right now I really have no clue to why this is happening.

regnerus, 2019-07-03:
I did a few more tests and eventually found out that yes indeed the issue was caused by interference. It is still super weird to me that I had no issues when I have the CC2531 connected (through an USB extension) to my RPi but only when I connected it to the NUC (while leaving the CC2531 in exact the same place) when I start getting the issue.  Even-though both the RPi and NUC have wifi/bluetooth disabled.

Through trail and error I found a location where the CC2531 was able to reconnect to the devices, however moving it as little as 30 cm sometimes made it lose the signal again! 

I ordered some parts for the Hack-a-day CC2531 antenna mod (https://hackaday.io/project/163505-cc2531-usb-adapter-antenna-mod) which hopefully will result in a more stable signal.

closed 2019-07-03 by regnerus

Koenkk, 2019-07-03:
@regnerus thanks for letting us know!