Koenkk/zigbee2mqtt #1658: Support for Paul Neuhaus 8095-16 Q-FLAG CCT LED Panel

DoraDeu, 2019-06-21:
It would be great if the Pauls Neuhaus 8095-16 Q-FLAG CCT LED Panel could be added to the supported devices.

It is a CCT panel that can be dimmed and the light temperature can be controlled.
https://www.amazon.de/Paul-Neuhaus-8095-16-Deckenleuchte-Fernbedienung/dp/B01APEPIFC/ref=sr_1_2?keywords=led+panel+30x30&qid=1561152872&refinements=p_89%3APaul+Neuhaus&s=lighting&sr=1-2

The device and remote control could be successfully paired with zigbee adapter in iobroker. 
The remote control has been identified as "NLG-remote" and the LED panel as "undefined".

DoraDeu, 2019-08-08:
Any progress here? 
It could be the case that the communication is similar / identical to the other already supported Paul Neuhaus controller.

Koenkk, 2019-08-09:
To support a new device please follow: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

It will probably work with a config like: 
```js
    {
        zigbeeModel: ['TODO'],
        model: 'TODO',
        vendor: 'Pauls Neuhaus',
        description: 'TODO',
        extend: generic.light_onoff_brightness_colortemp,
    },
```

For the zigbeemodel you have to fill in the `modelId` from `database.db`.

closed 2019-10-08 by Koenkk