Koenkk/zigbee2mqtt #1660: Color Problem OSRAM Flex RGBW 4052899926110

TimBB, 2019-06-22:
Hello

I am using FHEM to connect via zigbee to the device.

on, off brightness, color_temp works great, but i cannot set the color.

When i send the color, nothing happens, it look like i give the wrong format, because the values are "null"

## Debug Info
zigbee2mqtt version 1.4.0 (commit #927c4db)
Coordinator firmware version: '20190425'

debug: Received MQTT message on 'zigbee2mqtt/0x7cb03eaa00ab4309/set' with data '{"color":"RRGGBB"}' 

info: Zigbee publish to device '0x7cb03eaa00ab4309', lightingColorCtrl - moveToColor - {"transtime":0,"colorx":null,"colory":null} - {"manufSpec":0,"disDefaultRsp":0} - null

info: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00ab4309', payload '{"state":"ON","linkquality":49,"brightness":165,"color":{},"color_temp":417,"color_mode":2}'
debug: Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"currentHue":0,"currentSaturation":0,"remainingTime":0,"currentX":24939,"currentY":24701}}' of device 'LIGHTIFY Indoor Flex RGBW' (0x7cb03eaa00ab4309) of endpoint 3
info: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00ab4309', payload '{"state":"ON","linkquality":49,"brightness":165,"color":{"x":0.381,"y":0.377},"color_temp":417,"color_mode":2}'
debug: Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"colorTemperature":417,"colorMode":2,"numPrimaries":3,"primary1X":44891,"primary1Y":20637}}' of device 'LIGHTIFY Indoor Flex RGBW' (0x7cb03eaa00ab4309) of endpoint 3
info: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00ab4309', payload '{"state":"ON","linkquality":49,"brightness":165,"color":{"x":0.381,"y":0.377},"color_temp":417,"color_mode":2}'
debug: Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"primary1Intensity":254,"primary2X":11796,"primary2Y":47533,"primary2Intensity":254,"primary3X":8134}}' of device 'LIGHTIFY Indoor Flex RGBW' (0x7cb03eaa00ab4309) of endpoint 3
debug: Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"primary3Y":3788,"primary3Intensity":254,"primary4X":0,"primary4Y":0,"primary4Intensity":0}}' of device 'LIGHTIFY Indoor Flex RGBW' (0x7cb03eaa00ab4309) of endpoint 3
debug: Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"primary5X":0,"primary5Y":0,"primary5Intensity":0,"primary6X":0,"primary6Y":0}}' of device 'LIGHTIFY Indoor Flex RGBW' (0x7cb03eaa00ab4309) of endpoint 3
debug: Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"primary6Intensity":0,"enhancedCurrentHue":0,"enhancedColorMode":2,"colorLoopActive":0,"colorLoopDirection":0}}' of device 'LIGHTIFY Indoor Flex RGBW' (0x7cb03eaa00ab4309) of endpoint 3
debug: Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"colorLoopTime":25,"colorLoopStartEnhancedHue":8960,"colorLoopStoredEnhancedHue":0,"colorCapabilities":31,"colorTempPhysicalMin":125}}' of device 'LIGHTIFY Indoor Flex RGBW' (0x7cb03eaa00ab4309) of endpoint 3
debug: Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"20480":0,"colorTempPhysicalMax":666}}' of device 'LIGHTIFY Indoor Flex RGBW' (0x7cb03eaa00ab4309) of endpoint 3

How the format must be to change the color?

Thx for help

TimBB, 2019-06-22:
when I use "MQTT Explorer" and publish {"state":"ON","color": {    "x": 0.153,    "y": 0.32} } it works.

Koenkk, 2019-06-24:
It looks that FHEM publishes `'{"color":"RRGGBB"}'` which is not a valid color code, so this should be fixed there. Closing as it isn't a zigbee2mqtt issue.

closed 2019-06-24 by Koenkk