Koenkk/zigbee2mqtt #1661: SP600 Status LEDs and behaviour.

9600, 2019-06-22:
Wondering if anyone knows what the various status modes of the Salus SP600 LEDs are? I've seen:

* slow flashing red
* slow flashing green
* alternating between red, blue and green
* yellow
* 3x short red flashes

Can't seem to find documentation anywhere on what these mean.

When the coordinator crashes I see the first of the above. Sometimes after restarting the coordinator I need to re-pair them — hold the button down for >10s —  before they will join the network again. Whereas other times they will join when the coordinator is restarted. However, sometimes they rejoin and the red flashing stops, but other times the red LED continues to flash and won't go away until I follow the re-join procedure — even though it's otherwise still a functioning node. Odd.

Jim-Noble, 2019-07-17:
I have one of these. Gradually getting a rough idea of what the patterns might mean. 

3 short flashes means it's is trying to pair 
Slow flashing green seems to be for poor or loss of signal. eg if you pressed the local button, but the status update message isn't getting through perhaps.
Slow flashing red seem to be for a more significant network or complete loss of signal type issue.
R->G->B repeated is some sort of reset mode I think. I can only get it to re-pair by triggering this procedure first - not entirely clear what the precise method is though!
Haven't seen yellow so much so far.

James

closed 2019-09-22 by stale[bot]

gausie, 2019-10-30:
I have two SP600s but haven't been able to get my CC2531 to see them at all - no relevant ouput whatsoever to the zigbee2mqtt logs, and I'm in device pairing mode of course. @9600 any suggestions on how to get them to pair?