Koenkk/zigbee2mqtt #1662: Binding Philips Hue Outdoor Sensor SML002 directly to Bulb via ZLL

dotmaster, 2019-06-22:
I would like to pair the hue motion outdoor sensor directly to a bulb an let it send ZLL genOnOff Commands directly without using the coordinator (and not using the Occupancy triggers).

SImilarily to how the ikea motion sensor works. See https://github.com/Koenkk/zigbee2mqtt/issues/102

I have been sniffing the traffic for when the motion sensor comes online (see screenshots below).
As far as I can see the motion sensor has two endpoints. One of the endpoints seems to have Output Clusters for the genOnOff command, which leads me into thinking that the hue outdoor motion might be operated that way.

I therefore have tried to bind a ZLL bulb directly to the sensor (via MQTT message: zigbee2mqtt/bridge/bind/hue-motion
PAYLOAD: innr-candle

I would expect the sensor to send genOnOff commands to the bulb now, but it still just sends out occupancy/illuminance reports (see screenshots).

Am I doing something wrong or is there some hidden toggle, that makes the hue outdoor motion change behaviour?

Her are the results from sniffing:

Active Endpoint Response tells me there are 2 Endpoints:

[image]

Simple Descriptor Response on Endpoint 0x01: ZLL, telling me it's a "reserved" device

[image]

Simple Descriptor Response on Endpoint 0x02: ZHA, telling me it's an occopancy sensor

[image]

 

Bind Request goes to Endpoint 0x01:

[image]

Bind Response tells me it was successful:

[image]

Still when I trigger a motion it just sends occupancy reports:

[image]

Any ideas?

Koenkk, 2019-06-24:
I think endpoint 1 instead of the 2 from the motion sensor is bound, I've added some extra logging in the latest dev branch. Can you check that out and enable debug logging (`log_level: debug`) (http://www.zigbee2mqtt.io/configuration/configuration.html)

dotmaster, 2019-06-24:
Hi,
just checked out dev and ran in debug mode.
This is the relevant part:

>   zigbee2mqtt:debug 2019-6-24 21:03:06 Received MQTT message on 'zigbee2mqtt/bridge/bind/hue-motion' with data 'innr-candle1'
  zigbee2mqtt:debug 2019-6-24 21:03:06 binding cluster '5' from '0x00178801064575c9' (1) to '0x00158d0002f48514'
  zigbee2mqtt:debug 2019-6-24 21:03:06 binding cluster '6' from '0x00178801064575c9' (1) to '0x00158d0002f48514'
  zigbee2mqtt:debug 2019-6-24 21:03:06 binding cluster '8' from '0x00178801064575c9' (1) to '0x00158d0002f48514'
  zigbee2mqtt:debug 2019-6-24 21:03:06 Binding  'hue-motion' - 5
  zigbee-shepherd:request REQ --> ZDO:bindReq +5s
  zigbee-shepherd:msgHdlr IND <-- ZDO:bindRsp +8s
  zigbee2mqtt:debug 2019-6-24 21:03:07 Successfully bound  'hue-motion' - 5
  zigbee2mqtt:info 2019-6-24 21:03:07 Successfully bound cluster '5' from '0x00178801064575c9' to '0x00158d0002f48514'
  zigbee2mqtt:debug 2019-6-24 21:03:07 Binding  'hue-motion' - 6
  zigbee-shepherd:request REQ --> ZDO:bindReq +501ms
  zigbee2mqtt:debug 2019-6-24 21:03:07 Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '1561402987380'
  zigbee-shepherd:msgHdlr IND <-- ZDO:bindRsp +513ms
  zigbee2mqtt:debug 2019-6-24 21:03:07 Successfully bound  'hue-motion' - 6
  zigbee2mqtt:info 2019-6-24 21:03:07 Successfully bound cluster '6' from '0x00178801064575c9' to '0x00158d0002f48514'
  zigbee2mqtt:debug 2019-6-24 21:03:07 Binding  'hue-motion' - 8
  zigbee-shepherd:request REQ --> ZDO:bindReq +501ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:bindRsp +521ms
  zigbee2mqtt:debug 2019-6-24 21:03:08 Successfully bound  'hue-motion' - 8
  zigbee2mqtt:info 2019-6-24 21:03:08 Successfully bound cluster '8' from '0x00178801064575c9' to '0x00158d0002f48514'

Looks like endpoint 1 is used.

Koenkk, 2019-06-25:
Can you check if this is fixed in the latest dev branch? (ep 2 should be used now).

closed 2019-08-31 by stale[bot]

willu, 2020-06-04:
Was this ever sorted out.  I've just tried to do the same thing and had no success.

My Hue outdoor motion sensor is "Huw outdoor motion sensor", and the light controller is "Stitchy Dim 2".  Then this:

$ mosquitto_pub -t "zigbee2mqtt/bridge/bind/Hue outdoor motion sensor/ep1" -m 'Stitchy Dim 2' -q 1

This gave this log:

zigbee2mqtt:debug 2020-06-04 19:37:23: Received MQTT message on 'zigbee2mqtt/bridge/bind/Hue outdoor motion sensor/ep1' with data 'Stitchy Dim 2'                    
zigbee2mqtt:debug 2020-06-04 19:37:23: binding cluster 'genScenes' from 'Hue outdoor motion sensor' to 'Stitchy Dim 2'                                               
zigbee2mqtt:info  2020-06-04 19:37:23: Successfully bound cluster 'genScenes' from 'Hue outdoor motion sensor' to 'Stitchy Dim 2'                                    
zigbee2mqtt:info  2020-06-04 19:37:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"Hue outdoor motion sensor","to
":"Stitchy Dim 2","cluster":"genScenes"}}'                                                                                                                           
zigbee2mqtt:debug 2020-06-04 19:37:23: binding cluster 'genOnOff' from 'Hue outdoor motion sensor' to 'Stitchy Dim 2'                                                
zigbee2mqtt:info  2020-06-04 19:37:24: Successfully bound cluster 'genOnOff' from 'Hue outdoor motion sensor' to 'Stitchy Dim 2'                                     
zigbee2mqtt:info  2020-06-04 19:37:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"Hue outdoor motion sensor","to
":"Stitchy Dim 2","cluster":"genOnOff"}}'                                                                                                                            
zigbee2mqtt:debug 2020-06-04 19:37:24: binding cluster 'genLevelCtrl' from 'Hue outdoor motion sensor' to 'Stitchy Dim 2'                                            
zigbee2mqtt:info  2020-06-04 19:37:24: Successfully bound cluster 'genLevelCtrl' from 'Hue outdoor motion sensor' to 'Stitchy Dim 2'                                 
zigbee2mqtt:info  2020-06-04 19:37:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"Hue outdoor motion sensor","to
":"Stitchy Dim 2","cluster":"genLevelCtrl"}}'                                                                                                                        

That all looks good.  But when I move in front of the motion sensor the light doesn't turn on.  I can see ep2 of the motion sensor activate its occupancy attribute (I previously set the timeout to 20 seconds, and that seems to work).

I suspect the problem is that no scene has been configured.  Does anyone know how to do that?

felix63, 2021-01-23:
Anyone? Trying to do the same

Koenkk, 2021-01-24:
I'm quite certain the outdoor sensor does not support this.