Koenkk/zigbee2mqtt #1663: Device limit? Devices not routing.

silsmaria, 2019-06-22:
# Bug Report

I'm trying to pair multiple devices, all unsuccessfully. I think it is because I have hit a device limit due to end devices not routing through routers.

Trust Remote ZYCT-202 - This pairs, but I when I use it, nothing comes up on the zigbee2mqtt log. When I restart z2m I get the following error: 

> PM Failed to configure 0x00158d0001314818 (0x00158d0001314818) ('Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.') (attempt #1)

Xiaomi Switch (w\ gyro) - Pairs, but doesn't actually work. Nothing happens when I press the button. It sent battery level and link quality at first a few times, but after that nothing.

Orvibo Smart Sticker Switch - This doesn't pair at all (presumably because it is not a supported device though?)

I have exactly 15 end devices that are correctly paired and working with the controller. I have a further 3 routers spread around the house. I am concerned that the end devices aren't routing through the routers, and that I have potentially hit the device limit on the controller.

To check that, I looked at my network map (I'm just sending graphviz to zigbee2mqtt/bridge/networkmap). It shows most devices connecting directly to the controller, and a few devices that aren't connected to anything (even though they do work...) But no kind of routing through the routers at all. Perhaps my network map is not accurate?

## Debug Info
zigbee2mqtt version: 1.4
CC2531 firmware version: 20190109

JLFN, 2019-06-24:
You have very old firmware try this one https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing

pcb1962, 2019-07-02:
@silsmaria did the new firmware help? I have a similar issue.

silsmaria, 2019-07-03:
Thanks @JLFN !

@pcb1962 - sorry - only just got around to updating the firmware. Yes! That did fix it. Didn't have to re-pair old devices either.

Still haven't got the Orvibo Smart Sticker Switch to even pair. I realise it is not a supported device, but I should at least be able to pair it and manually add the mapping. If anyone has had any luck with it, I'd be interested to hear! It's potentially a great little remote.

Thanks!

closed 2019-09-08 by stale[bot]