Koenkk/zigbee2mqtt #1664: Cannot turn on/off Xiaomi Power Socket (lumi.plug)

Nyqi, 2019-06-23:
When I try to turn on/off Xiaomi plug with `zigbee2mqtt/xiaomi_plug_veranda/set` and data `{"state": "ON"}` it throws an exception and it did not turn on/off.

Log: [https://hastebin.com/utodepijet.apache](https://hastebin.com/utodepijet.apache)

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: 20190425

Koenkk, 2019-06-23:
Try `rm -rf node_modules && npm i`

Nyqi, 2019-06-23:
Is this ^ safe for docker-compose container?

Koenkk, 2019-06-24:
Do you use a custom zigbee-shepherd-converters? (mapped via volume with docker), if so, you either need to remove it or update zigbee-shepherd-converters.

tseroo, 2019-07-05:
Similar issue here ... docker image 1.4.0 with shepherd-converters 10.0.2. When sending an ON or OFF to an INNR SP 120, the container crashes. 
I've tried removing zigbee-shepherd-converters => container crashes immediately after starting, then I've downloaded the latest zigbee-shepherd-converters 10.1.1 => container crashes immediately after starting. Nothing is written to the log file. 

Any ideas how to fix that?

tseroo, 2019-07-06:
It's working with docker container latest-dev. Seems only 1.4.0 has issues with the latest zigbee-shepherd-converters release ...

Koenkk, 2019-07-07:
Please provide logs of the crash

tseroo, 2019-07-07:
As written above, nothing is written to the log file...

tseroo, 2019-07-07:
The last message written to the log is 
7/6/2019, 3:53:33 AM - debug: Received MQTT message on 'zigbee2mqtt/eg/sd/sd01/set' with data '{"state": "ON" }'
And then the container restarts permanently, every time creating a new empty log file.

Koenkk, 2019-07-07:
do you also see nothing with the `docker logs -f [container_name]`?

tseroo, 2019-07-07:
I tried again, now looking at docker logs ... using docker image 1.4.0 and zigbee-shepherd-converters 10.1.1. The output is [here](https://drive.google.com/file/d/1nmCVHKo62FpwkHOcQeI4nwu4izRc46xn/view?usp=sharing).

Koenkk, 2019-07-07:
In the zigbee-shepherd-converters folder you need to do `rm -rf node_modules && npm install`.

tseroo, 2019-07-07:
The zigbee-shepherd-converters folder is mapped to the docker container! I cannot do this in the container, as the container is restarting all the time.

tseroo, 2019-07-07:
During the last 2 hours I've had severe issues with my CC2531! Starting zigbee2mqtt container failed, because zigbee-shepherd could not access the stick. My docker host is running as a VM on KVM. The stick was visible within the docker VM after reboot, but after a few minutes it disappeared! I did reboots of the docker VM as well as the KVM host, but without success. Then suddenly it was working again. Very strange! I hope this does not happen more often though...

tseroo, 2019-07-08:
Today I did a rename of a newly added device and afterwards a restart of the docker container, and again the container is restarting all the time:

`CONTAINER ID        IMAGE                                 COMMAND                  CREATED             STATUS                         PORTS                                                                                                              NAMES
655b67eae55e        koenkk/zigbee2mqtt:1.4.0              "./run.sh"               43 seconds ago      Restarting (1) 2 seconds ago   '       

docker log:
Using '/app/data' as data directory

> zigbee2mqtt@1.4.0 start /app
> node index.js

  zigbee2mqtt:info 7/8/2019, 5:41:44 PM Logging to directory: '/app/data/log/2019-07-08.17-41-44'
  zigbee2mqtt:debug 7/8/2019, 5:41:44 PM Removing old log directory '/app/data/log/2019-07-08.17-38-45'
  serialport/bindings loading LinuxBinding +0ms
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/zcl-id'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/app/node_modules/zigbee-shepherd-converters/converters/toZigbee.js:5:15)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-08T17_41_44_916Z-debug.log

Koenkk, 2019-07-09:
Can you post the output of `docker inspect [CONTAINER_ID]` on pastebin.com and link it here?

tseroo, 2019-07-09:
I was not able to get it up an running with 1.4.0 anymore, thus I tried the latest-dev again, and that worked immediately. The inspect is now from the currently running latest-dev container.
[link](https://pastebin.com/raw/Dr6YBK9n)

Koenkk, 2019-07-09:
I see that you are still using a custom zigbee-shepherd-converters folder, `"/etc/docker/zigbee2mqtt/data/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters",`, removing this volume mapping will allow you to run both 1.4.0 and -dev

tseroo, 2019-07-09:
I have that still mapped as I'm using a Lupus socket, which you added recently (#507). So I guess I can either use 1.4.0 with custom mapping or the latest-dev without the mapping, correct? Thanks so far for your support.

Koenkk, 2019-07-09:
I would recommend using the latest-dev, as the latest zigbee-shepherd-converters is not guaranteed to be compatible with 1.4.0.

Closing this issue as the problem has been solved.

closed 2019-07-09 by Koenkk