Koenkk/zigbee2mqtt #1665: On Click Information or no Message at all on Xiaomi Wireless Switch

UlrichFrank, 2019-06-23:
#bug 

Similar to #1632 

- Devices are paired and fully functional.
- A day later connection seemed to be lost.
- I paired devices again -> full functional
- Hours later, click information is not available anymore

Could this be related with device limit? I habe 21 devices (11 Window  sensor MCCGQ01LM, 3 temperature sensors WSDCGQ11LM, 5 Wireless Switches WXKG11LM and two osram Smart+ Plugs)

## Debug Info
zigbee2mqtt Version: 1.3.1
CC253X firmware  Version: 20190425

Koenkk, 2019-06-23:
see https://github.com/Koenkk/zigbee2mqtt/issues/1632#issuecomment-504780096

UlrichFrank, 2019-06-24:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3319368/log.txt)

line 1556: button_004 with last "click" event (genOnOff)
line 1714: button_004 without "click" event (genBasic)

button_005 is still there but unknown/undefined (it has been there before like all the others)

Koenkk, 2019-06-24:
The message on line 1714 seems to be triggered by a battery report of the device (on line 1713) therefore it does not contain a click.

UlrichFrank, 2019-06-24:
Hello Konkk, thanks for the reply.
I‘ll try to generate a smaller log (as soon as possible). Actually I did not verify at which time I have pressend the button, but I am pretty sure that i have pressed it multiple times over a Time Period of Hours in Order to verify the function (While monitoring the Messages on my phone). ;)
How can you see what the trigger of a message has been?

Koenkk, 2019-06-25:
If you see `genBasic` (like line 1713) it's a battery message, `genOnOff` are typically button presses.

UlrichFrank, 2019-07-07:
The device is broken - same behavior with dresden-elektronik/deconz

closed 2019-07-07 by UlrichFrank