Koenkk/zigbee2mqtt #1668: Zigbee2mqtt crashes after plugging a brand new IKEA LED1650r5

gericho, 2019-06-24:
# Bug Report
## What happened

## What did you expect to happen
Pair the device
How to reproduce it (minimal and precise)
Turn ON the LED1650R5
## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version:CC2531_20190425

Koenkk, 2019-06-25:
You need to fix the permission on your `configuration.yaml`

```
Error: EACCES: permission denied, open '/opt/zigbee2mqtt/data/configuration.yaml'
```

gericho, 2019-06-25:
Ok thanks, so the current permission are:

`-rwxr--r-- 1 homeassistant homeassistant 1338 Jun 24 23:01 configuration.yaml`

and needs to be? Thank you very much!

Koenkk, 2019-06-26:
The user which run zigbee2mqtt should have write access, in case you want to go for the quick and dirty hack: `sudo chmod 777 configuration.yaml`.

gericho, 2019-07-04:
I have to reopen this, unfortunately the permission keeps changing back to: 

`
-rwxr--r--  1 homeassistant homeassistant  1239 Jun 30 21:36 configuration.yaml`

it changes back after the machine reboots or HA restart, I'm not 100% sure, setting again to 777 is a workaround that works but a fix is welcome.

Koenkk, 2019-07-04:
There is not much to fix from the zigbee2mqtt side, it's a general linux permission issue. You have to make sure that the user who runs zigbee2mqtt has write access to the file and that others don't change permission on it.

closed 2019-07-04 by gericho