Koenkk/zigbee2mqtt #1670: Log enhancement (add Friendly name)

VivantSenior, 2019-06-25:
## Log enhancement proposal
Now I can see in the logs (e.g.):
> zigbee2mqtt:error 6/22/2019, 6:20:36 AM Failed to ping 0x001...1
zigbee2mqtt:error 6/23/2019, 1:19:37 AM Zigbee publish to device '0x001...2', genTime - readRsp - [{"attrId":0,"status":0,"attrData":34543543,"dataType":226}] - {"direction":1,"seqNum":187,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
zigbee2mqtt:error 6/23/2019, 4:45:06 AM Failed to ping 0x001...2
zigbee2mqtt:error 6/24/2019, 5:59:38 AM Failed to ping 0x001...3

I think it would be more readable to see also `Friendly name` in such logs, e.g.:

> zigbee2mqtt:error 6/22/2019, 6:20:36 AM Failed to ping 0x001...1 **(Friendly name 1)**
zigbee2mqtt:error 6/23/2019, 1:19:37 AM Zigbee publish to device '0x001...2' **(Friendly name 2)**, genTime - readRsp - [{"attrId":0,"status":0,"attrData":34543543,"dataType":226}] - {"direction":1,"seqNum":187,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
zigbee2mqtt:error 6/23/2019, 4:45:06 AM Failed to ping 0x001...2 **(Friendly name 2)**
zigbee2mqtt:error 6/24/2019, 5:59:38 AM Failed to ping 0x001...3 **(Friendly name 3)**

Thanks,
Viv

andreasbrett, 2019-06-25:
This has already been added to the code in the dev branch some weeks ago. Either wait for it to be merged to the master branch or switch over to the dev branch to haved this feature.

Pull Request back then: https://github.com/Koenkk/zigbee2mqtt/pull/1524

Koenkk, 2019-06-25:
@andreasbrett thanks for looking at this, closing.

closed 2019-06-25 by Koenkk