Koenkk/zigbee2mqtt #1672: `Failed to reenable joining` after `Failed to configure ... Timed out after 10000 ms`

vogler, 2019-06-27:
# Bug Report
Just updated from 82dc8437a122366ac2657b4dbd8c631a4f2a9fb2 to 51df6b7e8a1b793e6c66b9c524300d725e7a6625 (dev branch).
Tried several times, but always got stuck repeating `Failed to reenable joining` after `Failed to configure ... ('Error: Timed out after 10000 ms') (attempt #1)` (different devices).
Checked out the old commit and everything is working fine again.

Here are two logs with the new commit that show the problem:
https://termbin.com/5to8
https://termbin.com/d1zt

And here after checking out the old commit again:
https://termbin.com/5iy4

Maybe related issues:
https://github.com/Koenkk/zigbee2mqtt/issues/1631
https://github.com/Koenkk/zigbee2mqtt/issues/1601

## Debug Info
CC253X firmware version: CC2530_20190425

vogler, 2019-08-27:
Still an issue.

Koenkk, 2019-08-27:
When zigbee-herdsman lands after 1.6 in dev this might be fixed

magiva, 2019-10-24:
still an issue with ikea remote 
Failed to configure 0xd0cf5efffe409765 (0xd0cf5efffe409765) ('Error: Timed out after 10000 ms') (attempt #1)

closed 2019-12-30 by stale[bot]