Koenkk/zigbee2mqtt #1674: Inaccurate linkquality from routers

mikeage, 2019-06-28:
# Bug Report
## What happened

I know that linkquality is not a particularly meaningful metric, but it seems like there's some corruption when reporting linkquality for routers. I have a network with one coordinator and two routers, named `downstairsrouter` and `staircaserouter`. Currently, the staircase one is sitting right next to the coordinator, and the downstairs one is far enough away that it barely gets a connection.

However, I sometimes get MQTT messages from `downstairsrouter` that show an excellent link quality; seems like it's the exact value from `staircaserouter`

Snippet showing the attReports and MQTT messages sent
```
  zigbee2mqtt:debug 6/28/2019, 1:06:31 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b0018e1f3e7) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:06:31 PM MQTT publish: topic 'zigbee2mqtt/staircaserouter', payload '{"state":false,"linkquality":134,"description":"48516/0x00124B0018E1F73C","type":"PARENT","rssi":40}'
  zigbee2mqtt:debug 6/28/2019, 1:07:16 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.router' (0x00124b0018e1f73c) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:07:16 PM MQTT publish: topic 'zigbee2mqtt/downstairsrouter', payload '{"state":true,"linkquality":0}'
  zigbee2mqtt:debug 6/28/2019, 1:07:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b0018e1f3e7) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:07:30 PM MQTT publish: topic 'zigbee2mqtt/staircaserouter', payload '{"state":false,"linkquality":134,"description":"48516/0x00124B0018E1F73C","type":"PARENT","rssi":40}'
  zigbee2mqtt:debug 6/28/2019, 1:08:16 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.router' (0x00124b0018e1f73c) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:08:16 PM MQTT publish: topic 'zigbee2mqtt/downstairsrouter', payload '{"state":true,"linkquality":134}'
  zigbee2mqtt:debug 6/28/2019, 1:08:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b0018e1f3e7) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:08:30 PM MQTT publish: topic 'zigbee2mqtt/staircaserouter', payload '{"state":false,"linkquality":134,"description":"48516/0x00124B0018E1F73C","type":"PARENT","rssi":40}'
  zigbee2mqtt:debug 6/28/2019, 1:09:16 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.router' (0x00124b0018e1f73c) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:09:16 PM MQTT publish: topic 'zigbee2mqtt/downstairsrouter', payload '{"state":true,"linkquality":131}'
  zigbee2mqtt:debug 6/28/2019, 1:09:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b0018e1f3e7) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:09:30 PM MQTT publish: topic 'zigbee2mqtt/staircaserouter', payload '{"state":false,"linkquality":134,"description":"48516/0x00124B0018E1F73C","type":"PARENT","rssi":40}'
  zigbee2mqtt:debug 6/28/2019, 1:10:16 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.router' (0x00124b0018e1f73c) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:10:16 PM MQTT publish: topic 'zigbee2mqtt/downstairsrouter', payload '{"state":true,"linkquality":134}'
  zigbee2mqtt:debug 6/28/2019, 1:10:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b0018e1f3e7) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:10:30 PM MQTT publish: topic 'zigbee2mqtt/staircaserouter', payload '{"state":false,"linkquality":134,"description":"48516/0x00124B0018E1F73C","type":"PARENT","rssi":40}'
  zigbee2mqtt:debug 6/28/2019, 1:11:16 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.router' (0x00124b0018e1f73c) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:11:16 PM MQTT publish: topic 'zigbee2mqtt/downstairsrouter', payload '{"state":true,"linkquality":0}'
  zigbee2mqtt:debug 6/28/2019, 1:11:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b0018e1f3e7) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:11:31 PM MQTT publish: topic 'zigbee2mqtt/staircaserouter', payload '{"state":false,"linkquality":131,"description":"48516/0x00124B0018E1F73C","type":"PARENT","rssi":40}'
  zigbee2mqtt:debug 6/28/2019, 1:12:16 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.router' (0x00124b0018e1f73c) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:12:16 PM MQTT publish: topic 'zigbee2mqtt/downstairsrouter', payload '{"state":true,"linkquality":0}'
  zigbee2mqtt:debug 6/28/2019, 1:12:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b0018e1f3e7) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:12:30 PM MQTT publish: topic 'zigbee2mqtt/staircaserouter', payload '{"state":false,"linkquality":131,"description":"48516/0x00124B0018E1F73C","type":"PARENT","rssi":40}'
  zigbee2mqtt:debug 6/28/2019, 1:13:16 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.router' (0x00124b0018e1f73c) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:13:16 PM MQTT publish: topic 'zigbee2mqtt/downstairsrouter', payload '{"state":true,"linkquality":0}'
  zigbee2mqtt:debug 6/28/2019, 1:13:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b0018e1f3e7) of endpoint 8
  zigbee2mqtt:info 6/28/2019, 1:13:30 PM MQTT publish: topic 'zigbee2mqtt/staircaserouter', payload '{"state":false,"linkquality":131,"description":"48516/0x00124B0018E1F73C","type":"PARENT","rssi":40}'
```
## What did you expect to happen

I expect to see reports with a high link quality from the staircaserouter and a very low linkquality from downstairs

## How to reproduce it (minimal and precise)

Not sure.

## Debug Info
zigbee2mqtt version: latest master
CC253X firmware version: coordinator: CC2531_20190425.zip routers: CC2531_router_2018_09.zip

clockbrain, 2019-06-30:
Bringing the link quality issue back here from #652 

I don't know exactly how the link quality is generated for received messages - perhaps it is just reporting the link quality of the last hop and in the cases where you are seeing a high link quality from downstairs it is when traffic is routed via the staircase to the coordinator and in the other cases it is reporting direct link quality from downstairs when messages aren't being routed, i.e. are direct from downstairs to coordinator. 

To confirm you could turn off the staircase router and you should then only see low link quality from downstairs.

mikeage, 2019-06-30:
I did, but the issue is that I know the downstairs router is barely connected. For the test, I placed the staircase router right next to the coordinator; they show 130-140 between them, with downstairs fluctuating between the exact same value as staircase and 0. 

They're both CC2531s, so I don't think they have significantly different reception; it's a different FW, of course, but I wouldn't have expected this type of change. 

My house is made of concrete and rebar, with a few interior walls being cinder block. There is no way it has such a high quality link directly between the two floors! The only thing I can think of is that I have very high coord<->staircase, very low coord<->downstairs, and very low staircase<->downstairs, but the downstairs link is being reported by its last hop. Is that possible? In that case, the network graph is actually correct, but the status on the router itself is, while perhaps not inaccurate, effectively useless.

closed 2019-09-05 by stale[bot]