Koenkk/zigbee2mqtt #1675: Zigbee msg for Selectorswitch trigger scripts only once when multiple identical msg's are sent

wifiminime, 2019-06-28:
# Bug Report
## What happened
zigbee2mqqt plugin for domoticz:

 Philips Hue dimmer switch, press same button multiple times --> scripts only triggered once because device is mapped to Selector switch and state does not change.
For instance: msg: up-press is used to increase light, but you can only do this once... 
Mqqt message is there, not propagated to domoticz because state of selector switch does not change, suppressed in zigbee2mqqt plugin for domoticz

## What did you expect to happen
I expect that when i have a zigbee device that has pushbuttons(!) that the corresponding device in domoticz also behaves like this: after button is released this will also be effective in the state of the domoticz device. (I do realize that this might cause problems for scenarios like press-hold-release)

## How to reproduce it (minimal and precise)
use Zigbee2mqqt plugin for domoticz, add a pushbutton zigbee device that maps on a selector switch (Mi aqara / philips dimmer switch).
Press button (single click) ,wait, press button again, wait, press button again etc
Second and following button clicks will not be shown in domoticz log, and will not trigger scripts.

## Debug Info
zigbee2mqtt version:
latest (installed 27-06-2019)
CC253X firmware version:
not sure

Koenkk, 2019-06-28:
I think this belongs here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

wifiminime, 2019-06-28:
> I think this belongs here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

You are right, how should i transfer this issue? just file new issue?

Koenkk, 2019-06-28:
Just create a new issue there.

closed 2019-06-28 by Koenkk