Koenkk/zigbee2mqtt #1677: Yale lock issues

CanuckianOz, 2019-06-29:
Hey @Koenkk, 

If you recall... you helped me get my Yale lock working and the other models thereafter... 

https://github.com/Koenkk/zigbee2mqtt/issues/1032

It works mostly but occasionally have an issue where if I restart HA or zigbee2mqtt it sometimes stops working and identifies as a slightly different Yale lock model and can't get setup. As far as I can tell, the lock models are identical between the YRD226, 246, 426 and 446. They're all the same lock, just different market or with/without key lock (therefore 4 combinations).  Can you combine all 4 models into one config in devices.js? IE

`        zigbeeModel: ['YRD446 BLE TSDB', 'YRD226/246 TSDB','YRD426NRSC', 'YRD226NRSC'],`

I tried doing this myself but I'm on Hass.io and I get a ton of errors and looping zigbee shepard converter restart loops. 

I'm running zigbee2mqtt hassio 1.4, so the latest version... 
Cheers,
CanuckianOz

Koenkk, 2019-07-01:
Can you provide the debug logs? I'm interested to see where this issue comes from.

poma, 2019-07-05:
I have `YRD226/246 TSDB` lock which is not currently in devices.js but it works when I put model list that @CanuckianOz suggests in place of `['YRD446 BLE TSDB']`

CanuckianOz, 2019-08-24:
@Koenkk Sorry bud, I tried replicating it to get you a log but it randomly assigns one of those models when it reconnects, ie when I restart zigbee2MQTT or Hass.io. When it assigns one that isn't "YRD446 BLE TSDB", I usually can't reliably communicate with the lock and have to restart zigbee2mqtt and hope it works or re-pair the lock. I've been ignoring it since it comes and goes but this needs to be fixed. I had it assign "YRD426NRSC" just now but I didn't have debug logging on (had trouble changing the config). Then got debugging to work and can't replicate the problem. 

I'll fiddle with it again and send you a log when I get another chance.

CanuckianOz, 2019-08-30:
@Koenkk 
Apologies, finally got it to replicate after a restart of the add-on with debug on. It's giving inconsistent lock/unlock statuses which makes automations not useful. Eg right now it says locked, but it's not. 
https://pastebin.com/aexUjvkn

Search for device 0x000d6f0010f9f93f. Looks like it's initially identifying as the YRD426NRSC then setting up as the YRD446 BLE TSDB. I didn't disconnect it from the network; this is just after a restart of the add-on. 

It responds to lock/unlock commands fine, but throws back the same "No converter available" for YRD426NRSC, even though the device is setup as the YRD446 BLE TSDB.

closed 2019-11-05 by stale[bot]