Koenkk/zigbee2mqtt #1678: Help on adding Stelpro SMT402 Thermostat

MrNickll, 2019-06-29:
Hi,
I'm new here and currently setting up a Home Assistant hub.
I confirmed my Zigbee2mqtt works by installing a Lightify bulb. Works nicely and it's recognized as expected. Now the whole reason I wanted a HA hub in the first place is climate control. I'm in Canada and so I have line voltage thermostats. I figured the Stelpro SMT402 should work as it's a fairly recent Zigbee line voltage thermostat.
- Paired it > Zigbee2mqtt sees it.
- Log shows "MaestroStat\u0000\n\u000f\u000f�\u0000�\u0001C�\u0000��\u0001�\u0000���\u0001" unsupported device. Add new device here: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

My HA is ran on Hassio.
- I followed those instructions instead: http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html
- I managed to get in the running docker container, but then I can never copy nor open the devices.js file @ node_modules/zigbee-shepherd-converters/devices.js
I keep getting messages for no such file or directory. 
I want to add this to my devices.js file:

// Stelpro Maestro Thermostat White
{
  zigbeeModel: ['MaestroStat '],
  model: 'SMT402',
  vendor: 'Stelpro',
  description: 'Stelpro Maestro Thermostat',
  supports: 'temperature ',
  fromZigbee: [],
  toZigbee: [],
},

But I can't even open my devices.js.
- I also tried using the option for "zigbee_shepherd_devices": true and adding my own devices.js to share/zigbee2mqtt, but then Zigbee2mqtt doesn't start because I get a "unexpected token" error. I browsed through other similar issues and found I'm supposed to edit a working devices.js, which brings me back to square 1(or 2). 
My understanding right now is:
A. SMT402 pairs.
B. SMT402 not supported.
C. Need to add data to devices.js
D. Can't open/reach/copy existing devices.js

What I need help with to get started and move forward: opening or copying the devices.js in the hassio container. I reach step 4 here: http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html
My issue is at step 5, when I go back to the standard instructions: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Maybe if someone can provide the original devices.js I could start experimenting by using it as a custom devices.js?

## Debug Info
zigbee2mqtt version: 1.4
CC253X firmware version: 20190223

Koenkk, 2019-07-01:
The original devices.js can be found here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js

MrNickll, 2019-07-03:
Great thanks!
Progress so far: 
Taking from the Stelpro ST218 integration and going through database.db, I've added the following to my devices.js (note that I've only added the "msRelativeHumidity" component in the configure section as this is present in database.db):
    `
{
        zigbeeModel: ['MaestroStat'],
        model: 'SMT402',
        vendor: 'Stelpro',
        description: 'Maestro Thermostat',
        supports: 'temperature and humidity',
        fromZigbee: [fz.thermostat_att_report, fz.thermostat_dev_change, fz.xiaomi_humidity],
        toZigbee: [
            tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint,
        ],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genBasic', coordinator, cb),
                (cb) => device.bind('genIdentify', coordinator, cb),
                (cb) => device.bind('genGroups', coordinator, cb),
                (cb) => device.bind('genTime', coordinator, cb),
                (cb) => device.bind('hvacThermostat', coordinator, cb),
                (cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.bind('msRelativeHumidity', coordinator, cb),
                (cb) => device.report('hvacThermostat', 'localTemp', 300, 3600, 0, cb),
            ];
            execute(device, actions, callback);
        },
    },`

Now, the result is that the thermostat seems to connect as shown in the log with 2 devices joined, but fails to configure. It goes to sleep (or at least lcd turning off) quite quickly so I figured I'd stay waving in front of the unit (it has proximity detection) while zigbee2mqtt restarts, but that doesn't make any difference. 
Also, MQTT discovery is turned on, but the thermostat doesn't get added, so I guess I'm missing something. If I remove the entire configure section, the failed to configure message goes away and no other change happens.
See log:

7/3/2019, 2:47:10 PM - info: Logging to directory: '/share/zigbee2mqtt/log/2019-07-03.14-47-10'
7/3/2019, 2:47:18 PM - info: Starting zigbee2mqtt version 1.4.0 (commit #unknown)
7/3/2019, 2:47:18 PM - info: Starting zigbee-shepherd
7/3/2019, 2:47:22 PM - info: zigbee-shepherd started
7/3/2019, 2:47:22 PM - info: Coordinator firmware version: '20190223'
7/3/2019, 2:47:22 PM - info: Currently 2 devices are joined:
7/3/2019, 2:47:22 PM - info: 0x7cb03eaa00a46257 (0x7cb03eaa00a46257): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
7/3/2019, 2:47:22 PM - info: MaestroStat (0xf8f005ffffd13e56): SMT402 - Stelpro Maestro Thermostat (Router)
7/3/2019, 2:47:22 PM - info: Zigbee: disabling joining new devices.
7/3/2019, 2:47:22 PM - info: Connecting to MQTT server at mqtt://homeassistant:1883
7/3/2019, 2:47:24 PM - warn: Failed to configure MaestroStat (0xf8f005ffffd13e56) ('No device') (attempt #1)
7/3/2019, 2:47:24 PM - warn: This can be ignored if the device is working properly
7/3/2019, 2:47:24 PM - info: zigbee-shepherd ready
7/3/2019, 2:47:29 PM - info: Connected to MQTT server
7/3/2019, 2:47:29 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
7/3/2019, 2:47:30 PM - info: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a46257', payload '{"state":"OFF","linkquality":0,"brightness":254,"color":{"x":0.381,"y":0.377,"saturation":100},"color_temp":370,"color_mode":2}'
7/3/2019, 2:47:30 PM - info: MQTT publish: topic 'zigbee2mqtt/MaestroStat', payload '{"local_temperature":22,"linkquality":57}'
7/3/2019, 2:47:30 PM - info: MQTT publish: topic 'homeassistant/light/0x7cb03eaa00a46257/light/config', payload '{"brightness":true,"color_temp":true,"xy":true,"schema":"json","command_topic":"zigbee2mqtt/0x7cb03eaa00a46257/set","state_topic":"zigbee2mqtt/0x7cb03eaa00a46257","name":"0x7cb03eaa00a46257_light","unique_id":"0x7cb03eaa00a46257_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x7cb03eaa00a46257","name":"0x7cb03eaa00a46257","sw_version":"Zigbee2mqtt 1.4.0","model":"LIGHTIFY LED RGBW A19 (73693)","manufacturer":"Sylvania"},"availability_topic":"zigbee2mqtt/bridge/state"}'
7/3/2019, 2:47:30 PM - info: MQTT publish: topic 'homeassistant/sensor/0x7cb03eaa00a46257/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x7cb03eaa00a46257","json_attributes_topic":"zigbee2mqtt/0x7cb03eaa00a46257","name":"0x7cb03eaa00a46257_linkquality","unique_id":"0x7cb03eaa00a46257_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x7cb03eaa00a46257","name":"0x7cb03eaa00a46257","sw_version":"Zigbee2mqtt 1.4.0","model":"LIGHTIFY LED RGBW A19 (73693)","manufacturer":"Sylvania"},"availability_topic":"zigbee2mqtt/bridge/state"}'
7/3/2019, 2:47:30 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190223,"log_level":"info","permit_join":false}'

MrNickll, 2019-07-03:
For the purpose of documenting for helping others in the future, since last post I have:
- Deleted SMT402 again from database.db, state.json and configuration.yaml.
- Removed the configuration section of devices.js.
- Paired again.
- Once Zigbee is turned on, it shows the radio channel. At this point I cycled through all menus first before exiting, then got out of the menu by long pressing center button.
- Added the configuration section to devices.js.
- Restarted zigbee2mqtt.

Log now shows successfully configured MaestroStat:
7/3/2019, 4:31:32 PM - info: Logging to directory: '/share/zigbee2mqtt/log/2019-07-03.16-31-32'
7/3/2019, 4:31:39 PM - info: Starting zigbee2mqtt version 1.4.0 (commit #unknown)
7/3/2019, 4:31:39 PM - info: Starting zigbee-shepherd
7/3/2019, 4:31:44 PM - info: zigbee-shepherd started
7/3/2019, 4:31:44 PM - info: Coordinator firmware version: '20190223'
7/3/2019, 4:31:44 PM - info: Currently 2 devices are joined:
7/3/2019, 4:31:44 PM - info: Lightify Chambre 2 (0x7cb03eaa00a46257): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
7/3/2019, 4:31:44 PM - info: MaestroStat (0xf8f005ffffd13e56): SMT402 - Stelpro Maestro Thermostat (Router)
7/3/2019, 4:31:44 PM - info: Zigbee: disabling joining new devices.
7/3/2019, 4:31:44 PM - info: Connecting to MQTT server at mqtt://homeassistant:1883
7/3/2019, 4:31:45 PM - info: zigbee-shepherd ready
7/3/2019, 4:31:51 PM - info: Successfully configured MaestroStat (0xf8f005ffffd13e56)
7/3/2019, 4:31:52 PM - info: Connected to MQTT server
7/3/2019, 4:31:52 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
7/3/2019, 4:31:53 PM - info: MQTT publish: topic 'zigbee2mqtt/Lightify Chambre 2', payload '{"state":"OFF","linkquality":0,"brightness":254,"color":{"x":0.381,"y":0.377,"saturation":100},"color_temp":370,"color_mode":2}'
7/3/2019, 4:31:53 PM - info: MQTT publish: topic 'zigbee2mqtt/MaestroStat', payload '{"local_temperature":22,"linkquality":57}'
7/3/2019, 4:31:53 PM - info: MQTT publish: topic 'homeassistant/light/0x7cb03eaa00a46257/light/config', payload '{"brightness":true,"color_temp":true,"xy":true,"schema":"json","command_topic":"zigbee2mqtt/Lightify Chambre 2/set","state_topic":"zigbee2mqtt/Lightify Chambre 2","name":"Lightify Chambre 2_light","unique_id":"0x7cb03eaa00a46257_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x7cb03eaa00a46257","name":"Lightify Chambre 2","sw_version":"Zigbee2mqtt 1.4.0","model":"LIGHTIFY LED RGBW A19 (73693)","manufacturer":"Sylvania"},"availability_topic":"zigbee2mqtt/bridge/state"}'
7/3/2019, 4:31:53 PM - info: MQTT publish: topic 'homeassistant/sensor/0x7cb03eaa00a46257/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/Lightify Chambre 2","json_attributes_topic":"zigbee2mqtt/Lightify Chambre 2","name":"Lightify Chambre 2_linkquality","unique_id":"0x7cb03eaa00a46257_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x7cb03eaa00a46257","name":"Lightify Chambre 2","sw_version":"Zigbee2mqtt 1.4.0","model":"LIGHTIFY LED RGBW A19 (73693)","manufacturer":"Sylvania"},"availability_topic":"zigbee2mqtt/bridge/state"}'
7/3/2019, 4:31:53 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190223,"log_level":"info","permit_join":false}'

zhapwey, 2019-07-04:
Does it work correctly?

MrNickll, 2019-07-04:
Right now no.
But I believe I have a MQTT problem. I get it to configure successfully, report localtemp and link quality but it never gets added to HA as a new entity.

zhapwey, 2019-07-05:
besides, only the temperature is removed, can not you act on it?

It's a very nice thermostat, but is it quite expensive to order one and play with it?

Koenkk, 2019-07-05:
To have it added to home assistant, add the following to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L694

```js
    'SMT402': [configurations.thermostat],
```

MrNickll, 2019-07-05:
@zhapwey Yes and no. It is somewhat expensive but to start with, the options for line-voltage smart thermostats are very limited. Then most of the other options are either cloud-based or made by startups or both. Now this one is made by a local (to me) company that only builds thermostats and heating fixtures and I picked it up on sale. 

@Koenkk Thanks that looks like it's the information I was missing!

Change of plans yesterday. I got fed up of restarting(mostly because my test rig is hassio on a pi zero). So I installed a Lubuntu VM and carried on with an actual zigbee2mqtt install. 
Ctrl-C FTW!!
Progress:
I now get report for: local temp, link quality, occupied_heating_setpoint, pi_heating_demand and humidity.
I believe that's all it can report on the thermostat side of things judging by the Smarttings implementation here: 
https://github.com/stelpro/maestro-thermostat/blob/master/devicetypes/stelpro/stelpro-maestro-thermostat.src/stelpro-maestro-thermostat.groovy
There's also on interface cfg temperature display mode and keypad lockout. I get missing converters messages for those as well. Plus I'd love to be able to send it the outdoor temp from HA.
This all brings me to questions:
**1.** In devices.js, in device configure section I have this line: 
`(cb) => device.report('hvacThermostat', 'localTemp', 300, 1800, 0, cb),`
What are the numbers exactly? Min time, max time and what? Or something else entirely? I figured that if I decrease the 1st one to 30, I get a payload every 30 seconds or so, but it seems to repeat the same message over and over. If I change the heat setpoint on the thermostat it still shows the previous setting. Would that mean it's only going to do an actual update at the time set for the second number? So like payload every 30 secs, update every 30 mins?
Is there another way I could get it to report upon value change instead? 
**2.** How would I go about sending the thermostat a value read from another entity? In my case, outdoor temperature acquired from a HA entity? I think it would be a toZigbee converter. Anyone has an example of something similar?
**3**. The thermostat supports remote change for keypad lockout and temp display mode.
If I play in the menu and change those options, I get messages for missing converters. 
Now for each option, I get 2 errors, one for attReport, one for devChange. Example:
```
No converter (cid: hvacuserinterfacecfg, type: attReport, data: Tempdisplaymode:0)
No converter (cid: hvacuserinterfacecfg, type: devChange, data: Tempdisplaymode:0)
```
In this case, `TempDisplayMode: 0` is degrees C, `TempDisplayMode: 1` is degrees F.
Similar situation applies for keypad lockout option. 
A. Is it possible to send commands from zigbee2mqtt to change those options?
B. If I add converters in fromZigbee.js, I know those errors will go away. Would it allow control of options or do I need a toZigbee.js converter?

Koenkk, 2019-07-06:
1. minimal report within 300 seconds, max 1800 seconds, `0` = no change required, if you set it to e.g. `1` it will only report when there is a change (or if the 1800 seconds are expired).
2. You cannot do this with zigbee2mqtt itself, use e.g. home assistant or a handwritten script to transfer the values
3. 
a. yes, just add a converter for this message to `fromZigbee.js`
b. controlling = toZigbee, receiving = fromZigbee

MrNickll, 2019-07-07:
Thanks great!
Finally making progress.
So I fixed my MQTT discovery issues for good. Now on HA side I have:
[image]

On zigbee2mqtt I have that, so I believe it now reports all the things it can:

`zigbee2mqtt:info 2019-7-7 14:46:55 MQTT publish: topic 'zigbee2mqtt/MaestroStat', payload '{"local_temperature":22.5,"linkquality":39,"system_mode":"heat","occupied_heating_setpoint":23.5,"pi_heating_demand":100,"humidity":50,"tempDisplayMode":0,"keypadLockout":0}'`
`zigbee2mqtt:info 2019-7-7 14:51:28 MQTT publish: topic 'zigbee2mqtt/MaestroStat', payload '{"local_temperature":22.5,"linkquality":86,"system_mode":"heat","occupied_heating_setpoint":17,"pi_heating_demand":0,"humidity":50,"tempDisplayMode":0,"keypadLockout":0}'`

So I added the thermostat manually on HA. It reports the correct topics, but if I try to control it, then that breaks something and I need to delete from database (and all other entries), reset and pair again.
The exact sequence is:
- Change set temperature from HA.
- Temp change is shown on thermostat screen.
- "write" entry shows up on zigbee2mqtt log.
- Device with modelID '' is not supported shows up on zigbee2mqtt log.

I think it has something to do with the write command, but I don't understand what I did wrong. New/different converter required?

Log entry is the following:
`zigbee2mqtt:info 2019-7-7 17:09:49 MQTT publish: topic 'zigbee2mqtt/MaestroStat', payload '{"local_temperature":22,"linkquality":86,"system_mode":"heat","occupied_heating_setpoint":16,"pi_heating_demand":0,"humidity":40}'`
`zigbee2mqtt:info 2019-7-7 17:10:25 Zigbee publish to device '0xf8f005ffffd13e56', hvacThermostat - write - [{"attrId":18,"dataType":41,"attrData":1500}] - {"manufSpec":0,"disDefaultRsp":0} - null`
`zigbee2mqtt:warn 2019-7-7 17:10:28 Device with modelID '' is not supported.`
[image]

Koenkk, 2019-07-08:
Please share:
- `database.db`
- Log when publish and having in `configuration.yaml`.

```yaml
advanced:
  log_level: debug
```

MrNickll, 2019-07-09:
@Koenkk Here they are:

[database_snip.txt](https://github.com/Koenkk/zigbee2mqtt/files/3371094/database_snip.txt)
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/3371095/database.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3371096/log.txt)

The Database_snip file is only the Maestro thermostat entity, once before and once after the write command that seems to corrupt it.
In the log, all is good until 2019-7-8 22:53:25. At 2019-7-8 22:55:09, message is received from MQTT, then something happens.

MaestroStat devices.js currently looks like this:
{
    zigbeeModel: ['MaestroStat'],
    model: 'SMT402',
    vendor: 'Stelpro',
    description: 'Maestro Thermostat',
    supports: 'temperature and humidity',
    fromZigbee: [fz.thermostat_att_report, fz.thermostat_dev_change, fz.maestro_humidity_report, fz.maestro_interface_change, fz.maestro_interface_report, fz.ignore_humidity_change],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_temperature_display_mode, tz.maestro_keypad_lockout, tz.thermostat_occupied_heating_setpoint, tz.thermostat_system_mode],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const device = shepherd.find(ieeeAddr, 25);
        const actions = [
            (cb) => device.bind('genBasic', coordinator, cb),
            (cb) => device.bind('genIdentify', coordinator, cb),
            (cb) => device.bind('genGroups', coordinator, cb),
            (cb) => device.bind('genTime', coordinator, cb),
            (cb) => device.bind('hvacThermostat', coordinator, cb),
            (cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
            (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
            (cb) => device.bind('msRelativeHumidity', coordinator, cb),
            (cb) => device.report('hvacThermostat', 'localTemp', 20, 1800, 1, cb),
        ];
        execute(device, actions, callback);
    },
},

Koenkk, 2019-07-09:
The reason for this is that on line 46  a wrong the `modelId` is send by the device which is saved into the `database.db`.

Could you share your `/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js` file?

MrNickll, 2019-07-11:
@Koenkk Here's the file. I don't think I've modified anything in there.
[shepherd.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/3380263/shepherd.js.txt)

Koenkk, 2019-07-11:
Can you replace it with: https://gist.github.com/Koenkk/53e91be26dbc1388d032bcfea17a1ea3 and provide the log again?

MrNickll, 2019-07-12:
Done. Here's the resulting log.
[log_20190712.txt](https://github.com/Koenkk/zigbee2mqtt/files/3384993/log_20190712.txt)

Koenkk, 2019-07-12:
Sorry, I think the statements I've added are not logged in the log file but are only shown on stdout, can you post that?

MrNickll, 2019-07-12:
Ok here they are. However for the first time I couldn't get it to crash  somehow :(

[log_nofault_20190712.txt](https://github.com/Koenkk/zigbee2mqtt/files/3388226/log_nofault_20190712.txt)
[log_pair_20190712.txt](https://github.com/Koenkk/zigbee2mqtt/files/3388227/log_pair_20190712.txt)

Koenkk, 2019-07-13:
Can you try with https://gist.github.com/Koenkk/53e91be26dbc1388d032bcfea17a1ea3

MrNickll, 2019-07-14:
@Koenkk Sorry I didn't test with that file. Slight change of plans due to v1.5.0.
On 1.5.0 it looks like writing doesn't make it corrupted. I've changed the heating setpoint 9 times on these 2 logs here and not once has the database become corrupted. Looking good!

If you want, I can still try with that file, but where is shepherd.js located on 1.5.0?

As of now, the only thing I've seen is that sometimes HA doesn't get the feedback instantly on write. i.e.: HA sends MQTT message > zigbee2mqtt writes to thermostat (confirmed looking at the unit) > HA displays previous heating setpoint. Then when the next message with linkquality gets published it reports what has been written. This is intermittent. 

[log_nofault_20190714_0101.txt](https://github.com/Koenkk/zigbee2mqtt/files/3389499/log_nofault_20190714_0101.txt)
[log_nofault_20190714_0035.txt](https://github.com/Koenkk/zigbee2mqtt/files/3389500/log_nofault_20190714_0035.txt)

Koenkk, 2019-07-15:
Can you change the actions to

```js
const actions = [
(cb) => device.bind('genBasic', coordinator, cb),
(cb) => device.bind('genIdentify', coordinator, cb),
(cb) => device.bind('genGroups', coordinator, cb),
(cb) => device.bind('genTime', coordinator, cb),
(cb) => device.bind('hvacThermostat', coordinator, cb),
(cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
(cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
(cb) => device.bind('msRelativeHumidity', coordinator, cb),
(cb) => device.report('hvacThermostat', 'occupiedHeatingSetpoint', 0, 1800, 1, cb),
(cb) => device.report('hvacThermostat', 'localTemp', 0, 1800, 1, cb),
];
```

MrNickll, 2019-07-18:
@Koenkk if I add the occupied heating setpoint device.report, I get a configuration failure. As soon as I remove it, back to normal. I tried it with only
`(cb) => device.report('hvacThermostat', 'localTemp', 0, 1800, 1, cb),`
and it looks like it reports as soon as I change the set temperature anyway. 

I left zigbee2mqtt running for a few days, when I came back I had a log with the deviceID crashed again. I also noticed a few lines earlier on the log:
`7/14/2019, 1:03:06 AM - warn: No converter available for 'SMT402' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","modelId":"MaestroStat\u0000y\u0000�N\u0001\u001d\\\u0004\u0001\u0001�\u0000�\u0000\u0000\u00015\u0000\u0000","powerSource":1}}'`
`7/14/2019, 1:03:39 AM - warn: No converter available for 'SMT402' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"identifyTime":0}}'`
I figured if I add the converters for ignore_basic_change and ignore_genIdentify_change, maybe it could help? Testing now I'll report back whether or not it crashes.

Koenkk, 2019-07-18:
@MrNickll that won't help, you need to use the fix from https://github.com/Koenkk/zigbee2mqtt/issues/1678#issuecomment-511126765

MrNickll, 2019-07-18:
@Koenkk ok great I'll do that. My previous question remains however: on v1.5.0, where is shepherd.js located? The previous directory is no longer present if I'm not mistaken.
As a side-note, in the meantime, I've tried yet another thing, removed all those:
`(cb) => device.bind('genBasic', coordinator, cb),
(cb) => device.bind('genIdentify', coordinator, cb),
(cb) => device.bind('genGroups', coordinator, cb),
(cb) => device.bind('genTime', coordinator, cb),`
from the configure section. Changing temperature hasn't caused a crash yet and neither did changing system mode (which also used to). Would that make sense?

Koenkk, 2019-07-18:
That could be as now `genBasic` messages (which caused the `modelId` to change) are not send anymore. Let's monitor it for a few days.

bobzer, 2019-07-18:
I also got some SMT402 and one SMC402 I plan to do this
so thank you very much @MrNickll 
I'm a little lost, but i would be glad to help you make this work.
also did you find how to get the power consumption ?
Thanks @Koenkk and @MrNickll for your work

MrNickll, 2019-07-21:
@Koenkk Reporting back after 3 days: still running. No modelID '' not supported message, as I write this thermostat still sends localtemp, linkquality, occupied_heating_setpoint, pi_heating_demand, humidity and system_mode.

@bobzer Power consumption? Is this something Stelpro advertises for Maestro thermostat? As per the smartthings Maestro code, I think it only supports the things I've got it to report already:
``` 
        //Read the configured variables
        zigbee.readAttribute(0x201, 0x0000),	//Read Local Temperature
    	zigbee.readAttribute(0x201, 0x0012),	//Read Heat Setpoint
        zigbee.readAttribute(0x201, 0x0008),	//Read PI Heating State
        zigbee.readAttribute(0x204, 0x0000),	//Read Temperature Display Mode
    	zigbee.readAttribute(0x204, 0x0001),	//Read Keypad Lockout
        zigbee.readAttribute(0x405, 0x0000),	//Read Local Humidity
```        

Unless it's the PI heating demand? So far it reports it, but I'm not sure it always updates.

MrNickll, 2019-07-21:
@Koenkk It started crashing again when playing with temperature and mode. Where do I put that shepherd.js in 1.5.0 please?

MrNickll, 2019-07-22:
@Koenkk Nevermind, found it in zigbee-herdsman/dist.
Here's the resulting log with error on setting the occupied_heating_setpoint to 13.0 C.
[log_fault_20190722_0015.txt](https://github.com/Koenkk/zigbee2mqtt/files/3415703/log_fault_20190722_0015.txt)

Koenkk, 2019-07-22:
No luck yet, updated: https://gist.github.com/Koenkk/53e91be26dbc1388d032bcfea17a1ea3 plesae try again and provide the log.

bobzer, 2019-07-22:
> @bobzer Power consumption? Is this something Stelpro advertises for Maestro thermostat? As per the smartthings Maestro code, I think it only supports the things I've got it to report already:

indeed stelpro advertise this and because i got a smc402 i can see my power consummation by day.
I want to get the same with this 

I'll be glad to help but I'm kind of loss, i don't even know how to start

MrNickll, 2019-07-23:
@bobzer This is done with the SMC402. 
See http://pro.stelpro.com/contenu/ca/pdf/fiches/SMT402.pdf. The energy consumption reports are available *with the Maestro ecosystem. It doesn't look like an available feature when using SMT402 with a Zigbee hub.

MrNickll, 2019-07-23:
@Koenkk Here's the log for today. Failed again sadly.
I see at line 81 it skips the record of attrId 5, but somehow it looks like it's publishing it twice? Also is it possible that attrId 6 at line 107 also sends incorrect data? If I read the log right that would be "datecode"?
[log_fault_20190722_2054.txt](https://github.com/Koenkk/zigbee2mqtt/files/3419948/log_fault_20190722_2054.txt)

bobzer, 2019-07-23:
> @bobzer This is done with the SMC402.
> See http://pro.stelpro.com/contenu/ca/pdf/fiches/SMT402.pdf. The energy consumption reports are available *with the Maestro ecosystem. It doesn't look like an available feature when using SMT402 with a Zigbee hub.

the SMC402 communicate by zigbee to the SMT402, so I gues we should be able to make it works.

@Koenkk maybe i could sniff the communication between the SMT and the SMC . Do you things it could help ?

Koenkk, 2019-07-26:
@MrNickll no luck yet, I need some more information, please try with: https://gist.github.com/Koenkk/53e91be26dbc1388d032bcfea17a1ea3 (won't fix the problems but produces more logging which I need).

@bobzer that could help indeed.

MrNickll, 2019-07-26:
Alright this hasn't crashed yet with this shepherd.js. I'll leave this log here for now anyway in case there's something useful in it.
[log_nofault_20190726_1549.txt](https://github.com/Koenkk/zigbee2mqtt/files/3437225/log_nofault_20190726_1549.txt)

Koenkk, 2019-07-27:
I need a log with crash, let me know when you get one.

MrNickll, 2019-07-28:
@Koenkk Finally here's one. 
[log_fault_20190728_1837.txt](https://github.com/Koenkk/zigbee2mqtt/files/3439639/log_fault_20190728_1837.txt)

Koenkk, 2019-07-29:
@MrNickll updated again (https://gist.github.com/Koenkk/53e91be26dbc1388d032bcfea17a1ea3). May fix the problem.

MrNickll, 2019-07-30:
@Koenkk Unfortunately it's crashed again. Line 242 is when the write command gets sent this time. 
[log_fault_20190730_0024.txt](https://github.com/Koenkk/zigbee2mqtt/files/3445213/log_fault_20190730_0024.txt)

Koenkk, 2019-07-30:
Honestly I have no clue where the `modelId` is updated. You could try to update to zigbee2mqtt 1.5.1 to see if it solves the problem. 

Also you can try to modify the `msg.zclMsg.payload` directly in `af.js` (which can also be found in zigbee-shepherd).

Anyway, I'm currently refactoring the whole zigbee-shepherd library (as the current code is a mess and hard to trace). After that we can fix it for sure.

MrNickll, 2019-09-19:
@Koenkk Hi! I've updated to 1.6 dev this week. I browsed through the devices.js and updated the SMT402 section accordingly to this:
```
{
        zigbeeModel: ['MaestroStat'],
        model: 'SMT402',
        vendor: 'Stelpro',
        description: 'Maestro Thermostat',
        supports: 'temperature, humidity',
        fromZigbee: [
            fz.thermostat_att_report,
            fz.generic_humidity,
            fz.maestro_interface_change,
            fz.maestro_interface_report,
        ],
        toZigbee: [
            tz.thermostat_local_temperature,
            tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_temperature_display_mode, tz.thermostat_keypad_lockout,
            tz.thermostat_system_mode,
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(25);
            const binds = [
                'genBasic', 'genIdentify', 'genGroups', 'hvacThermostat', 'hvacUserInterfaceCfg',
                'msTemperatureMeasurement',
            ];
            await bind(endpoint, coordinatorEndpoint, binds);
            await configureReporting.thermostatTemperature(endpoint);
            await configureReporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await configureReporting.thermostatPIHeatingDemand(endpoint);
        },
    },
```
As of now I've changed the occupied_heating_setpoint a few times and the thermostat still responds. I'll do more testing in the coming days.

bobzer, 2019-10-05:
Hi, 

I succeed to sniff the stelpro zigbee network. I mean kind of because i don't have the network key.
So i quit the network and link back and record everything.
I looked for a device announcement message but there is none but there is association message and they seem successfull.

[reattach.pcapng.txt](https://github.com/Koenkk/zigbee2mqtt/files/3693789/reattach.pcapng.txt)
hope this file can help but if not tell me what could i do ?

MrNickll, 2019-10-08:
@bobzer 
What are we seeing here? All I see is unknown characters. I thought by sniffing between SMC402 and SMT402 we would get the cluster and attribute IDs they use, shouldn't we see that?

bobzer, 2019-10-14:
@MrNickll 
I was hoping to misunderstand the file and that you was able to get the network key from it since there is an association successfull message. 
Problem is i realise i am probably wrong since the thermostat that i remove to put back while recording is not anymore connecting.
Since the other day my thermostat is not visible anymore in the list of attach thermostat and not controllable either. So i screwed up my zigbee stelpro network it seems.

I'm still trying to put back my thermostat in the network provide by the maestro controller.
Still hoping to record the whole thing to get the network key.
But right now i don't really know what to do or try because, i tried a lot already.
Maybe i should quit the network from the controller and do a new one but i'm afraid to screw it up event more.

If you have idea i'm really interested

bobzer, 2019-11-28:
HI, 
I think I got it :smiley: i got the their network key 
so it's 4d:61:6e:79:20:65:74:20:53:69:6d:61:72:64:3a:29
i put that on my wireshark and now everything is decrypt 
I got some sniffing file also don't know what i kind do to help improve the support on this 
is it possible to pair the zigbee2mqtt coordinator with them to be able to control the controller ?
I mean now i guess we should be able to make the thermostat fully works but the controller should be more difficult ?

bobzer, 2019-11-28:
just realize that actually it's provide often 
there is often a paquet which appear as info : link status 
and this give the network key in the header

carldebilly, 2019-12-29:
@bobzer @MrNickll Please check the following PRs:
* Zigbee2Mqtt: https://github.com/Koenkk/zigbee2mqtt/pull/2628
* Herdsman-converters: https://github.com/Koenkk/zigbee-herdsman-converters/pull/853
* Herdsman: https://github.com/Koenkk/zigbee-herdsman/pull/115

Should works with both Stelpro Ki and Stelpro Maestro thermostats!

MrNickll, 2019-12-29:
@carldebilly This is amazing!
Thanks for this work! So most of the things currently not working like the humidity report were in fact typos in herdsman?

Now pardon my git ignorance, but how would I use this PR in my existing instance?

Koenkk, 2020-01-06:
Closing this as https://github.com/Koenkk/zigbee-herdsman-converters/pull/853 has been merged, big thanks to @carldebilly !

closed 2020-01-06 by Koenkk

coryben, 2022-01-10:
> 

@bobzer Did you ever figure out how to connect the controller unit to your HA zigbee network? I have been able to use my Google Home with assistant relay to send commands from HA, but cannot get any information back from the controller into HA. Any help would be much appreciated.

bobzer, 2022-01-10:
Sadly no I didn't 

> > 
> 
> @bobzer Did you ever figure out how to connect the controller unit to your HA zigbee network? I have been able to use my Google Home with assistant relay to send commands from HA, but cannot get any information back from the controller into HA. Any help would be much appreciated.

Sadly no I didn't, I would love to but no idea how could I do this.
I guess the best would be to replace the firmware like : https://www.youtube.com/watch?v=Kdf6W_Ied4o
If you figure out a way I be glad to test it :-)