Koenkk/zigbee2mqtt #1681: Latest dev  not  working

drbios, 2019-07-01:
`pi@hassbian:/opt/zigbee2mqtt $ DEBUG=* npm start

> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:info 6/30/2019, 10:01:10 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-30.22-01-10'
  zigbee2mqtt:debug 6/30/2019, 10:01:10 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 6/30/2019, 10:01:10 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 6/30/2019, 10:01:10 PM Starting zigbee2mqtt version 1.4.0 (commit #51df6b7)
  zigbee2mqtt:info 6/30/2019, 10:01:10 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 6/30/2019, 10:01:10 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":false}}'
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyAMA0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +5ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyAMA0 +7ms
  serialport/stream _write 1 bytes of data +1ms
  serialport/binding-abstract write 1 bytes +12ms
  serialport/stream _read reading +1ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned null 1 +5ms
  serialport/bindings/unixWrite wrote 1 bytes +0ms
  serialport/bindings/unixWrite Finished writing 1 bytes +1ms
  serialport/stream binding.write write finished +8ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +7ms
  serialport/bindings/poller Polling for "readable" +14ms
  cc-znp The serialport /dev/ttyAMA0 is opened. +1s
  zigbee-shepherd:request REQ --> SYS:version +0ms
  cc-znp:SREQ --> SYS:version, {} +0ms
  unpi:writer --> frame [254,0,33,2,35] +0ms
  serialport/stream _write 5 bytes of data +1s
  serialport/binding-abstract write 5 bytes +1s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s
  serialport/bindings/unixWrite write returned null 5 +1ms
  serialport/bindings/unixWrite wrote 5 bytes +0ms
  serialport/bindings/unixWrite Finished writing 5 bytes +0ms
  serialport/stream binding.write write finished +2ms
  serialport/bindings/poller received "readable" +1s
  serialport/binding-abstract read +8ms
  serialport/bindings/unixRead Starting read +1s
  serialport/bindings/unixRead Finished read 19 bytes +0ms
  serialport/stream binding.read finished +7ms
  unpi:parser <-- [254,14,97,2,2,0,2,6,3,155,21,52,1,2,0,0,0,0,209] +0ms
  unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,155,21,52,1,2,0,0,0,0,209] +1ms
  unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,155,21,52,1,2,0,0,0,0] - 209 +1ms
  cc-znp Frame {
  cc-znp   type: 'SRSP',
  cc-znp   subsystem: 1,
  cc-znp   commandID: 2,
  cc-znp   data: [ 2, 0, 2, 6, 3, 155, 21, 52, 1, 2, 0, 0, 0, 0 ],
  cc-znp   length: 14,
  cc-znp   fcs: 209,
  cc-znp   csum: 209,
  cc-znp   sof: 254,
  cc-znp   len: 14,
  cc-znp   subsys: 'SYS',
  cc-znp   cmd: 'version',
  cc-znp   payload:
  cc-znp    { transportrev: 2,
  cc-znp      product: 0,
  cc-znp      majorrel: 2,
  cc-znp      minorrel: 6,
  cc-znp      maintrel: 3,
  cc-znp      revision: 20190619 } } +40ms
  unpi:parser --- parseNext [] +25ms
[... 1809 lines cut ...]
  cc-znp   commandID: 133,
  cc-znp   data: [ 0, 0, 0, 0, 0, 7, 11, 6, 5, 4, 3, 2, 1 ],
  cc-znp   length: 13,
  cc-znp   fcs: 198,
  cc-znp   csum: 198,
  cc-znp   sof: 254,
  cc-znp   len: 13,
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'activeEpRsp',
  cc-znp   payload:
  cc-znp    { srcaddr: 0,
  cc-znp      status: 0,
  cc-znp      nwkaddr: 0,
  cc-znp      activeepcount: 7,
  cc-znp      activeeplist: <Buffer 0b 06 05 04 03 02 01> } } +9ms
  unpi:parser --- parseNext [] +5ms
  serialport/stream _read reading +6ms
  serialport/binding-abstract read +8ms
  serialport/bindings/unixRead Starting read +7ms
  cc-znp:AREQ <-- ZDO:activeEpRsp, { srcaddr: 0, status: 0, nwkaddr: 0, activeepcount: 7, activeeplist: <Buffer 0b 06 05 04 03 02 01> } +30ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +30ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +8ms
  serialport/bindings/poller Polling for "readable" +24ms
  zigbee2mqtt:debug 6/30/2019, 10:01:13 PM Mounted the cieApp (epId 11)
`

drbios, 2019-07-01:
``> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 6/30/2019, 10:04:04 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-30.22-04-03'
  zigbee2mqtt:debug 6/30/2019, 10:04:04 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-06-30.20-51-49'
  zigbee2mqtt:debug 6/30/2019, 10:04:04 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 6/30/2019, 10:04:04 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 6/30/2019, 10:04:04 PM Starting zigbee2mqtt version 1.4.0 (commit #51df6b7)
  zigbee2mqtt:info 6/30/2019, 10:04:04 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 6/30/2019, 10:04:04 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":false}}'
  zigbee2mqtt:info 6/30/2019, 10:04:08 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 6/30/2019, 10:05:08 PM Starting zigbee-shepherd
  zigbee2mqtt:error 6/30/2019, 10:05:15 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 6/30/2019, 10:05:15 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 6/30/2019, 10:05:15 PM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 6/30/2019, 10:05:15 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-07-01T02_05_15_628Z-debug.log

drbios, 2019-07-01:
`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.4.0~prestart: zigbee2mqtt@1.4.0
6 info lifecycle zigbee2mqtt@1.4.0~start: zigbee2mqtt@1.4.0
7 verbose lifecycle zigbee2mqtt@1.4.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.4.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/s$
9 verbose lifecycle zigbee2mqtt@1.4.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.4.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.4.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.4.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.4.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.4.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.56-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.0
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.4.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.4.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
`

drbios, 2019-07-01:
`zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt

node index.js

zigbee2mqtt:info 6/30/2019, 10:04:04 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-06-30.22-04-03'
zigbee2mqtt:debug 6/30/2019, 10:04:04 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-06-30.20-51-49'
zigbee2mqtt:debug 6/30/2019, 10:04:04 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:debug 6/30/2019, 10:04:04 PM Saving state to file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info 6/30/2019, 10:04:04 PM Starting zigbee2mqtt version 1.4.0 (commit #51df6b7)
zigbee2mqtt:info 6/30/2019, 10:04:04 PM Starting zigbee-shepherd
zigbee2mqtt:debug 6/30/2019, 10:04:04 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":false}}'
zigbee2mqtt:info 6/30/2019, 10:04:08 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
zigbee2mqtt:info 6/30/2019, 10:05:08 PM Starting zigbee-shepherd
zigbee2mqtt:error 6/30/2019, 10:05:15 PM Error while starting zigbee-shepherd! (Error: request timeout)
zigbee2mqtt:error 6/30/2019, 10:05:15 PM Press the reset button on the stick (the one closest to the USB) and start again
zigbee2mqtt:error 6/30/2019, 10:05:15 PM Failed to start
{"message":"request timeout","stack":"Error: request timeout\n at CcZnp. (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n at Object.onceWrapper (events.js:286:20)\n at CcZnp.emit (events.js:198:13)\n at Timeout. (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n at ontimeout (timers.js:436:11)\n at tryOnTimeout (timers.js:300:5)\n at listOnTimeout (timers.js:263:5)\n at Timer.processTimers (timers.js:223:10)"}
zigbee2mqtt:error 6/30/2019, 10:05:15 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.4.0 start: node index.js
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! /home/pi/.npm/_logs/2019-07-01T02_05_15_628Z-debug.log`

Koenkk, 2019-07-01:
Is the OP of a failing start? I don't see an error there.

drbios, 2019-07-01:
@Koenkk master branch works (with some errors),  but the device list supported is different
what you mean with OP??

Koenkk, 2019-07-01:
Is this the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/1656#issuecomment-507334118? Then let's close this one and continue there.

drbios, 2019-07-01:
@Koenkk  ok

closed 2019-07-01 by drbios