Koenkk/zigbee2mqtt #1682: error with clean install Raspbian-Buster

daveknoop, 2019-07-02:
Hi!

I'm using a Pi2b with a clean install with the latest Raspbian-Buster system. I followed the instructions to install zigbee2mqtt but every time I get this error when I use "npm install" at the folder "ops/zigbee2mqtt" 

Can somebody help me to get it works?

--------------
root@raspberrypi:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'semver'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/index.js:1:16)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-02T18_36_38_381Z-debug.log
root@raspberrypi:/opt/zigbee2mqtt#
-----------------------

Node en Npm are up-to-date

drbios, 2019-07-03:
same here
https://github.com/Koenkk/zigbee2mqtt/issues/1656

Koenkk, 2019-07-03:
In your post you mention `npm install`, while the log shows you are doing `npm start`, it actually looks like you didn't do a `npm install`.

closed 2019-09-08 by stale[bot]