Koenkk/zigbee2mqtt #1684: Experiment: Turned off the coordinator for several days. Xiaomi devices no longer communicate

directman66, 2019-07-03:
Experiment: Turned off the coordinator for several days.
When turned on, some xiaomi devices fell asleep and on waking up refused to connect to the coordinator cc2561. (wall switches without zero (QBKG04LM and QBKG03LM), motion sensors (both versions RTCGQ01LM and RTCGQ11LM), leakage sensor SJCGQ11LM, wall-mounted remote control WXKG02LM.

When trying to control the switches, we get the error:
error: Failed to ping 0x00158d00029b4352
2019-7-2 17:52:42 - info: zigbee publish to device '0x00158d00022bc3ab', genOnOff - on - {} - {"manufSpec": 0, "disDefaultRsp": 0} - 3
2019-7-2 17:52:42 - info: MQTT publish: topic 'zigbee2mqtt / bridge / log', payload '{"type": "zigbee_publish_error", "message": "Error: AF data request fails, status code : 205. No network route. Please confirm the network. "," Meta ": {" entity ": {" ID ":" 0x00158d00022bc3ab "," type ":" device "," friendlyName ":" 0x00158d00022bc3ab "}," message ":" {\ "state \ 😕" OFF \ "}"}} '
2019-7-2 17:52:42 - Zigbee publish to device '0x00158d00022bc3ab', genOnOff - off - {} -

Re-activation, awakening, etc. does not work. It helped only to exclude the device from the zigbee network and re-enable it.

Other devices Ikea, Gledopto, Nue, Philips, Ksentry Electronics, Livolo network is not lost.

clockbrain, 2019-07-04:
Known issue with Xiaomi end devices. See e.g. [https://www.reddit.com/r/SmartThings/comments/b9qyg7/anyone_using_xiaomi_aqara_sensors_successfully/](https://www.reddit.com/r/SmartThings/comments/b9qyg7/anyone_using_xiaomi_aqara_sensors_successfully/) which relates to SmartThings but the issue is the same for Zigbee2mqtt.

In short they are non-standard and if they miss checking in to the network they leave and don't come back.

Koenkk, 2019-07-06:
Can you try the following firmware? https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190706.zip

I've increased the end device timeout to the max value (11 days).

odx, 2019-07-06:
Just wanted to add my experiences:
I have installed about 30 Aqara contact sensors and four motion sensors. My Zigbee network was down for about one week and I had no issues with devices dropping out of the network.

directman66, 2019-07-08:
> Can you try the following firmware? https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190706.zip
> 
> I've increased the end device timeout to the max value (11 days).

updated firmware cc2531. I continue to observe.

patrul, 2019-08-29:
> Can you try the following firmware? https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190706.zip
> 
> I've increased the end device timeout to the max value (11 days).

@Koenkk I have just installed this firmware. I have many Aqara devices but the ones giving me more problems are the wall switches (QBKG03LM). Sometimes the smart socket (QBCZ11LM), but allways are wired devices, I do not know if this makes sense.

patrul, 2019-08-30:
**UPDATE (+24H):** it continues showing errors. All but one work.
 
```
  zigbee2mqtt:error 8/30/2019, 7:01:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:01:37 PM Failed to ping 'Detector_gas'
  zigbee2mqtt:error 8/30/2019, 7:03:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:04:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:05:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:05:27 PM Zigbee publish to device 'Pulsador_salon_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620503522,"dataType":226}] - {"direction":1,"seqNum":240,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 8/30/2019, 7:06:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:07:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:08:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:09:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:09:37 PM Failed to ping 'Detector_gas'
  zigbee2mqtt:error 8/30/2019, 7:10:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:11:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:12:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:13:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:13:28 PM Zigbee publish to device 'Pulsador_dormitorio_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620504008,"dataType":226}] - {"direction":1,"seqNum":174,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:error 8/30/2019, 7:14:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:15:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:16:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:17:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:18:13 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:19:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:19:09 PM Zigbee publish to device 'Pulsador_salon_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620504343,"dataType":226}] - {"direction":1,"seqNum":255,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 8/30/2019, 7:20:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:21:13 PM Zigbee publish to device 'Pulsador_terraza_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620504468,"dataType":226}] - {"direction":1,"seqNum":207,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 8/30/2019, 7:22:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:23:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:24:14 PM Zigbee publish to device 'Pulsador_terraza_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620504654,"dataType":226}] - {"direction":1,"seqNum":210,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:error 8/30/2019, 7:25:22 PM Zigbee publish to device 'Pulsador_salon_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620504722,"dataType":226}] - {"direction":1,"seqNum":6,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:error 8/30/2019, 7:26:25 PM Zigbee publish to device 'Pulsador_terraza_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620504779,"dataType":226}] - {"direction":1,"seqNum":213,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 8/30/2019, 7:26:31 PM Zigbee publish to device 'Pulsador_salon_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620504785,"dataType":226}] - {"direction":1,"seqNum":7,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 8/30/2019, 7:29:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:29:31 PM Zigbee publish to device 'Pulsador_terraza_pared', genTime - readRsp - [{"attrId":0,"status":0,"attrData":620504966,"dataType":226}] - {"direction":1,"seqNum":216,"disDefaultRsp":1} - 1 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 8/30/2019, 7:30:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:32:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:33:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:34:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:35:07 PM Failed to ping 'Enchufe_multimedia'
[... cut ...]
  zigbee2mqtt:error 8/30/2019, 7:46:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:47:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:50:07 PM Failed to ping 'Enchufe_multimedia'
  zigbee2mqtt:error 8/30/2019, 7:50:21 PM Zigbee publish to device 'Pulsador_despacho_pared', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 2 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
```

Koenkk, 2019-08-31:
@patrul #943 is also interesting for you I guess, QBKG03LM is known to be a fishy device.

patrul, 2019-08-31:
Thanks @Koenkk for that. I do not think we can do so much to continue using these devices, wait 5 seconds to respond it is not practical. Personally, I will search for existing switch alternatives.

closed 2019-11-07 by stale[bot]