Koenkk/zigbee2mqtt #1689: Zigbee 3.0 Konke devices

kirovilya, 2019-07-04:
Konke - is a band of IoT http://www.ikonke.us/ with a cheap zigbee-devices:
Temperature sensor https://ru.aliexpress.com/item/33015776787.html
Contact sensor https://ru.aliexpress.com/item/33015624877.html
Motion sensor https://ru.aliexpress.com/item/33014773963.html
Mutifunctional button https://ru.aliexpress.com/item/33015636907.html

I bought a button and a motion sensor. With the button everything is fine, but with a motion sensor - no.

He gives the model in an unusual way. We had to slightly change the receipt of the model https://github.com/kirovilya/zigbee-shepherd/commit/0d857c2f5362c31765ccab607303871c80879fb4

Also, it is not possible to configure the device to send messages about the motion.
I added configure code https://github.com/kirovilya/zigbee-shepherd-converters/blob/master/devices.js#L4567 :
```
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
            execute(device, actions, callback);
        }
```
But have error in log:
```
2019-07-04T11:32:10.498Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1
2019-07-04T11:32:10.499Z cc-znp:SREQ --> AF:dataRequest, { dstaddr: 1684, destendpoint: 1, srcendpoint: 1, clusterid: 1280, transid: 1, options: 48, radius: 30, len: 14, data: <Buffer 00 01 02 10 00 f0 1e 74 e8 18 00 4b 12 00> }
2019-07-04T11:32:10.508Z cc-znp { sof: 254,  len: 1,  type: 'SRSP',  subsys: 'AF',  cmd: 'dataRequest',  payload: { status: 0 },  fcs: 100,  csum: 100 }
2019-07-04T11:32:10.508Z cc-znp:SRSP <-- AF:dataRequest, { status: 0 }
2019-07-04T11:32:10.508Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2019-07-04T11:32:17.262Z cc-znp { sof: 254,  len: 3,  type: 'AREQ',  subsys: 'AF',  cmd: 'dataConfirm',  payload: { status: 240, endpoint: 1, transid: 1 },  fcs: 55,  csum: 55 }
2019-07-04T11:32:17.263Z cc-znp:AREQ <-- AF:dataConfirm, { status: 240, endpoint: 1, transid: 1 }
2019-07-04T11:32:17.263Z zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: {"status":240,"endpoint":1,"transid":1}
2019-07-04T11:32:17.264Z zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1
2019-07-04T11:32:17.266Z zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'Error: AF data request fails, status code: 240. MAC transaction expired.'
Configure 0x086bd7fffee9826b 3AFE14010402000D Error: AF data request fails, status code: 240. MAC transaction expired.
```
But than I successfully get zoneStatus message:
```
2019-07-04T11:34:57.791Z cc-znp { sof: 254,  len: 29,  type: 'AREQ',  subsys: 'AF',  cmd: 'incomingMsg',  payload:   { groupid: 0,     clusterid: 1280,     srcaddr: 1684,     srcendpoint: 1,     dstendpoint: 1,     wasbroadcast: 0,     linkquality: 126,     securityuse: 0,     timestamp: 15763610,     transseqnumber: 0,     len: 9,     data: <Buffer 09 08 00 01 00 00 17 00 00> },  fcs: 66,  csum: 66 }
2019-07-04T11:34:57.792Z cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1280, srcaddr: 1684, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 126, securityuse: 0, timestamp: 15763610, transseqnumber: 0, len: 9, data: <Buffer 09 08 00 01 00 00 17 00 00> }
2019-07-04T11:34:57.792Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":1684,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":126,"securityuse":0,"timestamp":15763610,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[9,8,0,1,0,0,23,0,0]}}
2019-07-04T11:34:57.794Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-07-04T11:34:57.794Z zigbee:controller event msg { groupid: 0,  clusterid: 1280,  srcaddr: 1684,  srcendpoint: 1,  dstendpoint: 1,  wasbroadcast: 0,  linkquality: 126,  securityuse: 0,  timestamp: 15763610,  transseqnumber: 0,  len: 9,  data:   { '0': 9,     '1': 8,     '2': 0,     '3': 1,     '4': 0,     '5': 0,     '6': 23,     '7': 0,     '8': 0 },  zclMsg:   { frameCntl:      { frameType: 1, manufSpec: 0, direction: 1, disDefaultRsp: 0 },     manufCode: 0,     seqNum: 8,     cmdId: 'statusChangeNotification',     payload: { zonestatus: 1, extendedstatus: 0 } } } { modelId: '3AFE14010402000D' }
Device 0x086bd7fffee9826b incoming event:{"groupid":0,"clusterid":1280,"srcaddr":1684,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":126,"securityuse":0,"timestamp":15763610,"transseqnumber":0,"len":9,"data":{"0":9,"1":8,"2":0,"3":1,"4":0,"5":0,"6":23,"7":0,"8":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":0},"manufCode":0,"seqNum":8,"cmdId":"statusChangeNotification","payload":{"zonestatus":1,"extendedstatus":0}}}
Publish {"linkquality":126}
2019-07-04T11:34:57.795Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":1684,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":126,"securityuse":0,"timestamp":15763610,"transseqnumber":0,"len":9,"data":{"0":9,"1":8,"2":0,"3":1,"4":0,"5":0,"6":23,"7":0,"8":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":0},"manufCode":0,"seqNum":8,"cmdId":"statusChangeNotification","payload":{"zonestatus":1,"extendedstatus":0}}}
2019-07-04T11:34:57.795Z zigbee:controller debug handleMessage { type: 'statusChange',  endpoints:   [ Endpoint {       isLocal: [Function],       device: [Device],       profId: 260,       epId: 1,       devId: 1026,       inClusterList: [Array],       outClusterList: [Array],       clusters: [Ziee],       onAfDataConfirm: null,       onAfReflectError: null,       onAfIncomingMsg: null,       onAfIncomingMsgExt: null,       onZclFoundation: [Function: bound onZclFoundation],       onZclFunctional: null,       foundation: [Function],       functional: [Function],       bind: [Function],       unbind: [Function],       read: [Function],       write: [Function],       report: [Function] } ],  data: { cid: 'ssIasZone', zoneStatus: 1 },  linkquality: 126 }
handleMessage. {"type":"statusChange","endpoints":[{"device":{"_id":3,"type":"EndDevice","ieeeAddr":"0x086bd7fffee9826b","nwkAddr":1684,"manufId":4098,"manufName":"Konke","powerSource":"Unknown","dateCode":"","modelId":"3AFE14010402000D","zclVersion":3,"appVersion":20,"stackVersion":0,"hwVersion":1,"epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":"[Circular]"}},"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[3],"clusters":{"zapp":null,"genBasic":{"dir":{"cid":"genBasic","sid":"dir","value":1},"attrs":{"cid":"genBasic","sid":"attrs"}},"genPowerCfg":{"dir":{"cid":"genPowerCfg","sid":"dir","value":1},"attrs":{"cid":"genPowerCfg","sid":"attrs"}},"genIdentify":{"dir":{"cid":"genIdentify","sid":"dir","value":3},"attrs":{"cid":"genIdentify","sid":"attrs"}},"ssIasZone":{"dir":{"cid":"ssIasZone","sid":"dir","value":1},"attrs":{"cid":"ssIasZone","sid":"attrs"}}},"onAfDataConfirm":null,"onAfReflectError":null,"onAfIncomingMsg":null,"onAfIncomingMsgExt":null,"onZclFunctional":null}],"data":{"cid":"ssIasZone","zoneStatus":1},"linkquality":126}
2019-07-04T11:34:57.796Z zigbee:controller event statusChange { type: 'statusChange',  endpoints:   [ Endpoint {       isLocal: [Function],       device: [Device],       profId: 260,       epId: 1,       devId: 1026,       inClusterList: [Array],       outClusterList: [Array],       clusters: [Ziee],       onAfDataConfirm: null,       onAfReflectError: null,       onAfIncomingMsg: null,       onAfIncomingMsgExt: null,       onZclFoundation: [Function: bound onZclFoundation],       onZclFunctional: null,       foundation: [Function],       functional: [Function],       bind: [Function],       unbind: [Function],       read: [Function],       write: [Function],       report: [Function] } ],  data: { cid: 'ssIasZone', zoneStatus: 1 },  linkquality: 126 } { cid: 'ssIasZone', modelId: '3AFE14010402000D' }
Device 0x086bd7fffee9826b emit event statusChange with data:{"cid":"ssIasZone","zoneStatus":1}
```
(sorry, it is logs from ioBroker.zigbee)

There is also new cmdID:
```
2019-07-04T11:31:57.297Z zigbee-shepherd:af Message has unsupported cmdID: statusChangeNotification
```

Can the device be normally configured, but did not respond to the request? How to handle this situation?

I will fill this issue with my research on these new devices.

tb-killa, 2019-07-04:
Please provide your modification via PR because we need them for other devices too.
Do you add the ``statusChangeNotification`` to your https://github.com/Koenkk/zigbee-shepherd/blob/f6dc5b4469a87c90007dfc9f0d329cdb01d35072/lib/components/af.js#L691 and do your test again?

kirovilya, 2019-07-04:
I completely forgot: it's work only on 15, 20, 25 channels

kirovilya, 2019-07-04:
@tb-killa I added `statusChangeNotification `as you requested. And than also `enrollReq` command.
Attached some logs
[iobroker.2019-07-04.log](https://github.com/Koenkk/zigbee2mqtt/files/3360294/iobroker.2019-07-04.log)

tb-killa, 2019-07-05:
Please test with this:
``
const actions = [
                (cb) => device.bind('ssIasZone', coordinator, cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
``

kirovilya, 2019-07-05:
@tb-killa Thank you, I will try in evening

tb-killa, 2019-07-07:
Hi @kirovilya any news?

kirovilya, 2019-07-07:
@tb-killa get error `TypeError: attrId should be a number or a string.`
I think that the correct format is
`(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),`
but `AF data request fails, status code: 240. MAC transaction expired.` :(

kirovilya, 2019-07-07:
Strange, if I leave only one line in the configuration, it is sometimes successful, and sometimes not.
```
2019-07-07T08:11:03.455Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":34884,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":149,"securityuse":0,"timestamp":2214121,"transseqnumber":0,"len":4,"data":{"0":24,"1":1,"2":4,"3":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":1,"cmdId":"writeRsp","payload":[{"status":0}]}}
2019-07-07T08:11:03.459Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 2
2019-07-07T08:11:03.460Z cc-znp:SREQ --> AF:dataRequest, { dstaddr: 34884, destendpoint: 1, srcendpoint: 1, clusterid: 1280, transid: 2, options: 48, radius: 30, len: 6, data: <Buffer 00 02 0c 00 00 f0> }
2019-07-07T08:11:03.462Z zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'OK'
Successfully configured 0x086bd7fffee9826b 3AFE14010402000D
```
When two lines of configuration, sometimes it falls not on the first, but on the second line:
```
2019-07-07T07:05:05.053Z cc-znp:SREQ --> AF:dataRequest, { dstaddr: 34884, destendpoint: 1, srcendpoint: 1, clusterid: 1280, transid: 2, options: 48, radius: 30, len: 6, data: <Buffer 00 02 0c 00 00 f0> }
2019-07-07T07:05:05.060Z zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'OK'
2019-07-07T07:05:05.082Z cc-znp { sof: 254,  len: 1,  type: 'SRSP',  subsys: 'AF',  cmd: 'dataRequest',  payload: { status: 0 },  fcs: 100,  csum: 100 }
2019-07-07T07:05:05.084Z cc-znp:SRSP <-- AF:dataRequest, { status: 0 }
2019-07-07T07:05:05.085Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2019-07-07T07:05:05.370Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 3
2019-07-07T07:05:05.372Z cc-znp:SREQ --> AF:dataRequest, { dstaddr: 34884, destendpoint: 1, srcendpoint: 1, clusterid: 1280, transid: 3, options: 48, radius: 30, len: 5, data: <Buffer 01 03 00 00 17> }
2019-07-07T07:05:05.386Z cc-znp { sof: 254,  len: 1,  type: 'SRSP',  subsys: 'AF',  cmd: 'dataRequest',  payload: { status: 0 },  fcs: 100,  csum: 100 }
2019-07-07T07:05:05.388Z cc-znp:SRSP <-- AF:dataRequest, { status: 0 }
2019-07-07T07:05:05.389Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2019-07-07T07:05:06.007Z cc-znp { sof: 254,  len: 3,  type: 'AREQ',  subsys: 'AF',  cmd: 'dataConfirm',  payload: { status: 240, endpoint: 1, transid: 3 },  fcs: 53,  csum: 53 }
2019-07-07T07:05:06.010Z cc-znp:AREQ <-- AF:dataConfirm, { status: 240, endpoint: 1, transid: 3 }
2019-07-07T07:05:06.011Z zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: {"status":240,"endpoint":1,"transid":3}
2019-07-07T07:05:06.014Z zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 3
2019-07-07T07:05:06.018Z zigbee-shepherd-converters:devices Configured '(cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb)' with result 'Error: AF data request fails, status code: 240. MAC transaction expired.'
Configure 0x086bd7fffee9826b 3AFE14010402000D Error: AF data request fails, status code: 240. MAC transaction expired.
Dev 0x086bd7fffee9826b 3AFE14010402000D not configured yet, will try again in latest 300 sec.
```

kirovilya, 2019-07-07:
it seems the sensor goes to sleep does not accept configuration commands. only sometimes, when he is active, is it possible to transfer a command to it.

kirovilya, 2019-07-07:
One of the assumptions. Such devices must be configured at the time when they themselves send a message to coordinator. It may be possible at this moment to send them a command.

For example, there were first messages from a device with the `enrollRsp `type, then they were replaced with the `cmdEnrollReq `type. Now comes `statusChangeNotification `at motion detection.
Those. as if configuration messages arrive, but not immediately.

kirovilya, 2019-07-07:
Also with `genPowerCfg`. The command is sent, but the response is not waiting. Then come the data on the battery.
```
2019-07-07 12:56:41.126  - [34mdebug[39m: zigbee.0 2019-07-07T09:56:41.126Z zigbee-shepherd:request REQ --> ZDO:bindReq
2019-07-07 12:56:41.128  - [34mdebug[39m: zigbee.0 2019-07-07T09:56:41.128Z cc-znp:SREQ --> ZDO:bindReq, { dstaddr: 34884, srcaddr: '0x086bd7fffee9826b', srcendpoint: 1, clusterid: 1, dstaddrmode: 3, addr_short_long: '0x00124b0018e87346', dstendpoint: 1 }
2019-07-07 12:56:41.141  - [34mdebug[39m: zigbee.0 2019-07-07T09:56:41.141Z cc-znp { sof: 254,  len: 1,  type: 'SRSP',  subsys: 'ZDO',  cmd: 'bindReq',  payload: { status: 0 },  fcs: 69,  csum: 69 }
2019-07-07 12:56:41.142  - [34mdebug[39m: zigbee.0 2019-07-07T09:56:41.142Z cc-znp:SRSP <-- ZDO:bindReq, { status: 0 }
2019-07-07 12:56:51.128  - [34mdebug[39m: zigbee.0 2019-07-07T09:56:51.128Z zigbee-shepherd:request REQ --> ZDO:bindReq
2019-07-07 12:56:51.130  - [34mdebug[39m: zigbee.0 2019-07-07T09:56:51.130Z cc-znp:SREQ --> ZDO:bindReq, { dstaddr: 34884, srcaddr: '0x086bd7fffee9826b', srcendpoint: 1, clusterid: 1, dstaddrmode: 3, addr_short_long: '0x00124b0018e87346', dstendpoint: 1 }
2019-07-07 12:56:51.147  - [34mdebug[39m: zigbee.0 2019-07-07T09:56:51.147Z cc-znp { sof: 254,  len: 1,  type: 'SRSP',  subsys: 'ZDO',  cmd: 'bindReq',  payload: { status: 0 },  fcs: 69,  csum: 69 }
2019-07-07 12:56:51.151  - [34mdebug[39m: zigbee.0 2019-07-07T09:56:51.150Z cc-znp:SRSP <-- ZDO:bindReq, { status: 0 }
2019-07-07 12:57:01.135  - [34mdebug[39m: zigbee.0 2019-07-07T09:57:01.134Z zigbee-shepherd-converters:devices Configured '(cb) => device.bind('genPowerCfg', coordinator, cb)' with result 'Error: Timed out after 10000 ms'
2019-07-07 12:57:01.136  - [34mdebug[39m: zigbee.0 Configure 0x086bd7fffee9826b 3AFE14010402000D Error: Timed out after 10000 ms
2019-07-07 12:57:01.137  - [33mwarn[39m: zigbee.0 Dev 0x086bd7fffee9826b 3AFE14010402000D not configured yet, will try again in latest 300 sec.
2019-07-07 13:00:21.056  - [34mdebug[39m: zigbee.0 2019-07-07T10:00:21.056Z cc-znp { sof: 254,  len: 27,  type: 'AREQ',  subsys: 'AF',  cmd: 'incomingMsg',  payload:    { groupid: 0,     clusterid: 1,     srcaddr: 34884,     srcendpoint: 1,     dstendpoint: 1,     wasbroadcast: 0,     linkquality: 102,     securityuse: 0,     timestamp: 1553966,     transseqnumber: 0,     len: 7,     data: <Buffer 08 7e 0a 20 00 20 1c> },  fcs: 76,  csum: 76 }
2019-07-07 13:00:21.058  - [34mdebug[39m: zigbee.0 2019-07-07T10:00:21.058Z cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1, srcaddr: 34884, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 102, securityuse: 0, timestamp: 1553966, transseqnumber: 0, len: 7, data: <Buffer 08 7e 0a 20 00 20 1c> }
2019-07-07 13:00:21.060  - [34mdebug[39m: zigbee.0 2019-07-07T10:00:21.060Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1,"srcaddr":34884,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":1553966,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,126,10,32,0,32,28]}}
2019-07-07 13:00:21.075  - [34mdebug[39m: zigbee.0 2019-07-07T10:00:21.075Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-07-07 13:00:21.079  - [34mdebug[39m: zigbee.0 2019-07-07T10:00:21.078Z zigbee:controller event msg { groupid: 0,  clusterid: 1,  srcaddr: 34884,  srcendpoint: 1,  dstendpoint: 1,  wasbroadcast: 0,  linkquality: 102,  securityuse: 0,  timestamp: 1553966,  transseqnumber: 0,  len: 7,  data: { '0': 8, '1': 126, '2': 10, '3': 32, '4': 0, '5': 32, '6': 28 },  zclMsg:    { frameCntl: { frameType: 0, manufSpec: 0, direction: 1, disDefaultRsp: 0 },     manufCode: 0,     seqNum: 126,     cmdId: 'report',     payload: [ [Object] ] } } { modelId: '3AFE14010402000D' }
2019-07-07 13:00:21.081  - [34mdebug[39m: zigbee.0 Device 0x086bd7fffee9826b incoming event:{"groupid":0,"clusterid":1,"srcaddr":34884,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":1553966,"transseqnumber":0,"len":7,"data":{"0":8,"1":126,"2":10,"3":32,"4":0,"5":32,"6":28},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":0},"manufCode":0,"seqNum":126,"cmdId":"report","payload":[{"attrId":32,"dataType":32,"attrData":28}]}}
2019-07-07 13:00:21.082  - [34mdebug[39m: zigbee.0 Publish {"linkquality":102}
2019-07-07 13:00:21.085  - [34mdebug[39m: zigbee.0 2019-07-07T10:00:21.084Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1,"srcaddr":34884,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":1553966,"transseqnumber":0,"len":7,"data":{"0":8,"1":126,"2":10,"3":32,"4":0,"5":32,"6":28},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":0},"manufCode":0,"seqNum":126,"cmdId":"report","payload":[{"attrId":32,"dataType":32,"attrData":28}]}}
2019-07-07 13:00:21.097  - [34mdebug[39m: zigbee.0 2019-07-07T10:00:21.093Z zigbee:controller debug handleMessage { type: 'attReport',  endpoints:    [ Endpoint {       isLocal: [Function],       device: [Object],       profId: 260,       epId: 1,       devId: 1026,       inClusterList: [Array],       outClusterList: [Array],       clusters: [Object],       onAfDataConfirm: null,       onAfReflectError: null,       onAfIncomingMsg: null,       onAfIncomingMsgExt: null,  [... cut ...]
2019-07-07 13:02:01.133  - [34mdebug[39m: zigbee.0 2019-07-07T10:02:01.133Z zigbee-shepherd-converters:devices Configured '(cb) => device.bind('genPowerCfg', coordinator, cb)' with result 'Error: Timed out after 10000 ms'
2019-07-07 13:02:01.134  - [34mdebug[39m: zigbee.0 Configure 0x086bd7fffee9826b 3AFE14010402000D Error: Timed out after 10000 ms
2019-07-07 13:02:01.134  - [33mwarn[39m: zigbee.0 Dev 0x086bd7fffee9826b 3AFE14010402000D not configured yet, will try again in latest 300 sec.
```

tb-killa, 2019-07-10:
Any news on this?
I buy two devices, but they will arrived sadly at the end of this month.

kirovilya, 2019-07-11:
@tb-killa No news. :(
I added converters for these devices to my converters-fork, but I haven’t done the PR yet as I couldn't correctly configure the devices. Only as I described above - performed one configure-message at a time.

tb-killa, 2019-07-13:
Please provide your ``database.db`` content for the the ias device.
Maybe it support the polling cluster?
Or we have do some bad magic trick like hold device online by flooding with basic readinga and do our configuration part between this time slots.

kirovilya, 2019-07-15:
[shepherd.zip](https://github.com/Koenkk/zigbee2mqtt/files/3391053/shepherd.zip)

fairecasoimeme, 2019-07-17:
For motion sensor and door sensor, you have to manage 0x500 cluster with enrollment mechanism. 
The most important things that the two sensors talk to 0x15 endpoint.
you can see some traces here :
http://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home/

kirovilya, 2019-07-17:
@fairecasoimeme Great! Thanks for the detailed investigation! I read.

anshgh, 2019-07-19:
Hi
Why are Konke devices missing on this page?
http://www.zigbee2mqtt.io/information/supported_devices.html

kirovilya, 2019-07-19:
@anshgh Cause it's not complete yet

I tried to add endpoint 21 to the coordinator - I started receiving requests for cmdEnrollReq. Now need to learn how to respond to them from the coordinator.

tb-killa, 2019-07-20:
@kirovilya Any News ? My Devices arrived and i do some tests with manual adding ep15 to coordinator but this doesn´t seemt to work in right way.

Maybe we should forward ep:15 to ep:1 as we do with others via https://github.com/Koenkk/Z-Stack-firmware/commit/265ca6dc9027b61efd182b0cfaf55c706827a3dd#diff-722c0d664df72aed6df38b27a738c95eR74

@Koenkk Do you could compile Test-Firmware where we also forward EP:15 to EP:1 ?
This could maybe help to integrate this Product.

Koenkk, 2019-07-20:
15 or 21? https://github.com/Koenkk/zigbee2mqtt/issues/1689#issuecomment-513146534

tb-killa, 2019-07-21:
And that's the subtlety. Instead of addressing endpoint 0x01 (normally the coordinator), it sends its request to 0x15 .

Source from https://github.com/Koenkk/zigbee2mqtt/issues/1689#issuecomment-512192524

tb-killa, 2019-07-21:
@kirovilya @Koenkk 
If Devices joined and doing push on the motion sensor i got sadly this:

````
zigbee2mqtt:debug 2019-7-21 10:57:35 Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Konke","modelId":"3AFE14010402000D","powerSource":0}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^

TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:143:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:143:14)
    at Zigbee.onMessage (/opt/zigbee2mqtt/lib/zigbee.js:262:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:211:14)
```

As the Device use EP:1 for default i don´t know why this happened.

tb-killa, 2019-07-21:
@Koenkk do you know why this could happend ?
```
zigbee2mqtt:debug 2019-7-21 11:39:52 Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Konke","modelId":"3AFE14010402000D","powerSource":0}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^

TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:143:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:143:14)
    at Zigbee.onMessage (/opt/zigbee2mqtt/lib/zigbee.js:262:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:211:14)
```

Koenkk, 2019-07-21:
@kirovilya could you modify the firmware to also forward ep15?

@tb-killa looks like something is wrong with your manual zigbee-shepherd-covnerters setup, probably the format is not OK.

tb-killa, 2019-07-21:
@Koenkk I think this doesn´t seems to be anything with the converters.

```
zigbee2mqtt:debug 2019-7-21 16:26:18 Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":28}}' of device 'undefined' (0x086bd7fffee97be7) of endpoint 1
  zigbee2mqtt:warn 2019-7-21 16:26:18 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-7-21 16:26:18 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:debug 2019-7-21 16:26:18 Received zigbee message of type 'endDeviceAnnce' with data '"0x086bd7fffee97be7"' of device 'undefined' (0x086bd7fffee97be7) of endpoint 1
  zigbee2mqtt:warn 2019-7-21 16:26:18 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-7-21 16:26:18 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:debug 2019-7-21 16:26:50 Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Konke","modelId":"3AFE14010402000D","powerSource":0}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^

TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:143:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:143:14)
    at Zigbee.onMessage (/opt/zigbee2mqtt/lib/zigbee.js:262:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:216:14)
```

tb-killa, 2019-07-21:
After Restart and Rejoin the Same (now with ``attrReport``)
```
zigbee2mqtt:debug 2019-7-21 17:22:47 Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":28}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^

TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:143:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:143:14)
    at Zigbee.onMessage (/opt/zigbee2mqtt/lib/zigbee.js:262:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:184:14)
```

tb-killa, 2019-07-21:
Got new Messages =)

@Koenkk @kirovilya Maybe we doesn´t need some tweaks on firmware side because the ``iasZone`` is also in EP:1 by Default.

My devices.js entry

```
// Konke
	{
        zigbeeModel: ['3AFE14010402000D'],
        model: '3AFE14010402000D',
        vendor: 'Konke',
        description: 'Motion Sensor',
        supports: 'motion',
        fromZigbee: [fz.generic_batteryvoltage_3000_2500, fz.ignore_power_change,
fz.ignore_basic_change, fz.ignore_genIdentify_change],
        toZigbee: [],
		configure: (ieeeAddr, shepherd, coordinator, callback) => {
             const device = shepherd.find(ieeeAddr, 1);
             const actions = [
                 //(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
		(cb) => device.bind('ssIasZone', coordinator, cb),
		(cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
		(cb) => device.report('ssIasZone', 'zoneStatus', 0, 1000, null, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryVoltage', repInterval.HOUR, repInterval.MAX, cb),
             ];
             execute(device, actions, callback);
         }
    },
```

Trick: Push the little Button inside the Motion Sensor Multiple Times to Trigger the "Configuration" Part and wait for the Response.

```
2019-7-21 19:03:07 - debug: Received zigbee message of type 'devInterview' with data '"0x086bd7fffee97be7"'
2019-7-21 19:03:11 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
2019-7-21 19:03:11 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
2019-7-21 19:03:11 - warn: No converter available for '3AFE14010402000D' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}'
2019-7-21 19:03:11 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-7-21 19:03:11 - debug: Received zigbee message of type 'devInterview' with data '"0x086bd7fffee97be7"'
2019-7-21 19:03:15 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":13,"zoneStatus":205,"iasCieAddr":"0x0000000000000000","zoneId":255}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
2019-7-21 19:03:15 - debug: Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":13,"zoneStatus":205,"iasCieAddr":"0x0000000000000000","zoneId":255}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
2019-7-21 19:03:15 - warn: No converter available for '3AFE14010402000D' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":13,"zoneStatus":205,"iasCieAddr":"0x0000000000000000","zoneId":255}}'
2019-7-21 19:03:15 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-7-21 19:03:17 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"65533":1}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
2019-7-21 19:03:17 - debug: Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"65533":1}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
2019-7-21 19:03:17 - warn: No converter available for '3AFE14010402000D' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"65533":1}}'
2019-7-21 19:03:17 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-7-21 19:03:17 - debug: Received zigbee message of type 'devInterview' with data '"0x086bd7fffee97be7"'
zigbee-shepherd:af Message has unsupported cmdID: enrollReq
```

After adding the cmdID and Restart i got this after first Push on the EndDevice:

```
zigbee2mqtt:debug 2019-7-21 19:14:13 Received zigbee message of type 'cmdEnrollReq' with data '{"cid":"ssIasZone","data":{"zonetype":13,"manucode":4098}}' of device '3AFE14010402000D' (0x086bd7fffee97be7) of endpoint 1
  zigbee2mqtt:warn 2019-7-21 19:14:13 No converter available for '3AFE14010402000D' with cid 'ssIasZone', type 'cmdEnrollReq' and data '{"cid":"ssIasZone","data":{"zonetype":13,"manucode":4098}}'
  zigbee2mqtt:warn 2019-7-21 19:14:13 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Now we need to do the ``cmdEnrollReq`` Response ?
Ideas ?

tb-killa, 2019-07-21:
Okay i think i got it:
``cmdStatusChangeNotification`` represent the ``devChange`` from ``ssIasZone``

@Koenkk could we add multiple ``cid`` to the ``fromzigbee.js`` converter or have we build for every ``cid`` own ones ?
If last we have to ignore the normal ``devChange`` and use only the ``cmdStatusChangeNotification`` for our converter message.

tb-killa, 2019-07-21:
The Motion Sensor only Report True on Motion, it doesn´t report false.
I modify the bitron motion converter and use them for this device too.

tb-killa, 2019-07-21:
Will publish the end version of the ``devices.js`` entry and the modifications for ``af.js`` and the ``converter`` soon.

kirovilya, 2019-07-22:
Sorry, I was busy.
I tried made [Zive app](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zapp/cie.js#L15) with handlers:
```
// Init Command Response Handler
cieClusters.init('ssIasZone', 'cmdRsps', {
    enrollReq: (zapp, data, cb) => {
        zapp.functional(
            data.src.ieeeAddr,
            data.src.epId,
            'ssIasZone',
            'enrollRsp',
            {enrollrspcode: 0x00, zoneid : 6},
            {direction: 0},
            cb
        );
    },
    statusChangeNotification: (zapp, argObj, cb) => {},
});
```
and mount it to the 21 (0x15) endpoint (some [rewrite mount method here](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/shepherd.ts#L362) and parameters):
```
this.shepherd.mount(cieApp, 21, (err, epId) => {
    if (!err) {
        this.debug(`Mounted the cieApp (epId ${epId})`);
    } else {
        this.error(`Failed to mount the cieApp`);
    }
});
```
Now I request cmdEnrollReq and send enrollRsp from 21 (0x15) endpoint.

It worked for the contact sensor. But it did not work so well for the motion sensor.
Contact sensor send data to endpoint 1:
```
zigbee:controller event cmdStatusChangeNotification { type: 'cmdStatusChangeNotification',  endpoints:    [ Endpoint {       isLocal: [Function],       device: [Object],       profId: 260,       epId: 1,       devId: 1026,       inClusterList: [Array],       outClusterList: [Array],       clusters: [Object],       onAfDataConfirm: null,       onAfReflectError: null,       onAfIncomingMsg: null,       onAfIncomingMsgExt: null,       onZclFoundation: [Function: bound onZclFoundation],       onZclFunctional: null,       foundation: [Function],       functional: [Function],       bind: [Function],       unbind: [Function],       read: [Function],       write: [Function],       report: [Function] } ],  data: { cid: 'ssIasZone', data: { zonestatus: 1, extendedstatus: 0 } },  linkquality: 157,  groupid: 0 } { cid: 'ssIasZone', modelId: '3AFE130104020015' }
Device 0x086bd7fffee96d7c emit event cmdStatusChangeNotification with data:{"cid":"ssIasZone","data":{"zonestatus":1,"extendedstatus":0}}
No converter available for '2AJZ4KPDR' with cid 'ssIasZone' and type 'cmdStatusChangeNotification'
```
But motion sensor send to endpoint 21 (0x15) and not found model :( :
```
zigbee:controller debug handleMessage { type: 'cmdStatusChangeNotification',  endpoints:    [ Coordpoint {       isLocal: [Function],       device: [Object],       profId: 260,       epId: 21,       devId: 1024,       inClusterList: [],       outClusterList: [Array],       clusters: [Object],       onAfDataConfirm: null,       onAfReflectError: null,       onAfIncomingMsg: null,       onAfIncomingMsgExt: null,       onZclFoundation: [Function: bound onZclFoundation],       onZclFunctional: [Function],       isDelegator: [Function],       foundation: [Function],       functional: [Function],       bind: [Function],       unbind: [Function],       read: [Function],       write: [Function],       report: [Function] } ],  data: { cid: 'ssIasZone', data: { zonestatus: 1, extendedstatus: 0 } },  linkquality: 123,  groupid: 0 }
zigbee:controller debug Message without modelId! undefined
```

bedenko, 2019-08-04:
@kirovilya: could you elaborate on how did you manage to pair Konke button?
I flashed CC2531ZNP-with-SBL.hex (from CC2531_20190425.zip).
I held reset button until led began to flash, but i get nothing in console. Not even with DEBUG=* flag set.

kirovilya, 2019-08-04:
it's work only on 15, 20, 25 channels

bedenko, 2019-08-04:
Ouch, i missed that part.
It is working now. Thank you!

elpaul, 2019-08-12:
I have a problem with the button. Sometimes it works well for a long time, and sometimes it stops working and then I see something in the logs:
`zigbee2mqtt:info 8/12/2019, 5:53:54 AM MQTT publish: topic 'zigbee2mqtt/0x086bd7fffebfe5ee', payload '{"linkquality":76,"battery":"NaN","voltage":null}'`

kirovilya, 2019-08-13:
Hi! I made changes https://github.com/kirovilya/zigbee-shepherd/commit/a4c71f5715a4461be886b1abfeca09ccd7e725d8 and https://github.com/kirovilya/zigbee-shepherd-converters/commit/8090ad0bc5511bb09cc0e8f08a00ab8840d69041

Can anyone else besides me verify the correct pairing and working of Konke motion sensor and contact?

For me It's began to pair and work correctly with these changes.

zhapwey, 2019-08-13:
good I have a konke temperature sensor is already done to include it in the library or is it pending?
What can I get with him

Best Regards

kirovilya, 2019-08-13:
@zhapwey temperature sensor is already works, but in dev-branch of z2m

else, 2019-08-18:
> Can anyone else besides me verify the correct pairing and working of Konke motion sensor and contact?

Could you provide the steps to build it? What version of zigbee2mqtt do I need to check out?

I have a motion sensor here and am interested in getting it mainlined.

elsbrock, 2019-09-08:
@kirovilya highlight in case you missed my reply

kirovilya, 2019-09-09:
@elsbrock Now there is no version of z2m that can work with this sensor. If you can, make changes to the code described above. I checked, but my sensor is not always configuring correctly.

elsbrock, 2019-09-09:
I am aware, was trying to understand how I can apply your changes to zigbee2mqtt, because it looks like one of the libraries was changed?

kirovilya, 2019-09-10:
@elsbrock You must find and modify the code lines in the same files in your libraries.
Nobody can confirm this changes, so I not to do PR to zigbee-shepherd for z2m.
after the moving to a new version of the zigbee-herdsman, I will do PR to it.

paulgking, 2019-09-15:
@kirovilya I've applied similar changes to my setup for the motion sensor and so far it appears to be working, has registered ok and I get notifications on sensor activation. I have done this for two sensors so appears to not be just blind luck.

Though no homeassistant MQTT messages, and the battery level I suspect needs work, both reporting 2800 voltage and 60.00%, hard to tell if this is accurate or not, but as both just out of the box would seem odd they are precisely the same level run down.

paulgking, 2019-09-15:
Subsequent to a restart or two, it is no longer working, getting:
```
/app/lib/extension/deviceReceive.js:109

                if (c.cid === cid) {

                      ^

TypeError: Cannot read property 'cid' of undefined

    at mappedDevice.fromZigbee.filter (/app/lib/extension/deviceReceive.js:109:23)

    at Array.filter (<anonymous>)

    at DeviceReceive.onZigbeeMessage (/app/lib/extension/deviceReceive.js:105:52)

    at extensions.filter.forEach (/app/lib/controller.js:143:31)

    at Array.forEach (<anonymous>)

    at Controller.onZigbeeMessage (/app/lib/controller.js:143:14)

    at Zigbee.onMessage (/app/lib/zigbee.js:275:18)

```

removing the devices seems to stop that, so I assume they aren't setting up properly.

kirovilya, 2019-09-19:
@paulgking I will rewrite the code a bit, after switching to zigbee-herdsman. And make PR

paulgking, 2019-09-19:
ok, let me know if you need anyone to test anything

tb-killa, 2019-09-19:
@kirovilya I´m on herdsman and sadly my work didn´t work anymore stable.
The first ones are that the Device Infos doesnt seems to be write to database.db file complete.
If i use the motion sensor the pairing process themselve doesnt work correctly to allow the finish with modelID and some other Values.
I think this seems to be a strange bug in herdsman because the IAS Join Process seems not work correctly too with other devices like bitron devices.

paulgking, 2019-09-20:
I've now sniffed the traffic between the konke hub and a motion sensor. I'm not seeing anything unusual. 

Device joins network.
Hub writes IASCIEAddress
Sensor Responds ok
Hub reads IASCIEAddress
Sensor Responds
Hub sends EnrollRequest
Sensor Response to EnrollRequest
Hub reads attributes (Zone State, Zone Type, Zone Status)
Sensor Responds State:Enrolled, Type: MotionSensor, Status 0

I can't see endpoint 21 anywhere, it's all on endpoint 1.

I'll reconfiqure the shepard-converters to the simple form and try sniffing that, probably tomorrow now.

I

paulgking, 2019-09-20:
Update on that I've traced trying to join to Zigbee2Mqtt with a basic setup.

This time the message setup is quite different, I never see the output IASCIEAddress stuff, but I do see an incoming Enroll Request on endpoint 21. 

Not clear why we'd get this different behaviour. 

One thing I note is that with the Konke hub, after joining the hub does send a "Match Descriptor" request for IAS and the response acknowledges specifying endpoint 1.

That isn't the same for Zigbee2Mqtt where we get a simple descriptor request and response, the response includes IAS as an input cluster.

This is beyond my understanding at this point, but will keep trying chip away.

tb-killa, 2019-09-21:
hi @kirovilya @paulgking 

I push some changes from my working implementation.
i test with motion and contact sensors at now more than 5 days and it seems to be stable.
please check and do test again and report.

https://github.com/Koenkk/zigbee-shepherd-converters/pull/628
https://github.com/Koenkk/zigbee-herdsman/pull/34
https://github.com/Koenkk/zigbee-herdsman/issues/35

tb-killa, 2019-09-22:
Does someone who own the konke hub too could check if the motion sensor show some sort of blue led trigger on motion?

With Z2M we see the motion only in converter state but would be interested if this could work too.

paulgking, 2019-09-22:
@tb-killa The konke app is pretty limited so no options to set, not even a battery level display for the motion sensor. No visible output on the sensor itself when triggered, just at Zone Status Change Notification message, with "Alarm 1" set.

paulgking, 2019-09-29:
Had this up and running for a few days now. One of the motion sensors didn't seem to register properly and was only reporting link quality and battery, at some point it started working, so perhaps to 100% but working reliably now. 

Thanks.

paulgking, 2019-10-05:
I noticed both my two motion sensors were saying voltage 2900 battery 80.00 out of the box, so did some experimentation with one of them.

Testing up to a bench power supply and seeing what I got with one of my devices (the other I assume will be the same)

At 3.00v it was reading voltage 2800, battery 60%. This is the nominal voltage of a new battery (in reality they are slightly higher, hence my 2900 80%), so it's clearly not reading very well.

At 2.70v it's reading voltage 2500, battery 0%.

At 2.50v it's reading voltage 2300, battery 0% (Not sure if this would work with a real battery, since the power supply can generate the current requirements, don't know enough about the batteries to know if it could, though it seems to be less the 20ma drawn at startup - maybe higher this is just by eye on a meter)

At 2.75v it's reading voltage 2600, battery 20%

At no stage did I see the low battery indicator in the alarm messages.

So basic observation is that it's reading somewhere around 0.2v low, the % steps are 20%. The low battery indicator doesn't work and the % remaining is pretty misleading.

Given the konke hub/app don't show battery on these, my guess would be the battery stuff simply hasn't been setup/calibrated and is likely just a side effect of the code they based theirs on - quite possibly setup for a higher initial voltage battery. 

So I doubt anyone should be relying on the battery values it gives for anything meaningful, later versions/revisions of the motion sensor firmware could resolve this I guess.

tb-killa, 2019-10-05:
I thought battery percentage are from Z2M internal calculation per Converter? We should check them up because we report only voltage. Maybe we need some tweak for this.
But thanks for the response because this show up the devices work with Z2M.

paulgking, 2019-10-05:
Good point, I hadn't considered we maybe generating the %. I've done some further testing and it works down to 2.1v from a power suppy reading 2000 for voltage at that stage. At 2v it stops starting up. So if it is Z2M computing % we can probably do something better, though I doubt it scales linearly.

Still the low battery flag in the alert never happens.

tb-killa, 2019-10-05:
It's here https://github.com/Koenkk/zigbee-shepherd-converters/blob/b38f3db77e8bf82526454057e3ca76785fed3489/converters/fromZigbee.js#L1456

This converter is used for the motion actually.
Please test and tweak so we could prepare an PR.

ukrtrip, 2019-10-15:
Hello! I purchased a few Konke sensors on aliexpress.
The motion sensor has a model 2AJZ4-KPBS and is detected by the Z2M as '3AFE27010402000D'. I added it to the devices.js - It works correctly:
(LOG: _MQTT publish: topic 'zigbee2mqtt/KonkeMotionSensor', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":42,"battery":"100.00","voltage":3000}'_)
devices.js: 
```
{
    zigbeeModel: ['3AFE14010402000D', '3AFE27010402000D'],
    model: '2AJZ4KPBS',
    vendor: 'Konke',
    description: 'Motion sensor',
    supports: 'occupancy',
    fromZigbee: [fz.iaszone_occupancy_1_with_timeout, fz.battery_3V],
    toZigbee: [],
},
```

Please add the '3AFE27010402000D' zigbeeModel to the main (and dev) repository.
This is tested in the last dev repository (zigbee-herdsman: 0.9.2, zigbee-herdsman-converters: 11.1.24)

[22 more replies]

closed 2020-03-13 by stale[bot]