Koenkk/zigbee2mqtt #1690: Generic motion sensor RH3040 (JIANSHU..?) Zigbee 3.0 support

codegraffiti, 2019-07-04:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

`7/4/2019,
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/3359789/database.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3359790/log.txt)

 8:13:43 PM - info: Logging to directory: 'C:\opt\zigbee2mqtt\data\log\2019-07-04.20-13-43'
7/4/2019, 8:13:43 PM - debug: Removing old log directory 'C:\opt\zigbee2mqtt\data\log\2019-06-27.23-01-27'
7/4/2019, 8:13:43 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"C:\\opt\\zigbee2mqtt\\data\\database.db","coordBackupPath":"C:\\opt\\zigbee2mqtt\\data\\coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
7/4/2019, 8:13:43 PM - debug: Loaded state from file C:\opt\zigbee2mqtt\data\state.json
7/4/2019, 8:13:43 PM - debug: Saving state to file C:\opt\zigbee2mqtt\data\state.json
7/4/2019, 8:13:43 PM - info: Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
7/4/2019, 8:13:43 PM - info: Starting zigbee-shepherd
7/4/2019, 8:13:45 PM - info: zigbee-shepherd started
7/4/2019, 8:13:45 PM - info: Coordinator firmware version: '20190425'
7/4/2019, 8:13:45 PM - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b000e896b65","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190425},"startTime":1562251425,"joinTimeLeft":0}
7/4/2019, 8:13:45 PM - info: Currently 0 devices are joined:
7/4/2019, 8:13:45 PM - warn: `permit_join` set to  `true` in configuration.yaml.
7/4/2019, 8:13:45 PM - warn: Allowing new devices to join.
7/4/2019, 8:13:45 PM - warn: Set `permit_join` to `false` once you joined all devices.
7/4/2019, 8:13:45 PM - info: Zigbee: allowing new devices to join.
7/4/2019, 8:13:45 PM - info: Connecting to MQTT server at mqtt://192.168.0.3
7/4/2019, 8:13:45 PM - info: zigbee-shepherd ready
7/4/2019, 8:13:45 PM - info: Connected to MQTT server
7/4/2019, 8:13:45 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
7/4/2019, 8:13:45 PM - debug: Mounted the cieApp (epId 11)
7/4/2019, 8:13:45 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"927c4db","coordinator":20190425,"log_level":"debug","permit_join":true}'
7/4/2019, 8:14:00 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"TUYATEC-53o41joc","modelId":"RH3040","powerSource":3}}' of device 'undefined' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:00 PM - warn: Device with modelID 'undefined' is not supported.
7/4/2019, 8:14:00 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:02 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"20180512"}}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:02 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:14:02 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:02 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"appVersion":67}}' of device 'undefined' (0x00124b000e896b65) of endpoint 11
7/4/2019, 8:14:02 PM - debug: Ignoring message from coordinator
7/4/2019, 8:14:02 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"appVersion":67}}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:02 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:14:02 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:02 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":30,"batteryPercentageRemaining":200}}' of device 'undefined' (0x00124b000e896b65) of endpoint 11
7/4/2019, 8:14:02 PM - debug: Ignoring message from coordinator
7/4/2019, 8:14:02 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":30,"batteryPercentageRemaining":200}}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:02 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:14:02 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:05 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:05 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:14:05 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:07 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:07 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:14:07 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:07 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":67,"stackVersion":0,"hwVersion":1,"manufacturerName":"TUYATEC-53o41joc"}}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:07 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:14:07 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:08 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:08 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:14:08 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:08 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"appVersion":67}}' of device 'undefined' (0x00124b000e896b65) of endpoint 11
7/4/2019, 8:14:08 PM - debug: Ignoring message from coordinator
7/4/2019, 8:14:08 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"appVersion":67}}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:14:08 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:14:08 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:14:08 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":30,"batteryPercentageRemaining":200}}' of device 'undefined' (0x00124b000e896b65) of endpoint 11
[... 54 lines cut ...]
7/4/2019, 8:15:20 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:15:22 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":1}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:15:22 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:15:22 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:15:23 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":1}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:15:23 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:15:23 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:15:25 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":1}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:15:25 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:15:25 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:16:20 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
[... cut ...]
7/4/2019, 8:16:25 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'RH3040' (0x000d6ffffe816e45) of endpoint 1
7/4/2019, 8:16:25 PM - warn: Device with modelID 'RH3040' is not supported.
7/4/2019, 8:16:25 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
7/4/2019, 8:17:05 PM - info: Succesfully reenabled joining
7/4/2019, 8:18:43 PM - debug: Saving state to file C:\opt\zigbee2mqtt\data\state.json
7/4/2019, 8:20:25 PM - info: Succesfully reenabled joining`

Koenkk, 2019-07-05:
Do you get a message when triggering the occupancy sensor?

codegraffiti, 2019-07-06:
Hello @Koenkk ,

I did some further investigation with the help of your wiki and below are the things added to different files

**devices.js**

`{
        zigbeeModel: ['RH3040'],
        model: 'RH3040 - TUYATEC Motion sensor',
        vendor: 'TUYATEC',
        description: 'Motion sensor',
        supports: 'occupancy ',
        fromZigbee: [
            fz.RH3040_motion_dev_change,
            fz.RH3040_motion_status_change,
            fz.RH3040_battery,
            fz.RH3040_battery_change,
            fz.RH3040_voltage,
            fz.RH3040_voltage_change,
            fz.ignore_basic_report,
        ],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.bind('genBasic', coordinator, cb),
                (cb) => device.bind('ssIasZone', coordinator, cb),
                (cb) => device.bind('genIdentify', coordinator, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
//                (cb) => device.report('genPowerCfg', 'batteryVoltage','batteryPercentageRemaining', 1, 1000, 1, cb),
            ];
            execute(device, actions, callback);
        },
    },
];

**fromZigbee.js**

RH3040_battery: {
        cid: 'genPowerCfg',
        type: ['attReport', 'readRsp'],
        convert: (model, msg, publish, options) => {
            if (msg.data.data.hasOwnProperty('batteryPercentageRemaining')) {
                return {battery: msg.data.data['batteryPercentageRemaining'] *100 / 200.0};
            }
        },
    },
    RH3040_battery_change: {
        cid: 'genPowerCfg',
        type: 'devChange',
        convert: (model, msg, publish, options) => {
            if (msg.data.data.hasOwnProperty('batteryPercentageRemaining')) {
                return {battery: msg.data.data['batteryPercentageRemaining'] *100 / 200.0};
            }
        },
    },
    RH3040_voltage: {
        cid: 'genPowerCfg',
        type: ['attReport', 'readRsp'],
        convert: (model, msg, publish, options) => {
            if (msg.data.data.hasOwnProperty('batteryVoltage')) {
                return {voltage: msg.data.data['batteryVoltage'] * 100.0};
            }
        },
    },
    RH3040_voltage_change: {
        cid: 'genPowerCfg',
        type: 'devChange',
        convert: (model, msg, publish, options) => {
            if (msg.data.data.hasOwnProperty('batteryVoltage')) {
                return {voltage: msg.data.data['batteryVoltage'] * 100.0};
            }
        },
    },
    RH3040_motion_dev_change: {
        cid: 'ssIasZone',
        type: 'devChange',
        convert: (model, msg, publish, options) => {
            if (msg.data.data.zoneType === 0x000D) { // type 0x000D = motion sensor
                const zoneStatus = msg.data.data.zoneStatus;
                return {
                    occupancy: (zoneStatus & 1) > 0, // Bit 0 = Alarm 1: Presence Indication
					tamper: (zoneStatus & 1<<2) > 0, // Bit 2 = Tamper status
                    battery_low: (zoneStatus & 1<<3) > 0, // Bit 3 = Battery LOW indicator (trips around 2.4V)
                };
            }
        },
    },
    RH3040_motion_status_change: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            const zoneStatus = msg.data.zoneStatus;
            return {
                occupancy: (zoneStatus & 1) > 0, // Bit 0 = Alarm 1: Presence Indication
                tamper: (zoneStatus & 1<<2) > 0, // Bit 2 = Tamper status
                battery_low: (zoneStatus & 1<<3) > 0, // Bit 3 = Battery LOW indicator (trips around 2.4V)
            };
        },
    },

I still get few warnings in the log

line 23:
`7/6/2019, 10:04:05 AM - warn: Device with modelID 'undefined' is not supported.`

line 34, 44, 55:
`7/6/2019, 10:04:07 AM - warn: Failed to configure 0x000d6ffffe816e45 (0x000d6ffffe816e45) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)`

line 76:
`7/6/2019, 10:04:19 AM - warn: No converter available for 'RH3040 - TUYATEC Motion sensor' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"appVersion":67}}'`

I see few messages from line number 71 and 72:
`7/6/2019, 10:04:19 AM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":30,"batteryPercentageRemaining":200}}' of device 'undefined' (0x00124b000e896b65) of endpoint 11
7/6/2019, 10:04:19 AM - debug: Ignoring message from coordinator`

I want help in finding out the values for this line of code in devices.js

`(cb) => device.report('genPowerCfg', 'batteryVoltage','batteryPercentageRemaining', 1, 1000, 1, cb),`

for what function is it used for and where to find the values for it.
I have attached the log file here

Best regards,
Chandan

Koenkk, 2019-07-06:
line 23: can be ignored, at this point the modelId is not retrieved by zigbee2mqtt yet, on line 29 it is and you will never receive this error again.

line 34, 44, 55: happens due to the `configure` section of the device. If the configure fails and the device still works, you should check if this section is needed at all. I would remove it action by action and check if everything keeps working

line 76: add to fz: `fz.ignore_basic_change`

line  71 and 72: ignore it if battery reporting works

I recommend using:
```js
(cb) => device.report('genPowerCfg', 'batteryVoltage','batteryPercentageRemaining', repInterval.MINUTE, repInterval.MAX, 1, cb),
```

first value = minimal reporting interval
second value = max reporting interval
1 = minimal change, requires the reported value to be changed from the last reported one (or if the max reporting interval expires)

codegraffiti, 2019-07-11:
@Koenkk 

I had some time to play with this device and I think the device is fully working. I have created the pull request for this device.

closed 2019-07-12 by codegraffiti