Koenkk/zigbee2mqtt #1691: Network map rendering is very slow

antst, 2019-07-04:
I am running dev version of Z2M.
After recent change of network map, when it got coloured bars, it became impossible to use it in HA. Neither in browser nor in IOS app.
Simple zooming or panning take AGES. 
Before I'd scroll mouse and zoom it realtime so much as I want. Now I scroll, it become unnoticeable larger after few seconds. Then I scroll again and it become unnoticeable larger again after few seconds. And so on.

clockbrain, 2019-07-05:
Most network map discussion is happening in #652 but help us understand this issue a bit better before we work out if it belongs there.

How many devices / routers do you have in your network?

What map renderer are you using: https://github.com/rgruebel/ha_zigbee2mqtt_networkmap or https://github.com/ben423423n32j14e/zigbee2mqttadminpanel or something else?

Is it really the colors that are slowing down the map or is it the additional route information on links? Can you do any testing to help isolate this. If it is the route detail then perhaps that could be made a configuration or message request option.

Do you see the same behaviour when you manually paste the graphviz map data into an online graphviz viewer such as http://www.webgraphviz.com/ or https://dreampuf.github.io/GraphvizOnline ?

antst, 2019-07-05:
It is 40+ devices, I use ha_zigbee2mqtt_networkmap.

From what I see, before update which broke rendering, all information was already there.

Online renderers render as slow as HA. Only difference is that in those you render image once and there is no manipulations. while in HA you can use mouse to shift/zoom (which leads to re-render), and rendering takes 5 sec at least.

antst, 2019-07-05:
OK, I checked. Problem comes from excess of dotted and dashed.

sed 's/dashed,//g;s/dashed//g;s/dotted,//g;s/dotted//g' makes it blazing fast again.

antst, 2019-07-09:
Thanks! Now it is as it should be again :)

andreasbrett, 2019-07-09:
Great! Could you close the issue then?

closed 2019-07-09 by Koenkk