Koenkk/zigbee2mqtt #1692: npm install fails on raspibian buster

68CodingMonkeys, 2019-07-05:
# Bug Report
## What happened
Doing step: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
(installing on pi)

## What did you expect to happen
If everything went correctly the output of npm install is similar to (the number of packages and seconds is probably different on your device):

node-pre-gyp info ok
added 383 packages in 111.613s

## How to reproduce it (minimal and precise)

Follow steps in documenation for mac os x

## Debug Info
zigbee2mqtt version: git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
CC253X firmware version: CC2531_DEFAULT_20190608.zip
Buster Version 10
root@darkmatter:/etc# npm -v
6.9.0
root@darkmatter:/etc# node -v
v10.16.0

Note: the npm debug log was too large for pastebin

root@darkmatter:/opt/zigbee2mqtt# npm install
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/zigbee-shepherd-converters-865657b2/converters/common.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/yargs-parser-11cb6ddb/LICENSE.txt'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/zigbee-shepherd-converters-865657b2/converters/fromZigbee.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/yargs-parser-11cb6ddb/README.md'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/zcl-id-5993891d/definitions/common.json'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/zcl-id-11c412c6/definitions/common.json'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/deep-diff-e6df6311/.vscode/launch.json'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/deep-diff-e6df6311/.vscode/tasks.json'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/source-map-52448a65/dist/source-map.js'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/mocha-c0ce7c63/index.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/acorn-cd6124d3/dist/acorn.js.map'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/mocha-c0ce7c63/LICENSE'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/yargs-96b4c070/completion.sh.hbs'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/yargs-96b4c070/index.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/source-map-23af578d/dist/source-map.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/mocha-c0ce7c63/mocha.css'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/yargs-96b4c070/LICENSE'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/yargs-96b4c070/README.md'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/yargs-96b4c070/yargs.js'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/resolve-093ee6f6/test/pathfilter/deep_ref/node_modules/deep/deeper/ref.js'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/winston-3913d932/lib/winston'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/moment-6f9b3406/package.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/moment-6f9b3406/README.md'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/chai-a886f99e/CODE_OF_CONDUCT.md'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/chai-a886f99e/CODEOWNERS'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/chai-a886f99e/CONTRIBUTING.md'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/chance-162cb67c/gulpfile.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/chance-162cb67c/LICENSE'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/chance-162cb67c/README.md'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/chance-162cb67c/yarn.lock'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/eslint-ee3ff7e2/LICENSE'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/mqtt-514b80e9/dist/mqtt.min.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/eslint-ee3ff7e2/README.md'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/resolve-093ee6f6/test/resolver/without_basedir/main.js'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/resolve-093ee6f6/test/resolver/without_basedir/node_modules'
npm ERR! err.code.match is not a function

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-05T06_40_34_657Z-debug.log

Koenkk, 2019-07-05:
Probably the same as https://github.com/Koenkk/zigbee2mqtt/issues/1102#issuecomment-464445828

68CodingMonkeys, 2019-07-10:
since you dont bother with this error, please close it.

closed 2019-07-10 by Koenkk