Koenkk/zigbee2mqtt #1693: Aqara bulbs issue: one specific bulb gets disconnected all the time

alexvenom, 2019-07-05:
# Bug Report
## What happened

Have multiple devices connected to Zigbee2Mqtt using CC2531 using 20190608.hex and most of them work well.
There is one Aqara Bulb though that keeps getting disconnected all the time. 
When it works, it works perfectly. But every now and then, twice a minute at least, the Bulb gets disconnected, the Lovelace card for the light gets grayed out and no commands are accepted. 
When it gets back, all commands sent before it went off are called in a row very quickly.
Other Aqara bulbs work well.

Log says Failed to ping the device

## What did you expect to happen

## How to reproduce it (minimal and precise)

Not sure how to reproduce as it works fine with other bulbs.

## Debug Info
zigbee2mqtt version: Latest (development) version but  behaviour happens with different versions, stable or development branches.
CC253X firmware version: 20190608 but tried different ones with same result

Koenkk, 2019-07-07:
Although not an end device, maybe related to https://github.com/Koenkk/zigbee2mqtt/issues/1684#issuecomment-508304011

alexvenom, 2019-07-10:
Though definitely not solved, it seems much more stable with that firmware.
I am getting a lot less errors in the log, but a few are still showing:

`zigbee2mqtt:error 7/9/2019, 8:15:12 PM Failed to ping 0x00158d00032caef7
zigbee2mqtt:info 7/9/2019, 8:06:40 PM Zigbee publish to device '0x00158d00032caef7', genOnOff - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:info 7/9/2019, 8:06:40 PM Zigbee publish to device '0x00158d00032caef7', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:error 7/9/2019, 8:07:03 PM Zigbee publish to device '0x00158d00032caef7', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.`

When those show, the light won't work, but it's taking a lot more time for those to show than it was taking before.

Koenkk, 2019-07-10:
Do you have the same behaviour with: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing?

alexvenom, 2019-07-10:
Unfortunately yes:

`zigbee2mqtt:error 7/10/2019, 7:46:44 PM Failed to ping 0x00158d00032caef7
  zigbee2mqtt:error 7/10/2019, 7:47:14 PM Zigbee publish to device '0x00158d00032caef7', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.`

And card still gets grayed out.
I used the CC2531_SOURCE_ROUTING_20190619 hex.

Koenkk, 2019-07-11:
Error 205 indicates there is no route, which is less likely to happen with the source routing firmware because it caches the routes. The real error seems to be the 233 no mac ack. This happens when the device cannot be reached, most of the times due to interference/out of range, can you put the bulb closer to the CC2531 and check if the problem persists?

alexvenom, 2019-07-11:
The bulbs are on a test table sitting literally side by side to the Raspberry Pi and the CC2531, so distance cannot be an issue. 
I was thinking... is it possible I have a bad CC2531? Or maybe a bad Aqara bulb? Or perhaps both?
Curiously, Aqara and Mijia sensors such as temperature, presence and door sensors all report correctly 100% of the times, and so does the Aqara and Mijia switches.

Maybe I have something wrong with my configuration?

`{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": false,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://homeassistant",
  "serial_port": "/dev/ttyACM0",
  "devices": [],
  "network_key": [],
  "mqtt_user": "myUsername",
  "mqtt_pass": "myPassword",
  "disable_led": true,
  "log_level": "info",
  "availability_timeout": 30,
  "report": true,
  "include_device_information": true
}`

Koenkk, 2019-07-12:
Can you try without availability timeout?

alexvenom, 2019-07-12:
Left during the night without the availability_timeout option and the behaviour continues to happen, but I rolled back the version to the one specified in #1684 (your first comment on this thread).

Tonight I will try to use https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing again and without the availability_timeout to see how it behaves.
Just to make sure, I shoud use CC2531_SOURCE_ROUTING_20190619.zip right?

I am also thinking about modding my CC2531 as this:
https://hassiohelp.eu/2019/03/02/zigbee-cc2531-antenna/

So I can have a little more range, even the lamps being around 5 inches from the CC2531 right now.

Something I also noticed, my Raspberry keeps complaining about low voltage. I am not really sure that could be the problem, but I will also replace the energy source for a more stable one to eliminate that variable as well.
Will report soon.

closed 2019-09-17 by stale[bot]