Koenkk/zigbee2mqtt #1696: Implement OTA

tb-killa, 2019-07-06:
As part of further preparations to implement a basic OTA server, a first OTA implementation based on the zcl-id cluster definition is required.
This is followed by a general implementation of the discovery process:

```
Before becoming part of the network, a device may be preprogrammed with the IEEE address of the
authorized upgrade server. In this case, once the device is part of the network, it shall discover the
network address of the upgrade server via ZDO network address discovery command.

If the device is not preprogrammed with the upgrade server’s IEEE address, the device shall discover
the upgrade server before it participates in any upgrade process. The device shall send Match
Descriptor Request (ZDO command) to discover an upgrade server by specifying a single OTA cluster
ID in the input Cluster attribute. If the receiving node is an upgrade server, it shall reply with Match
Descriptor Response, with the (active) endpoint that the OTA cluster is implemented on, hence,
identifying itself as acting as server in OTA Upgrade cluster. Since Match descriptor request may be
sent as unicast or broadcast, the client may get multiple responses if there are more than one server in
the network. The client shall use the first response received6 794 .. Each application profile should specify
the frequency of OTA server discovery done by the client. After discovering the OTA server’s short
ID via the ZDO Match descriptor request the client shall discover the IEEE address of the upgrade
server via ZDO IEEE address discovery command and store the value in UpgradeServerID attribute.
```

I thought we could just use @Koenkk's [responder ](https://github.com/Koenkk/zigbee2mqtt/commit/0d587f304ed6597596c047b0e9db4e6ccfa061a6)to intercept first requests, but apparently none are sent at all.

Possibly broadcasts are not forwarded ?
Maybe @Koenkk could say something about this ?

Koenkk, 2019-07-07:
Could you grab such a message with sniffing?

tb-killa, 2019-07-07:
Will check and report. 
I modify the ``zcl-id`` and ``zcl-packet`` to include ``genOta`` fields at first (use the Values from @HalleyAssist / @splitice).

From what i know that some of my devices now shown the own ``genOta`` Fields with Values but none of them seems to request for any of the defined ways for answer of OTA-Endpoint-Server.

Maybe our Default EndPoint from zigbee-shepherd dont allow them or maybe we need special init the needing genOta Cluster Part ?

tb-killa, 2019-07-07:
@Koenkk How we could add OTA cluster to inCluster on Simple Descriptor of our coordinator ?
This is needing to allow the Clients doing the requests.

tb-killa, 2019-07-07:
@Koenkk got some more logging messages:

```
 zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":2,"securityuse":0,"timestamp":3858323,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,3,1,0,113,16,235,0,225,0,0,0]}} +3m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +56s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3867505,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,4,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3876678,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,5,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3885840,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,6,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3895011,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,7,1,0,113,16,235,0,225,0,0,0]}} +3s
```

Koenkk, 2019-07-08:
I prefer to move them into the default ep as I want to get rid of ZAPP

tb-killa, 2019-07-08:
How we could do this?
Maybe we could combine this work and the ias part for removing the zapp?

Koenkk, 2019-07-08:
Eventually I want to integrate this into zigbee-herdsman (same goes for e.g. the ias zone enrolment, so we can completely remove it from the `configure` sections).

tb-killa, 2019-07-08:
@Koenkk Doesn´t we could add simple our Clusters to the ``inClusterList: []`` and ``outClusterList: []`` on 
https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/initializers/init_controller.js#L117
and remove the zapp include ? for ``genOta`` we need ``inClusterList: [25]`` and ``outClusterList: [25,1280]`` where last ones is for the ``ias``.

Koenkk, 2019-07-08:
I havent' checked that, but if it's that easy, why not?

tb-killa, 2019-07-09:
Doesn´t seems to work :(

```
/opt/zigbee2mqtt/node_modules/q/q.js:155
                throw e;
                ^

TypeError: Cannot read property 'device' of undefined
    at DeviceReceive.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:21:56)
    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:114:75)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeStarted (/opt/zigbee2mqtt/lib/controller.js:114:58)
    at zigbee.start (/opt/zigbee2mqtt/lib/controller.js:169:26)
    at shepherd.start (/opt/zigbee2mqtt/lib/zigbee.js:80:17)
    at /opt/zigbee2mqtt/node_modules/q/q.js:2055:17
    at runSingle (/opt/zigbee2mqtt/node_modules/q/q.js:137:13)
    at flush (/opt/zigbee2mqtt/node_modules/q/q.js:125:13)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
```

tb-killa, 2019-07-09:
Yes:
Seems to work now :)

Modify some Parts:
https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/initializers/init_controller.js#L81
``
dlgInfos =  [
            { profId: 0x0104, epId: 1, inClusterList: [25], outClusterList: [25,1280] }, 
			{ profId: 0x0101, epId: 2, inClusterList: [], outClusterList: [] }, 
			{ profId: 0x0105, epId: 3, inClusterList: [], outClusterList: [] },
            { profId: 0x0107, epId: 4, inClusterList: [], outClusterList: [] }, 
			{ profId: 0x0108, epId: 5, inClusterList: [], outClusterList: [] }, 
			{ profId: 0x0109, epId: 6, inClusterList: [], outClusterList: [] }
        ];
``
https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/initializers/init_controller.js#L117
``
var dlgDesc = { profId: dlgInfo.profId, epId: dlgInfo.epId, devId: 0x0005, inClusterList: dlgInfo.inClusterList, outClusterList: dlgInfo.outClusterList },
``

Comment out the complete Part of https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L151

After Restarting the Z2M the new Coordinator Entrys in the db File looks like this:
``
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b000XXXXXXX","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1562658293,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[25],"outClusterList":[25,1280],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"VCQlPYRg6jjw9DX3"}
``

Endpoint#1 now include the needing Clusters in the List.

Now we doesn´t need the EndPoint:11 from our own ZAPP no more.
We need some more Tests but this looks like we could now add different Clusters to our local Endpoints.

tb-killa, 2019-07-09:
Now my DEBUG Output look like this:
```
zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":860088,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[24,9,13,1]}} +14ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +17ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":860088,"transseqnumber":0,"len":4,"data":{"0":24,"1":9,"2":13,"3":1},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":9,"cmdId":"discoverRsp","payload":{"discComplete":1,"attrInfos":[]}}} +5ms
```

tb-killa, 2019-07-09:
I test this with the Bitron Device ``AV2010/34``

```
zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 48 +233ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":40689,"srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":5916132,"transseqnumber":0,"len":40,"data":{"type":"Buffer","data":[24,46,13,1,0,0,240,1,0,35,2,0,35,3,0,33,4,0,35,5,0,33,6,0,32,7,0,33,8,0,33,9,0,33,10,0,35,70,0,35]}} +19ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +68ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":40689,"srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":5916132,"transseqnumber":0,"len":40,"data":{"0":24,"1":46,"2":13,"3":1,"4":0,"5":0,"6":240,"7":1,"8":0,"9":35,"10":2,"11":0,"12":35,"13":3,"14":0,"15":33,"16":4,"17":0,"18":35,"19":5,"20":0,"21":33,"22":6,"23":0,"24":32,"25":7,"26":0,"27":33,"28":8,"29":0,"30":33,"31":9,"32":0,"33":33,"34":10,"35":0,"36":35,"37":70,"38":0,"39":35},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":46,"cmdId":"discoverRsp","payload":{"discComplete":1,"attrInfos":[{"attrId":0,"dataType":240},{"attrId":1,"dataType":35},{"attrId":2,"dataType":35},{"attrId":3,"dataType":33},{"attrId":4,"dataType":35},{"attrId":5,"dataType":33},{"attrId":6,"dataType":32},{"attrId":7,"dataType":33},{"attrId":8,"dataType":33},{"attrId":9,"dataType":33},{"attrId":10,"dataType":35},{"attrId":70,"dataType":35}]}}} +51ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 49 +276ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +17ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: {"status":0,"endpoint":1,"transid":49} +73ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 49 +75ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":40689,"srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":5916580,"transseqnumber":0,"len":45,"data":{"type":"Buffer","data":[24,47,1,0,0,0,240,255,255,255,255,255,255,255,255,1,0,0,35,255,255,255,255,2,0,0,35,6,0,0,0,3,0,0,33,2,0,4,0,0,35,255,255,255,255]}} +19ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +72ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":40689,"srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":5916580,"transseqnumber":0,"len":45,"data":{"0":24,"1":47,"2":1,"3":0,"4":0,"5":0,"6":240,"7":255,"8":255,"9":255,"10":255,"11":255,"12":255,"13":255,"14":255,"15":1,"16":0,"17":0,"18":35,"19":255,"20":255,"21":255,"22":255,"23":2,"24":0,"25":0,"26":35,"27":6,"28":0,"29":0,"30":0,"31":3,"32":0,"33":0,"34":33,"35":2,"36":0,"37":4,"38":0,"39":0,"40":35,"41":255,"42":255,"43":255,"44":255},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":47,"cmdId":"readRsp","payload":[{"attrId":0,"status":0,"dataType":240,"attrData":"0xffffffffffffffff"},{"attrId":1,"status":0,"dataType":35,"attrData":4294967295},{"attrId":2,"status":0,"dataType":35,"attrData":6},{"attrId":3,"status":0,"dataType":33,"attrData":2},{"attrId":4,"status":0,"dataType":35,"attrData":4294967295}]}} +55ms
  zigbee2mqtt:debug 2019-7-9 10:15:40 Received zigbee message of type 'readRsp' with data '{"cid":"genOta","data":{"upgradeServerId":"0xffffffffffffffff","fileOffset":4294967295,"currentFileVersion":6,"currentZigbeeStackVersion":2,"downloadedFileVersion":4294967295}}' of device 'AV2010/34' (0x00124b000959a3ea) of endpoint 4
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 50 +143ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +17ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: {"status":0,"endpoint":1,"transid":50} +76ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 50 +79ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":40689,"srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":5917039,"transseqnumber":0,"len":32,"data":{"type":"Buffer","data":[24,48,1,5,0,0,33,255,255,6,0,0,32,0,7,0,0,33,0,0,8,0,0,33,0,0,9,0,0,33,0,0]}} +15ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +44ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":40689,"srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":5917039,"transseqnumber":0,"len":32,"data":{"0":24,"1":48,"2":1,"3":5,"4":0,"5":0,"6":33,"7":255,"8":255,"9":6,"10":0,"11":0,"12":32,"13":0,"14":7,"15":0,"16":0,"17":33,"18":0,"19":0,"20":8,"21":0,"22":0,"23":33,"24":0,"25":0,"26":9,"27":0,"28":0,"29":33,"30":0,"31":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":48,"cmdId":"readRsp","payload":[{"attrId":5,"status":0,"dataType":33,"attrData":65535},{"attrId":6,"status":0,"dataType":32,"attrData":0},{"attrId":7,"status":0,"dataType":33,"attrData":0},{"attrId":8,"status":0,"dataType":33,"attrData":0},{"attrId":9,"status":0,"dataType":33,"attrData":0}]}} +32ms
[... cut ...] 5,"16":255,"17":255,"18":255},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":49,"cmdId":"readRsp","payload":[{"attrId":10,"status":0,"dataType":35,"attrData":0},{"attrId":70,"status":0,"dataType":35,"attrData":4294967295}]}} +13ms
  zigbee2mqtt:debug 2019-7-9 10:15:40 Received zigbee message of type 'readRsp' with data '{"cid":"genOta","data":{"70":4294967295,"imageStamp":0}}' of device 'AV2010/34' (0x00124b000959a3ea) of endpoint 4
```

tb-killa, 2019-07-09:
@Koenkk The only difference between our own build clusterlist from https://github.com/Koenkk/zigbee2mqtt/issues/1696#issuecomment-509532268 and using a ZAPP with the Mount Option via https://github.com/zigbeer/zigbee-shepherd/blob/master/lib/shepherd.js#L243 would be the Part of this: https://github.com/zigbeer/zigbee-shepherd/blob/master/lib/shepherd.js#L288-L301
Maybe we need this too to get the EP:1 Stuff working on right way ?

tb-killa, 2019-07-09:
The Messages are from ``incomingMsg`` who are going to ``targetEp.onAfIncomingMsg``.
Maybe we need here some bind to our resolver ?

Koenkk, 2019-07-10:
Probably, but before we change old shepherd code, I'm currently refactoring the complete zigbee-shepherd library and will take this into account to make such things much more easy.

tb-killa, 2019-07-12:
@splitice What type of Modification on zigbee-shepherd themselve it need to get the broadcast passed or to interact with the request and responses ? I build up the needing steps for the OTA Statemachine but doesn´t could pass the needing traffic.
Hope you could public some Informations as you mentioned that your OTA Server is closed-sourced (actually).

tb-killa, 2019-07-12:
Added more ``debug`` logic to ``zigbee-shepherd`` stuff:

```
zigbee-shepherd:msgHdlr _areqEventBridge(): msg: {"subsys":"AF","ind":"incomingMsg","data":{"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":15,"securityuse":0,"timestamp":10983335,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,3,1,0,113,16,235,0,225,0,0,0]}}} +59s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":15,"securityuse":0,"timestamp":10983335,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,3,1,0,113,16,235,0,225,0,0,0]}} +59s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +59s
  zigbee-shepherd:msgHdlr _areqEventBridge(): msg: {"subsys":"AF","ind":"incomingMsg","data":{"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":10992507,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,4,1,0,113,16,235,0,225,0,0,0]}}} +3s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":10992507,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,4,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
  zigbee-shepherd:msgHdlr _areqEventBridge(): msg: {"subsys":"AF","ind":"incomingMsg","data":{"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":13,"securityuse":0,"timestamp":11001690,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,5,1,0,113,16,235,0,225,0,0,0]}}} +3s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":13,"securityuse":0,"timestamp":11001690,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,5,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
  zigbee-shepherd:msgHdlr _areqEventBridge(): msg: {"subsys":"AF","ind":"incomingMsg","data":{"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":11010866,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,6,1,0,113,16,235,0,225,0,0,0]}}} +3s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":18,"securityuse":0,"timestamp":11010866,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,6,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
  zigbee-shepherd:msgHdlr _areqEventBridge(): msg: {"subsys":"AF","ind":"incomingMsg","data":{"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":21,"securityuse":0,"timestamp":11020026,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,7,1,0,113,16,235,0,225,0,0,0]}}} +3s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":25,"srcaddr":49018,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":21,"securityuse":0,"timestamp":11020026,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,7,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3s
```

dzungpv, 2019-07-19:
@tb-killa For the OTA you must setup a OTA server, and then activate OTA update process on the devices, not just change in zigbee-shepherd , I think this document helpful for you because our solution base on TI API: http://dev.ti.com/tirex/content/simplelink_zigbee_sdk_plugin_1_60_01_09/docs/zigbee_user_guide/html/zigbee/developing_zigbee_applications/ota_upgrade/ota_upgrade.html.

Before build OTA server yourself, try TI close sourced one first.

lkollar, 2019-09-17:
`zigpy` has an implementation for OTA updates: https://github.com/zigpy/zigpy/pull/213. This could be helpful in understanding the upgrade process.

tb-killa, 2019-10-17:
Close this Issue Post and move to https://github.com/Koenkk/zigbee-herdsman/issues/60

closed 2019-10-17 by tb-killa