Koenkk/zigbee2mqtt #1697: Philips hue dimmer remote control state has not changed

hawaltie, 2019-07-07:
## What happened
I paired the philips hue dimmer remote control, using zigbee2mqtt and home assistant. The device was paired correct. When I push the 'on' button (|), the mqtt log ia as follows:
`  zigbee2mqtt:info 7/7/2019, 7:38:29 AM MQTT publish: topic 'zigbee2mqtt/0x0017880104e7027a', payload '{"battery":100,"linkquality":97,"counter":1,"brightness":155,"action":"on-press","duration":0}'

zigbee2mqtt:info 7/7/2019, 7:38:29 AM MQTT publish: topic 'zigbee2mqtt/0x0017880104e7027a', payload '{"battery":100,"linkquality":97,"counter":1,"brightness":155,"action":""}'`

So the action attribute is changed to the value "on-press" and immediately changed back again to "". Because this happens in a split second, I am not able to use this trigger for automation scripts. The state of the sensor created in Home assistant for this Hue dimmer remote control is not updated.

## What did you expect to happen
In the past I used the state of sensor in home assistant for the Hue dimmer remote control, and that worked. For some reason the state is not updated.

## How to reproduce it (minimal and precise)
1. Pair the hue dimmer remote control to zigbee2mqtt and home assistant.
2. Push the on button
3. Open 2 windows: 1 showing the zigbee2mqtt log, one showing the state window of home assistant.
3. Check the log of Zigbee2mqtt to see if zigbee2mqtt has recieved the trigger.
4. Check if the trigger is recieved by home assistant: you should see the action attribute change rapidly.
5. Check if the state of the sensor of the hue dimmer remote control in home assistant has changed

## Debug Info
zigbee2mqtt version: 1.4
CC2531 firmware version: CC2531_DEFAULT_20190608.zip

Koenkk, 2019-07-07:
This is actually expected (otherwise automations wouldnt trigger, you can find an example here http://www.zigbee2mqtt.io/integration/home_assistant.html (responding to button click via home assistant entity)

hawaltie, 2019-07-07:
Thanx.

I try to write a test script, which sends a message to my mail when a button is pushed, but is doesn' t work.

`- id: '1543874885010'
  alias: Philips Hue Remote Control 1st Floor
  trigger:
    platform: mqtt
    topic: zigbee2mqtt/0x0017880104e7027a
  condition:
      condition: template
      value_template: "{{ trigger.payload_json.action in ['off-press', 'down-press'] }}"
  action:
    - service: notify.notify_mail
      data_template:
        title: "Trigger works"
        message: >
          {% if trigger.payload_json.action  = 'down-press' %}
          "Pushed trigger is Down"
          {% if trigger.payload_json.action  = 'off-press' %}
          "Pushed trigger is Off"
          {% else %}
          "Pushed trigger is Wrong"
          {% endif %}     `

The error message complains about the = sign. I don't have any clue.

hawaltie, 2019-07-07:
Thanx.

I try to write a test script, which sends a message to my mail when a button is pushed, but it doesn't work

`
id: '1543874885010'
  alias: Philips Hue Remote Control 1st Floor
  trigger:
    platform: mqtt
    topic: zigbee2mqtt/0x0017880104e7027a
  condition:
      condition: template
      value_template: "{{ trigger.payload_json.action in ['off-press', 'down-press'] }}"
  action:
    - service: notify.notify_mail
      data_template:
        title: "Trigger works"
        message: >
          {% if trigger.payload_json.action | string = 'down-press' %}
          "Pushed trigger is Down"
          {% if trigger.payload_json.action | string = 'off-press' %}
          "Pushed trigger is Off"
          {% else %}
          "Pushed trigger is Wrong"
          {% endif %}         

`

hawaltie, 2019-07-07:
Thanx. I tried to write an action to send an message when a button is pushed. But for some reason the action doesn't work.

This is my code:

`id: '1543874885010'
  alias: Philips Hue Remote Control 1st Floor
  trigger:
    platform: mqtt
    topic: zigbee2mqtt/0x0017880104e7027a
  condition:
      condition: template
      value_template: "{{ trigger.payload_json.action in ['off-press', 'down-press'] }}"
  action:
    - service: notify.notify_mail
      data_template:
        title: "Trigger works"
        message: >
          {% if trigger.payload_json.action = 'down-press' %}
          "Pushed trigger is Down"
          {% if trigger.payload_json.action = 'off-press' %}
          "Pushed trigger is Off"
          {% else %}
          "Pushed trigger is Wrong"
          {% endif %}`

Koenkk, 2019-07-07:
Use this integration http://www.zigbee2mqtt.io/integration/home_assistant.html#via-home-assistant-entity (don't use platform mqtt)

hawaltie, 2019-07-07:
I managed to do it via platform mqtt. I' ll try the other way as well.

closed 2019-07-07 by hawaltie