Koenkk/zigbee2mqtt #1698: Aqara ZNLDP12LM succesfully reenabled joinin

hagazan, 2019-07-07:
Hi All,

Please help me with my issue regarding light bulb Aqara ZNLDP12LM.

On Zigbee to MQTT Bridge i receive the following log lines:

zigbee2mqtt:info 7/7/2019, 11:25:41 AM Succesfully reenabled joining zigbee2mqtt:info 7/7/2019, 11:29:01 AM Succesfully reenabled joining zigbee2mqtt:info 7/7/2019, 11:29:29 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d3a4d9', payload '{"state":"ON","brightness":255,"color_temp":153,"linkquality":44}' zigbee2mqtt:info 7/7/2019, 11:32:21 AM Succesfully reenabledjoining``

But if i have a look into the integrations page or dev page, i cannot find the founded light bulb.

mqtt: broker: 192.168.0.5 discovery: true birth_message: topic: 'hass/status' payload: 'online' will_message: topic: 'hass/status' payload: 'offline'

Could you please help me?

Thanks in advance!

Hagazan

maorbenamer, 2019-07-12:
Same issue, Someone have any idea?

Koenkk, 2019-07-12:
Please post full logs (on pastebin.com) and link it here, also clearly describe what the issue is and what you expect, without this nobody can help.

maorbenamer, 2019-07-12:
Hi Koenkk,

thats the log from the zigbee2mqtt add-on: [https://pastebin.com/BWGpvNnM](url)
I managed to connect a new device but could not find it on the entities list or in the Integration.

Koenkk, 2019-07-12:
Did you search for `0x00158d00032ca6bd`? You see that it is discoverd on line 22, also, did you enable home assistant discovery? (http://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery)

maorbenamer, 2019-07-12:
I didnt find anything with 0x00158d00032ca6bd.
Also the discovery set to true.
This is the Zigbee2MQTT configuration.yaml : [https://pastebin.com/ymFqDvDk](url)
This is the HomeAssistant configuration.yaml : [https://pastebin.com/UM1KQe4X](url)
Thanks

Koenkk, 2019-07-15:
Looks good, can you 

- Start home assistant
- Wait a few minutes
- Start zigbee2mqtt
- Wait a minute
- Provide logs of home assistant

maorbenamer, 2019-07-15:
Hi,
Home Assistant Log:[https://pastebin.com/8YsBBeMK](https://pastebin.com/8YsBBeMK)
Zigbee2MQTT Log: [https://pastebin.com/0DEGYSQq](https://pastebin.com/0DEGYSQq)

Should I add manually the sensor to the configuration file?

Thanks

Koenkk, 2019-07-15:
Please provide more of the home assistant log (wait a few minutes as mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/1698#issuecomment-511288955)

maorbenamer, 2019-07-15:
The home-assistant.log is empty

Koenkk, 2019-07-15:
Ok, so im not sure how to get the home assistant log on hassio, but from the current logs I dont see anything strange

maorbenamer, 2019-07-15:
OK, 
The last option I have left is to reinstall everything from scratch.
Thanks for your help

maorbenamer, 2019-07-17:
Just want to update that everything work great after re-flashed the sdcard and the cc2531 usb. 
Thank you very much!

closed 2019-07-17 by Koenkk