Koenkk/zigbee2mqtt #1699: CC2530 Jamming Wifi

selmholt, 2019-07-07:
# Bug Report
## What happened
Zigbee2MQTT version: 1.4.0
CC2530 verison: CC2530_DEFAULT_20190608

When connectiong the CC2530 to my raspberry PI and starting Zigbee2MQTT the WIFI signal is jammed in some areas of my house, if I make a test the internet is totally dead. If i disconnect the CC2530, the internet works again. 

How can this be? 

## What did you expect to happen
I did not expect the wifi to get jammed. 

## How to reproduce it (minimal and precise)
I can send you my CC2530

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: CC2530_DEFAULT_20190608

SAGV, 2019-07-07:
Basically a direct quote from [Zigbe2mqtt.io](https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html)

> As Wifi and Zigbee both operate on the same frequency space (2.4 GHz); they can interfere with each other. By using the correct Zigbee channel interference with Wifi can (partly) be avoided. There is a good article available explaining this: https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence To change the Zigbee channel Zigbee2mqtt uses you have to set the channel in configuration.yaml.

boojew, 2019-07-09:
@SAGV any chance you would know what the default channel is?

SAGV, 2019-07-09:
@boojew maybe https://github.com/Koenkk/zigbee2mqtt/issues/650 or https://github.com/Koenkk/zigbee2mqtt/issues/178 would help?

SAGV, 2019-07-09:
Btw, what really helped me to jam the signal less and improve zigbee link quality is flashing firmware from here https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator , buying 1.5 metres usb extender, placing zigbee adapter on the top of my wardrobe and turning it by 90 degrees. 

You can check link quality in zigbee2mqtt logs for every device and see whether your adjustments made it better or worse.

selmholt, 2019-07-10:
Made som channel changes on zigbee2MQTT and my 2,4 GHz wifi - Now i looks like it is working.

closed 2019-07-10 by selmholt