Koenkk/zigbee2mqtt #1700: Complete SSL support

Diaoul, 2019-07-07:
I'd like to use SSL to connect to my Mosquitto server, currently, zigbee2mqtt isn't ready for it.
Several MQTT.js options are not exposed, this ticket is about making them configurable.

[Here](https://github.com/mqttjs/MQTT.js/blob/master/examples/tls%20client/mqttclient.js#L27-L36) is an example of the MQTT.js configuration I would like to achieve, missing keys are `key`, `cert` and `ca`, all of those are file paths.

Can you expose them?

Thanks!

andreasbrett, 2019-07-08:
I just exposed those 3 parameters. Could you modify your local "lib/mqtt.js" file and check if this works for your SSL/TLS secured MQTT server? If it works @Koenkk could merge this pull.

https://github.com/Koenkk/zigbee2mqtt/pull/1706

andreasbrett, 2019-07-08:
Also added a PR for the zigbee2mqtt documentation: https://github.com/Koenkk/zigbee2mqtt.io/pull/85

Koenkk, 2019-07-08:
Waiting for @Diaoul to verify this, @andreasbrett thanks again!

Diaoul, 2019-07-08:
Thank you very much! I've quickly reviewed it but don't hold your breath on the testing part because I will only be able to test when the CC2531 arrives (ordered it yesterday) and I'm back from holidays. That would be early august at best.

andreasbrett, 2019-07-08:
> Thank you very much! I've quickly reviewed it but don't hold your breath on the testing part because I will only be able to test when the CC2531 arrives (ordered it yesterday) and I'm back from holidays. That would be early august at best.

That's too bad but was a great opportunity for me to finally make my MQTT server more secure. I can now confirm that zigbee2mqtt will connect to my MQTT server that is set up to with a SSL/TLS configuration. Both client and server authenticate against each other without issue.

Koenkk, 2019-07-09:
@andreasbrett so for you it works?

andreasbrett, 2019-07-09:
Exactly. Server authentication only worked as well as server+client authentifiation. I used mosquitto as mqtt server and openssl for certificate generation.

Koenkk, 2019-07-10:
merged it, thanks @andreasbrett !!

closed 2019-07-10 by Koenkk

Jeoffreybauvin, 2019-07-12:
Hi there,

I'm testing this modification on my setup. It seems to be working fine ;) : 

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt:8883'
  ca: /etc/mosquitto/ssl/ca/ca.crt
  key: /etc/mosquitto/ssl/client/zigbee2mqtt.key
  cert: /etc/mosquitto/ssl/client/zigbee2mqtt.crt
serial:
  port: /dev/ttyACM0
```

After restarting : 

```
  zigbee2mqtt:info 2019-7-12 11:58:19 PM Connected to MQTT server
  zigbee2mqtt:info 2019-7-12 11:58:19 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-7-12 11:58:19 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0003130670', payload '{"battery":100,"voltage":3025,"contact":true,"linkquality":86}'
```