Koenkk/zigbee2mqtt #1701: AM Cannot get the Node Descriptor of the Device: 0x00158d0002672c7a (Error: Timed out after 10000 ms)

tronicanet, 2019-07-08:
please help. I can not pair the devices

```
zigbee2mqtt:error 7/8/2019, 1:41:36 AM Cannot get the Node Descriptor of the Device: 0x00158d0002672c7a (Error: Timed out after 10000 ms)
```

DEBUG=* npm start
```
root@odroidn2:/opt/zigbee2mqtt# DEBUG=* npm start

> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 7/8/2019, 1:37:21 AM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-08.01-37-21'
  zigbee2mqtt:debug 7/8/2019, 1:37:21 AM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-07-08.00-31-01'
  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:debug 7/8/2019, 1:37:21 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 7/8/2019, 1:37:21 AM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 7/8/2019, 1:37:21 AM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 7/8/2019, 1:37:21 AM Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
  zigbee2mqtt:info 7/8/2019, 1:37:21 AM Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyACM0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +2ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +4ms
  serialport/stream _write 1 bytes of data +1ms
  serialport/binding-abstract write 1 bytes +6ms
  serialport/stream _read reading +0ms
  serialport/binding-abstract read +0ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned null 1 +2ms
  serialport/bindings/unixWrite wrote 1 bytes +0ms
  serialport/bindings/unixWrite Finished writing 1 bytes +0ms
  serialport/stream binding.write write finished +4ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +3ms
  serialport/bindings/poller Polling for "readable" +6ms
  cc-znp The serialport /dev/ttyACM0 is opened. +1s
  zigbee-shepherd:request REQ --> SYS:version +0ms
  cc-znp:SREQ --> SYS:version, {} +0ms
  serialport/stream _write 5 bytes of data +1s
  serialport/binding-abstract write 5 bytes +1s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s
  serialport/bindings/unixWrite write returned null 5 +1ms
  serialport/bindings/unixWrite wrote 5 bytes +0ms
  serialport/bindings/unixWrite Finished writing 5 bytes +1ms
  serialport/stream binding.write write finished +5ms
  serialport/bindings/poller received "readable" +1s
  serialport/binding-abstract read +292ms
  serialport/bindings/unixRead Starting read +1s
  serialport/bindings/unixRead Finished read 30 bytes +1ms
  serialport/stream binding.read finished +292ms
  cc-znp { sof: 254,
  cc-znp   len: 6,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'SYS',
  cc-znp   cmd: 'resetInd',
  cc-znp   payload:
  cc-znp    { reason: 0,
  cc-znp      transportrev: 2,
  cc-znp      productid: 0,
  cc-znp      majorrel: 2,
  cc-znp      minorrel: 6,
  cc-znp      hwrev: 3 },
  cc-znp   fcs: 194,
  cc-znp   csum: 194 } +345ms
  cc-znp { sof: 254,
[... 1511 lines cut ...]
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 1030,
  cc-znp      srcaddr: 33624,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 131,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 446504,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 7,
  cc-znp      data: <Buffer 18 95 0a 00 00 18 01> },
  cc-znp   fcs: 37,
  cc-znp   csum: 37 } +6s
  serialport/stream _read reading +20ms
  serialport/binding-abstract read +26ms
  serialport/bindings/unixRead Starting read +23ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1030, srcaddr: 33624, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 131, securityuse: 0, timestamp: 446504, transseqnumber: 0, len: 7, data: <Buffer 18 95 0a 00 00 18 01> } +6s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1030,"srcaddr":33624,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":131,"securityuse":0,"timestamp":446504,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,149,10,0,0,24,1]}} +6s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +6s
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +4ms
  serialport/bindings/poller Polling for "readable" +33ms

```

arjanvroege, 2019-08-04:
I see similar issues running the latest release (1.5.1). I tried the default firmware and the source routing firmware of the CC2531 stick but this makes no change. I see this issue only with IKEA bulbs (led). I tried to reset the bulb multiple times but this doesn't solve the issue.  I did a re-flash of both CC2531 coordinators (default and source routing). That didn't solved the issue either.. It looks like a cosmetic error since I'm able to control the Bulb. Does anyone know what this error means?

Koenkk, 2019-08-05:
When the device indicates that it is joining the network, zigbee-herdsman (used zigbee library) will start to query device information (get node descriptors) about e.g. which clusters it supports. If the device doesn't respond you will get this error. Most of the time this succeeds for routers (as they are AC powered and always on), but is a bit more tricky for battery devices as they tend to go to sleep and therefore are not answering any requests.

I'm aiming to improve this procedure in my zigbee-herdsman refactor (https://github.com/koenkk/zigbee-herdsman).

If you keep running into this issue and cannot get a device to work, make sure you are using the correct pairing procedure. Some pairing procedures can be found by clicking on your device modelID here: https://www.zigbee2mqtt.io/information/supported_devices.html

TLDR; If you can control the device properly, there is nothing to worry about.

closed 2019-08-05 by Koenkk