Koenkk/zigbee2mqtt #1702: Messages from groups

directman66, 2019-07-08:
Currently, I am setting up an interface for interacting with the z2m with the majordomo automation system. I noticed that if some devices are added to the group, topics with the group name begin to arrive. Is it possible to add a sign to the topic by which it would be possible to determine that the received message is a group message. For now, I determine that a message is received from a group if the message is not associated with a physical device identifier. I really do not like this option. Is it possible to add a group to the text on json - a sign that the message is a group message.

Koenkk, 2019-07-08:
I prefer not to as you could easily do this by grabbing a list of all groups (which you already do), based on the friendly name in the topic (`zigbee2mqtt/[FRIENDLY_NAME]`) you could determine if it's from a group.

directman66, 2019-07-08:
Thank you, this is a good solution to the problem. You can close the issue.

closed 2019-07-08 by Koenkk