Koenkk/zigbee2mqtt #1703: "alert" is not sent to groups

lostsourcecode, 2019-07-08:
# Bug Report
## What happened
I was trying to send 'alert': 'select' to the groups of lights I've created so that I can visually verify which group I've selected using a remote. I ended up getting "No converter available for 'alert'... " errors.

## What did you expect to happen
I expected the alert to be sent to the group of lights.

## How to reproduce it (minimal and precise)
- Create a group with a light in it.
- Send an 'alert': 'select' to the group.

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: 20190109

I've looked at the [groupConverters list in devicePublish](https://github.com/Koenkk/zigbee2mqtt/blob/6b7546547d879ea42908591e4a8c68deb89468b5/lib/extension/devicePublish.js#L16). If I just add zigbeeShepherdConverters.toZigbeeConverters.light_alert to that list it all works fine (for IKEA tradfri and a few Hue color/white bulbs). 

Is there any particular reason why light_alert should not be included there?

Koenkk, 2019-07-08:
It can be added there, can you make a PR?

lostsourcecode, 2019-07-08:
Done, see PR #1704

PR #1704 (2019-07-08): Added "alert" for groups
Fixes #1703
closed 2019-08-11 by lostsourcecode