Koenkk/zigbee2mqtt #1705: Strange data reported by CC2531 with debug router firmware

stepanov1975, 2019-07-08:
I have C2531 with latest router-cc2531-diag firmware.
I see reports like below. It seems to me strange because:
1. Why two lines with the same information?
2. 0x00124B001BD809F4 is second router. This router at a distance of 5m. The coordinator is at a distance of 10m separated by two walls. So it seems unrealistic that it will have a much stronger signal.

So is it a bug in the report? Is it a way to debug this?

zigbee2mqtt/router3 {"state":false,"linkquality":18,"last_seen":"2019-07-08T16:11:28.297Z","description":"48229/0x00124B001BD809F4","type":"NEIGHBOR","rssi":1}
zigbee2mqtt/router3 {"state":false,"linkquality":18,"last_seen":"2019-07-08T16:11:28.336Z","description":"0/0x00124B0018E1E951","type":"COORD","rssi":26}
zigbee2mqtt/router3 {"state":false,"linkquality":18,"last_seen":"2019-07-08T16:11:28.352Z","description":"48229/0x00124B001BD809F4","type":"NEIGHBOR","rssi":1}

zigbee2mqtt version:1.4.0
coordinator:20190425
CC2531 firmware version: latest router firmware

zigbee2mqtt/bridge/config {"version":"1.4.0","commit":"927c4db","coordinator":20190425,"log_level":"info","permit_join":true}

Koenkk, 2019-07-08:
I assume a report is done for every child of the router (seen in `description`). @ptvoinfo can you verify?

stepanov1975, 2019-07-09:
> I assume a report is done for every child of the router (seen in `description`). @ptvoinfo can you verify?

But the question is why to lines for the same child?

closed 2019-09-14 by stale[bot]