Koenkk/zigbee2mqtt #1707: Not connected to MQTT server! with IPv6

clemensul, 2019-07-08:
# Bug Report
## What happened
I´m using Hass.io in an IPv6 network environment with zigbee2mqtt and [MQTT Server & Web client](https://github.com/hassio-addons/addon-mqtt/blob/master/README.md) (it is a Mosquitto broker under the hood) as Add-ons.
zigbee2mqtt is not able to connect to the mqtt broker. It was working before when I was able to use IPv4. 
I can connect to the broker from other mqtt clients by using the `IPv6` and port `1883` so I suspect the issue is on the side of zigbee2mqtt.
The add-on has this configuration:
```
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": true,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://[2a02:908:2212:****:****:****:****:****]:1883",
  "serial_port": "/dev/ttyACM0",
  "devices": [],
  "network_key": [],
  "mqtt_user": "user****",
  "mqtt_pass": "8zkD******************",
  "reject_unauthorized": false,
  "disable_led": true,
  "log_level": "debug"
}
```

Log only says the following:
```
zigbee2mqtt:error 7/8/2019, 4:26:24 PM Not connected to MQTT server!
```

## What did you expect to happen

That it is working like on IPv4.

## How to reproduce it (minimal and precise)

Hass.io on a Raspberry Pi 2 with IPv6.

Hass.io is only accesible via IPv6 configured in the configuation.yaml:
```
...
http:
   server_host: ::0
...
```

Config of `MQTT Server & Web client`
```
{
  "ssl": false,
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem",
  "broker": true,
  "allow_anonymous": false,
  "mqttusers": [
    {
      "username": "user****",
      "password": "8zkD******************",
      "readonly": false,
      "topics": [
        "#"
      ]
    }
  ]
}
```

## Debug Info
zigbee2mqtt version: 1.4.0 (edge)
CC253X firmware version: not sure, but unnecessary (?)

Koenkk, 2019-07-08:
Did you try `mqtt_server` without the `[` and `]`? (`  "mqtt_server": "mqtt://[2a02:908:2212:****:****:****:****:****]:1883",`

clemensul, 2019-07-08:
Yes I did.

andreasbrett, 2019-07-09:
Could you try connecting using the hostname? Just to be sure the error is not due to string formatting.

clemensul, 2019-07-09:
I tried several other options:
`mqtt://hassio.local:1883`
`mqtt://localhost:1883`
`mqtt://127.0.0.1:1883`
`mqtt://::1/128:1883`
`mqtt://[::1/128] :1883`

None of them worked

andreasbrett, 2019-07-09:
And could you try connecting via another tool like mosquitto_sub? It's very strange that not even 127.0.0.1 is worked which would technically be an ipv4 address.

clemensul, 2019-07-09:
`127.0.0.1` is not working.
This might be because of the way hassio is configured. It is only accessible via `IPv6` or `hassio.local`:
```
...

http:
   server_host: ::0
...
```

Using `mqtt://hassio.local:1883` in the mqtt client worked. From the device itself as well as from another device in the network.

closed 2019-09-14 by stale[bot]