Koenkk/zigbee2mqtt #1708: Can not pair any device

wbsu2003, 2019-07-09:
# Bug Report
## What happened
Sorry my native language is Chinese, I use Google Translate.

I followed this article [How to support new devices](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html),  **permit_join: true** is set in **configuration.yaml**,  but never appeared "2018-5-1 18:06:41 INFO New device with address 0x00158d0001b79111 connected!"

## What did you expect to happen
I hope to recognize the new device in zigbee2mqtt, as described in the documentation [How to support new devices](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html).

## How to reproduce it (minimal and precise)
I am using this device[Wulian红外入侵探测器](http://www.wulian.cc/zhinengcg/wlhwrq.shtml),

I sniff Zigbee traffic with Wireshark. and saved the [.pcapng file](https://www.dropbox.com/s/h17y3tqfcg66x0d/wulian20190709.pcapng?dl=0)

 Can anyone tell me what to do next?

## Debug Info
zigbee2mqtt version:1.4.0
coordinator:20190425
CC2531 firmware version: latest router firmware

Koenkk, 2019-07-10:
Please provide the complete log when trying to pair the device, with:

```
advanced:
  log_level: debug
```

in `configuration.yaml`.

wbsu2003, 2019-07-11:
Thank you for your reply, I  have added debugging information according to your requirements.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3379891/log.txt)

state.json is empty；
database.db :
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018df4653","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1562634015,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"ZZpXX4a16175GKcP"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018df4653","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1562806578,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"ZZpXX4a16175GKcP"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018df4653","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1562806578,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"ZZpXX4a16175GKcP"}
```

Koenkk, 2019-07-11:
Can you post the log when running with `DEBUG=* npm start` ?

wbsu2003, 2019-07-12:
An error message was collected：
[image]

[image]

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3384851/log.txt)

A log file is also generated under /root/.npm/_logs/

[2019-07-12T03_15_11_377Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/3384871/2019-07-12T03_15_11_377Z-debug.log)

BTW： I used xiaomi WXKG01LM in the same environment, I can recognize the device, and I can subscribe to the message in mqtt.

wbsu2003, 2019-07-12:
我重新插拔了一下设备，看起来似乎正常了。
I plugged in the device again and it looks normal.

[image]

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3385039/log.txt)

wbsu2003, 2019-07-12:
[image]

Koenkk, 2019-07-12:
https://github.com/Koenkk/zigbee2mqtt/issues/1708#issuecomment-510747716 indicates there is atleast some connection.

Can you follow:
- Reflash CC2531 stick
- Remove `data/database.db`
- Start zigbee2mqtt with `DEBUG=* npm start`
- Try to pair the device
- Wait 5 minutes
- Provide the logs on http://www.pastebin.com
- Provide `data/database.db` on http://www.pastebin.com

wbsu2003, 2019-07-15:
[ ]Reflash CC2531 stick
-->I have purchased CC-Debugger but have not received it yet.

[x]Remove data/database.db
-->Yes

[x]Start zigbee2mqtt with DEBUG=* npm start
-->Yes

[x]Try to pair the device
-->Yes

[x]Wait 5 minutes
-->Yes

[x]Provide the logs on http://www.pastebin.com
-->https://pastebin.com/vM6pi6aQ

[x]Provide **data/database.db** on http://www.pastebin.com
-->https://pastebin.com/ZzdwATfk

Koenkk, 2019-07-15:
I forgot to mention that the `DEBUG=*` is logged to the console, not to the log files, can you post that output?

wbsu2003, 2019-07-16:
[x]Reflash CC2531 stick
-->Yes

From the point of view, it is the same as before.

I used various methods, but only captured info、warn、and error,  other information on the console can not be captured.

I want to use the screenshot, but the scrolling is too fast.So I decided to use the screen recording method.

 MP4 Video File：[20190716_162552.mp4](https://www.dropbox.com/s/yr5ouht3kyoh06g/20190716_162552.mp4?dl=0)

wbsu2003, 2019-07-16:
```
DEBUG=* npm start 2>&1 | tee -a log.txt
```

[log.txt](https://pastebin.com/3RUktTyL)

Koenkk, 2019-07-18:
The reason why your does not join is because the `parentaddr` is 0 (`{ nwkaddr: 52956, extaddr: '0x00124b00041f8947', parentaddr: 0 },`, see https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/components/controller.ts#L125.

However I don't know why this check is added. Can you try removing it locally and check if it works?

wbsu2003, 2019-07-19:
I am using zigbee2mqtt in Docker container, I modified /app/node_modules/zigbee-shepherd/lib/components/controller.js
[image]

I recaptured the log file
[log.txt](https://pastebin.com/q8FLq1Qe)

Koenkk, 2019-07-19:
It looks like the device does not answer `nodeDescReq`, can you try to keep the device awake after pairing by pressing buttons on it?

wbsu2003, 2019-07-19:
I understand what you mean, so I release and start pressing the reset button every second to keep the device from going to sleep.

[log.txt](https://pastebin.com/S07q46SL)

Koenkk, 2019-07-20:
The device still doesn't seem to respond, can you confirm via sniffing that it's indeed not responding?

wbsu2003, 2019-07-22:
[image]

[image]

[image]

[image]

[image]

[log.txt](https://pastebin.com/77k7yi0W)

[zigbee2019072202.pcapng](https://www.dropbox.com/s/caade92r4ny4gk4/zigbee2019072202.pcapng?dl=0)

wbsu2003, 2019-07-22:
These two files are generated at the same time, I don't know if it helps you analyze the problem.

Koenkk, 2019-07-22:
In the logs you see that the device does not answer the node descriptor request which is needed in order to join. I don't know why the device doesn't answer.

wbsu2003, 2019-07-23:
我买了这家公司整套的产品，包括移动侦查、红外控制以及视频监控等等，当然还有其他公司的产品，比如小米，因为不同的公司需要安装不同的APP，所以我希望能将这些智能产品整合到HomeAssistant中，如果你方便提供一个收件地址，我可以邮寄设备给您分析，不胜感激，谢谢！
I bought the company's complete set of products, including mobile reconnaissance, infrared control and video surveillance, and of course other companies' products, such as Xiaomi, because different companies need to install different apps, so I hope to put these smart The product is integrated into HomeAssistant. If you are convenient to provide a receiving address, I can mail it to you for analysis. Thank you!

wbsu2003, 2019-07-23:
wbsu2003@gmail.com

Koenkk, 2019-07-24:
I'm afraid I cannot do anymore than you already did, one thing you could still try is to sniff the traffic when pairing it to the original gateway.

wbsu2003, 2019-08-02:
I tried it according to your suggestion, but I don't understand it.

[NEW1.pcapng](https://www.dropbox.com/s/eifle1gpksfxbyi/NEW1.pcapng?dl=0)

Koenkk, 2019-08-03:
Is this when pairing it with the gateway of the vendor (so NOT with zigbee2mqtt)?

wbsu2003, 2019-08-04:
Yes

Koenkk, 2019-08-04:
Can you provide the encryption key? (all messages are encrypted in wireshark without having it).

wbsu2003, 2019-08-04:
I've been looking for the encryption key, but I don't know where I can find it. I'll try again and contact you when it turns out.

k3erg, 2019-08-26:
Stuck in the same place, wanted to add zigbee2mqtt support for my Wulian WL-ZCSENPB-S0420-01, but can't connect them. After some research, findings are, that all Wulian devices come with their own Trust Center Link Code instead of “ZigBeeAlliance09” preloaded.

So either we can flash or reset that key, or find it somewhere. There is supposed to be an Wulian Zigbee SDK for big companies available, but all I found until now is this: https://github.com/Wulian-WLink/GatewaySDK_v6 and I did not yet wade thru the headers or .so libs to search for the key. 

My chinese Google-Fu is weak, but I look out for more pointers, so I can use my Wulian bricks.

@wbsu2003 
Since you have the gateway:
https://www.blackhat.com/docs/us-15/materials/us-15-Zillner-ZigBee-Exploited-The-Good-The-Bad-And-The-Ugly.pdf
https://www.blackhat.com/docs/us-15/materials/us-15-Zillner-ZigBee-Exploited-The-Good-The-Bad-And-The-Ugly-wp.pdf

closed 2019-11-01 by stale[bot]