Koenkk/zigbee2mqtt #1709: Zigbee2mqtt turning on lights randomly

darksanity, 2019-07-09:
Zigbee2mqtt seems to be firing off a turn on command to my tradfri lights after a restart and seemingly randomly as well.

The only thing special is I've bound a tradfri dimmer to the light as well as using voice commands, switches through automations. Everything is working fine except the random on commands being sent.

It is definitely not a rogue automation as I checked with nothing in my automation file.

JackGruber, 2019-07-10:
Do you have set a switch on command over mqtt with a retain flag? 
Then on every reconnect of zigbee2mqtt to the mqtt broker this will happens. 

Delete retaind message with mosquitto_pub:
`mosquitto_pub -h <BROKER> -t <TOPIC> -r -n`

darksanity, 2019-07-10:
I don't believe so. What would cause a on command to be sent with a retain flag?

JackGruber, 2019-07-10:
Can you provide some Logfiles from the restart of zigbee2mqtt and from the time when the light just goes on

darksanity, 2019-07-10:
[Startup Log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3377542/Startup.Log.txt)

See attached a log from startup. Will need to go through the old logs for a random start. You can see though it just straight up turns only that light on. For some reason commands are being heavily replicated as well which I'm not sure is normal.

Koenkk, 2019-07-10:
@darksanity please post the log with below in `configuration.yaml` when the lights turn on randomly.

```yaml
advanced:
  log_level: debug
```

darksanity, 2019-07-12:
See below log showing the random turn on events.

At 8:40 am i turn the light on at the mains (0x000b3cfffef42080) and then press the zigbee switch to turn it off.

At 9:04 a couple of commands go out not issued by me. One of the commands is the on command.

As I am at work I don't realise till 12:05 at which point I turn it off. You can see it happens again later on as well.

 zigbee2mqtt:info 7/12/2019, 8:40:44 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 7/12/2019, 8:46:13 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002fbe29b', payload '{"linkquality":68,"battery":100,"voltage":3042,"click":"single"}'
  zigbee2mqtt:info 7/12/2019, 8:46:13 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002fbe29b', payload '{"linkquality":68,"battery":100,"voltage":3042,"click":""}'
  zigbee2mqtt:info 7/12/2019, 8:46:13 AM Zigbee publish to device '0x000b3cfffef42080', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 7/12/2019, 8:46:14 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":0,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:46:14 AM Zigbee publish to device '0x000b3cfffef42080', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 7/12/2019, 8:46:14 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":0,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:46:14 AM Zigbee publish to device '0x000b3cfffef42080', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 7/12/2019, 8:46:14 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":5,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:46:14 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":5,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:46:14 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":5,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:46:14 AM Zigbee publish to device '0x000b3cfffef42080', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 7/12/2019, 8:46:15 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":0,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:46:15 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":0,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:46:15 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":0,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:48:46 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffe96a450', payload '{"occupancy":true,"linkquality":44,"battery":87}'
  zigbee2mqtt:info 7/12/2019, 8:48:56 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffe96a450', payload '{"occupancy":true,"linkquality":47,"battery":87}'
  zigbee2mqtt:info 7/12/2019, 8:49:03 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffe96a450', payload '{"occupancy":true,"linkquality":42,"battery":87}'
  zigbee2mqtt:info 7/12/2019, 8:49:06 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffe96a450', payload '{"occupancy":true,"linkquality":42,"battery":87}'
  zigbee2mqtt:warn 7/12/2019, 8:49:36 AM Failed to configure 0x000b57fffe96a450 (0x000b57fffe96a450) ('Error: Timed out after 10000 ms') (attempt #2)
  zigbee2mqtt:warn 7/12/2019, 8:49:36 AM This can be ignored if the device is working properly
  zigbee2mqtt:info 7/12/2019, 8:50:06 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffe96a450', payload '{"occupancy":false,"linkquality":42,"battery":87}'
  zigbee2mqtt:info 7/12/2019, 8:50:06 AM Zigbee publish to device '0x0017880104c40f9c', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 7/12/2019, 8:50:07 AM Zigbee publish to device '0x0017880104637a87', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 7/12/2019, 8:50:07 AM MQTT publish: topic 'zigbee2mqtt/0x0017880104c40f9c', payload '{"state":"OFF","linkquality":44,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:50:07 AM MQTT publish: topic 'zigbee2mqtt/0x0017880104637a87', payload '{"state":"OFF","linkquality":44,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:50:07 AM MQTT publish: topic 'zigbee2mqtt/0x0017880104c40f9c', payload '{"state":"OFF","linkquality":44,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:50:08 AM MQTT publish: topic 'zigbee2mqtt/0x0017880104637a87', payload '{"state":"OFF","linkquality":44,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:50:08 AM MQTT publish: topic 'zigbee2mqtt/0x0017880104637a87', payload '{"state":"OFF","linkquality":44,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 8:50:08 AM MQTT publish: topic 'zigbee2mqtt/0x0017880104637a87', payload '{"state":"OFF","linkquality":44,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 9:04:31 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002fbe29b', payload '{"linkquality":55,"battery":100,"voltage":3042}'
  zigbee2mqtt:info 7/12/2019, 9:04:31 AM Zigbee publish to device '0x000b3cfffef42080', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 7/12/2019, 9:04:32 AM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"ON","linkquality":0,"brightness":254}'
[... cut ...]
  zigbee2mqtt:info 7/12/2019, 1:42:13 PM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":21,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 1:42:13 PM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":21,"brightness":254}'
  zigbee2mqtt:info 7/12/2019, 2:05:19 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002fbe29b', payload '{"linkquality":55,"battery":100,"voltage":3042}'
  zigbee2mqtt:info 7/12/2019, 3:05:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002fbe29b', payload '{"linkquality":55,"battery":100,"voltage":3042}'

Koenkk, 2019-07-12:
this log does not contains the debug logging, see https://github.com/Koenkk/zigbee2mqtt/issues/1709#issuecomment-510138446 on how to enable it.

darksanity, 2019-07-13:
Interesting this is what I have in my config

devices:
  '0x000b3cfffef42080':
    advanced:
      log_level: debug
      report: true
    friendly_name: '0x000b3cfffef42080'
    retain: false

It doesn't seem to be enabling debug mode

Koenkk, 2019-07-13:
Is the indentation correct? You need to put two spaces for `log_level`.

darksanity, 2019-07-13:
yeah indent is correct

Im getting this in log

Exception in state_received when handling msg on 'zigbee2mqtt/0x000b3cfffef42080': ''
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 203, in state_received
    values = json.loads(msg.payload)
  File "/usr/local/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

Koenkk, 2019-07-13:
In that case you must see `debug` messages when starting zigbee2mqtt, can you verify this?

darksanity, 2019-07-13:
oh i see its not meant to be at the individual light level.

See attached startup log with debug messages.

I'll send through the random ones when it happens.

Looks like the actual light has a saved "on" state? Even though it was off when zigbee2mqtt restarted?

[Startup Log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3388905/Startup.Log.txt)

darksanity, 2019-07-14:
Ok finally have debug log of a random turn on event.

As follows:

zigbee2mqtt:debug 7/14/2019, 10:41:57 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b3cfffef42080) of endpoint 1
  zigbee2mqtt:info 7/14/2019, 10:41:57 PM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"OFF","linkquality":36,"brightness":254}'
  zigbee2mqtt:debug 7/14/2019, 10:42:23 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3042},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,9]},{"elmType":33,"elmVal":689},{"elmType":32,"elmVal":97}]}}' of device 'lumi.sensor_switch' (0x00158d0002fbe29b) of endpoint 1
  zigbee2mqtt:info 7/14/2019, 10:42:23 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002fbe29b', payload '{"linkquality":42,"battery":100,"voltage":3042}'
  zigbee2mqtt:debug 7/14/2019, 10:42:23 PM Received MQTT message on 'zigbee2mqtt/0x000b3cfffef42080/set' with data '{"state": "ON"}'
  zigbee2mqtt:info 7/14/2019, 10:42:23 PM Zigbee publish to device '0x000b3cfffef42080', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 7/14/2019, 10:42:24 PM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"ON","linkquality":36,"brightness":254}'
  zigbee2mqtt:debug 7/14/2019, 10:42:24 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b3cfffef42080) of endpoint 1
  zigbee2mqtt:info 7/14/2019, 10:42:24 PM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"ON","linkquality":34,"brightness":254}'
  zigbee2mqtt:debug 7/14/2019, 10:42:24 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b3cfffef42080) of endpoint 1
  zigbee2mqtt:info 7/14/2019, 10:42:24 PM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"ON","linkquality":34,"brightness":254}'
  zigbee2mqtt:debug 7/14/2019, 10:42:24 PM Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"16387":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b3cfffef42080) of endpoint 1
  zigbee2mqtt:info 7/14/2019, 10:42:24 PM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"ON","linkquality":34,"brightness":254}'
  zigbee2mqtt:debug 7/14/2019, 10:42:24 PM Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"65533":1}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b3cfffef42080) of endpoint 1
  zigbee2mqtt:debug 7/14/2019, 10:42:24 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b3cfffef42080) of endpoint 1
  zigbee2mqtt:info 7/14/2019, 10:42:24 PM MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef42080', payload '{"state":"ON","linkquality":34,"brightness":254}'

Koenkk, 2019-07-15:
I don't know what is logged before:

```
zigbee2mqtt:debug 7/14/2019, 10:41:57 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b3cfffef42080) of endpoint 1
```

but it seems that zigbee2mqtt didn't send the zigbee command to turn it off, do you have remotes in your zigbee network or was it turned off by a switch?