Koenkk/zigbee2mqtt #1710: Updated docker dev yesterday, all xiaomi contact reporting as false

way2many, 2019-07-09:
Hi all,

Weird bug since yesterday evening. I updated the dev docker version yesterday and since then all Xiaomi contact sensors are reporting false after a few minutes. I've enabled debug for log level and this is what I have:

```
7/9/2019, 8:52:39 AM - info: 0x00158d000312e7c8 (0x00158d000312e7c8): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
7/9/2019, 8:52:39 AM - info: 0x00158d0002d46d0c (0x00158d0002d46d0c): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
7/9/2019, 8:52:39 AM - info: 0x00158d000312e83d (0x00158d000312e83d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
```

```
7/9/2019, 8:52:40 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000312e7c8', payload '{"battery":0,"voltage":"!","contact":false,"linkquality":65}'
7/9/2019, 8:52:40 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d46d0c', payload '{"battery":0,"voltage":"!","contact":false,"linkquality":36}'
7/9/2019, 8:52:40 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000312e83d', payload '{"battery":0,"voltage":"!","contact":false,"linkquality":63}'
```

```
7/9/2019, 8:52:41 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000312e7c8/availability', payload 'online'
7/9/2019, 8:52:41 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d46d0c/availability', payload 'online'
7/9/2019, 8:52:41 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000312e83d/availability', payload 'online'
```

Note: Everything was working correctly until the update.

Any clue on what is going on?

Thank you very much.

antst, 2019-07-09:
Same story here. They flip suddenly to false.

way2many, 2019-07-10:
Went back to 1.4.0-arm32v6 and had to delete state.json. It is working again perfectly so something is very wrong with latest dev.

Koenkk, 2019-07-10:
On which version was it still working?

way2many, 2019-07-10:
I always used latest docker dev version for Raspberry Pi so I suppose it was 1.4.0-arm32v6 at that time.

Koenkk, 2019-07-10:
I think it's due to https://github.com/Koenkk/zigbee-herdsman/commit/b2fa80f970ef5d6dca2909cb9dfbda047ec7a019

antst, 2019-07-10:
Is it safe to roll back only this commit?
I will try.
It is annoying, light in my storage is always on )

antst, 2019-07-10:
OK, giving it a try with roll-back

Koenkk, 2019-07-10:
I verified that https://github.com/Koenkk/zigbee-herdsman/pull/2 is the issue, fixed in latest dev branch.

closed 2019-07-10 by Koenkk

way2many, 2019-07-14:
Been running with latest dev and everything runs smoothly. Thanks for the quick correction/action.

mar005, 2019-10-04:
I am having this issue right here where I just started with version 0.1.6 and the only contact sensor I have paired is constantly showing false on contact.