Koenkk/zigbee2mqtt #1711: control brightness of RGBW lesstrip

markokruijer, 2019-07-09:
Hi,

i've succesfully flashed a usb sniffer to use as a Zigbee gateway and am now in the proces of connecting a RGBW ledstrip and i'm almost there. Zigbee2mqtt detects the ledstrip (a Gledopto GL-C-008) and using mosquito messages i can change the led colors, and turn on/off the white leds. Connecting this to HASSIO also works, albeit with similar issues. I can use the coor wheel to change colors, use brightness to change rgb brightness, and color temp to change the temp of the rgb leds. However, for the white leds none of this work, and i cannot turn on/off the white leds at all from within HASSIO.

What would be the correct way of solving this? I can turn on the white led from zigbee2mqtt, but without brightness controls it feels like there is something wrong. 
I'm more than willing to debug and code some JS here (experienced developer) but I'm not sure where to start debugging this.

JackGruber, 2019-07-10:
Good morning,

do you get an error when you set the brightness?
I have also Gledopto GL-C-008 an set the brightness over mqtt

`{"state": "ON", "brightness":5 , "color_temp": 500, "transition":0}`

markokruijer, 2019-07-10:
Can you control both leds with that command? in my case this does not fly. With your command my rgb leds change, but i get an error, possibly related to why the white leds are not doing anything:

Also note, i'm sending the commands to the this device:
zigbee2mqtt/0x00124b001d04d7e2/set 
If can also target white and rgb directly (zigbee2mqtt/0x00124b001d04d7e2/white/set) but that's not what i want of course ;)

`  zigbee2mqtt:info 7/10/2019, 1:13:14 PM Zigbee publish to device '0x00124b001d04d7e2', genLevelCtrl - moveToLevelWithOnOff - {"level":5,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 7/10/2019, 1:13:21 PM Zigbee publish to device '0x00124b001d04d7e2', genLevelCtrl - moveToLevelWithOnOff - {"level":5,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:info 7/10/2019, 1:13:21 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x00124b001d04d7e2","type":"device","friendlyName":"0x00124b001d04d7e2"},"message":"{\"state\": \"ON\", \"brightness\":5 , \"color_temp\": 500, \"transition\":0}"}}'
  zigbee2mqtt:info 7/10/2019, 1:13:21 PM Zigbee publish to device '0x00124b001d04d7e2', lightingColorCtrl - moveToColorTemp - {"colortemp":500,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 7/10/2019, 1:13:21 PM Zigbee publish to device '0x00124b001d04d7e2', lightingColorCtrl - moveToColorTemp - {"colortemp":500,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:info 7/10/2019, 1:13:21 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x00124b001d04d7e2","type":"device","friendlyName":"0x00124b001d04d7e2"},"message":"{\"state\": \"ON\", \"brightness\":5 , \"color_temp\": 500, \"transition\":0}"}}'`

markokruijer, 2019-07-10:
I debugged some more and this error seemed to be related to proximity to the gateway (i was in the kitchen). I'm now closer and this error no longer is there.

If i send this command for example: {"state": "OFF", "brightness":200 , "color_temp":100, "transition":0}

The brightness changes, but the state OFF definitely does not work, so the question remains, how do i enable/disable the white led? brightness 0 doesn't work either.

JackGruber, 2019-07-10:
I use no HASSIO, only MQTT commands.

Your command would send a off and then a go to color 100 in 0 seconds, and your strip is on again.
I'm not 100% sure but for brightness is 100 the max value.

Switch off 
`{"state": "off"}`

Got from current HUE color to new HUE coller
`{"brightness":100, "color": {"hue": 200}, "transition": 10 }`

Or with RGB
`{"brightness":100, "color": {"r":100, "g":50, "b":200}, "transition": 10 }`

markokruijer, 2019-07-10:
i must say the plot thickens. After inspecting my logs I notice in my database.db that the controller is detected as: 
`"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13,15]`

And my npm instance reports this on adding devices:
`  zigbee2mqtt:info 7/10/2019, 2:25:42 PM Currently 1 devices are joined:
  zigbee2mqtt:info 7/10/2019, 2:25:42 PM 0x00124b001d04d7e2 (0x00124b001d04d7e2): GL-C-008 - Gledopto Zigbee LED controller RGB + CCT / RGBW / WWCW / Dimmer (Router)`

This does not seem right to me. Is this the same in your case?

markokruijer, 2019-07-10:
ok, it looks like this is default behaviour in the sense that a few types are identified as 008 and in my case the endpoints are set as follows:
`else if (device.epList.toString() === '11,13,15') {
                return {
                    'rgb': 11,
                    'white': 15,
                };`
So, there's separate ones for white and rgb. I'm now wondering how zigbee2mqtt determines which one to use, my best guess is the first one is also the default?

So, it looks like I have 2 controllers, that need to be handled apart form each other. How do i configure Hassio for this? surely this is possible?

Koenkk, 2019-07-10:
@markokruijer let's first check how we can integrate this via a manual configuration (afterwards we can do automatic), can you add to your home assistant `configuration.yaml`

```yaml
light:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/<FRIENDLY_NAME>"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    schema: "json"
    command_topic: "zigbee2mqtt/<FRIENDLY_NAME>/white/set"
```

Does this provide the good behaviour?

markokruijer, 2019-07-10:
@Koenkk this works, but for the white leds only. as expected since they have separate command endpoints. I can turn on/off the white leds and change brightness accordingly.

Continuing on my HASSIO integration, i managed to find a way to define a dummy mqtt light, which in turn will alias commands to the two separate endpoints. Seems doable, so will post back once I solved that puzzle.

davdav42, 2019-07-16:
Hello, I have a quite similar problem linked to this one.

I have a raspberry pi with Domoticz, And the ZigbeeM2qtt. I want to control a led ribbon with a GLEDOPTO RGB+CCT.
The led ribbon has 2 kind of led RBG and White and 6 wire (V+, RGB, W, WW).
I can successfully control the RGB, I can switch off/on change the color. but it's impossible to control the white leds. These one stay switched on and I cannot change the intensity.

You will find below the database.db for this device ::
{"id":3,"type":"Router","ieeeAddr":"0x00124b001ba6f3c7","nwkAddr":22174,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13,15],"zclVersion":1,"appVersion":1,"stackVersion":2,"hwV2,"hwVersion":1,"swBuildId":"2.0.2","status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":evId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{}},"15":{"profId":49246,"epId":15,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}}},"dateCode":"20180920","_id":"1X7PKOIHid2AmNLY"}
{"id":3,"type":"Router","ieeeAddr":"0x00124b001ba6f3c7","nwkAddr":22174,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13,15],"zclVersion":1,"appVersion":1,"stackVersion":2,"hwV2,"hwVersion":1,"swBuildId":"2.0.2","status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":evId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{}},"15":{"profId":49246,"epId":15,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}}},"dateCode":"20180920","_id":"1X7PKOIHid2AmNLY"}

and below the device.js :

// Gledopto
{
    zigbeeModel: ['GLEDOPTO', 'GL-C-008', 'GL-C-007'],
    model: 'GL-C-008',
    vendor: 'Gledopto',
    description: 'Zigbee LED controller RGB + CCT / RGBW / WWCW / Dimmer',
    extend: gledopto.light_onoff_brightness_colortemp_colorxy,
    ep: (device) => {
        if (device.epList.toString() === '11,12,13') {
            return {'': 12};
        } else if (device.epList.toString() === '10,11,13' || device.epList.toString() === '11,13') {
            return {'': 11};
        } else if (device.epList.toString() === '11,12,13,15') {
            return {
                'rgb': 12,
                'white': 15,
            };
        } else if (device.epList.toString() === '11,13,15') {
            return {
                'rgb': 11,
                'white': 15,
            };
        } else {
            return {};
        }
    },
},
On Domoticz the controller is a RGBWW.

Can somebody could help me. Than you very much in advance

I tried to add what koenkk wrote in the configuration.yaml . and reboot my raspberry. But there is no change. still can't control the white led. (I'm not sure of what to do after modify the configuration.yaml, It's take into account after a reboot or there is others things to do?)

markokruijer, 2019-07-17:
Using Hassio i created two lights, one for RGB and one for White. I'm not sure how you would do this using Domoticz, but I assume a similar principle applies. This is my Hassio config for the led strip:

`#mqtt light, manual entry
light:
    #rgb ledstrip endpoint
  - platform: "mqtt"
    name: "ledstrip_rgb"
    state_topic: "zigbee2mqtt/0x00124b001d04d7e2"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    color_temp: true
    xy: true
    schema: "json"
    command_topic: "zigbee2mqtt/0x00124b001d04d7e2/set"
    #white ledstrip endpoint
  - platform: "mqtt"
    name: "ledstrip_white"
    command_topic: "zigbee2mqtt/0x00124b001d04d7e2/white/set"
    availability_topic: "zigbee2mqtt/bridge/state"
    state_topic: "zigbee2mqtt/0x00124b001d04d7e2"
    brightness: true
    schema: "json"`

Note the separate command topic for the white led strip. That's the only way I can control the white leds, but in my case that is not a problem. I am still working on a better config, since my white state is currently not working so it's always on according to Hassio, but that's another problem ;)

G1K, 2019-07-18:
Works like this.
in a topic there is "state_white": "ON", but it is impossible to use it in the JSON scheme TK there is no parameter state_template/state_template_value ( https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/mqtt/light/schema_json.py#L206 , and will not do https://github.com/home-assistant/home-assistant/pull/22416 ) , as a result, there is no controlled control, so included optimistic .

```
  - platform: "mqtt"
    name: "ledstrip_rgb"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    command_topic: "zigbee2mqtt/0x00124b001a15a404/set"
    schema: "json"
    state_topic: "zigbee2mqtt/0x00124b001a15a404"
    xy: true
  - platform: "mqtt"
    name: "ledstrip_white"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    color_temp: true
    command_topic: "zigbee2mqtt/0x00124b001a15a404/white/set"
    optimistic: true
    schema: "json"
```

**The temperature is controlled in the white lamp, and switches the cold/warm led.**
Brightness is controlled though separately, but the state is common. Therefore, incorrect data are obtained on the first lamp, but you can control it.

The problem is slightly intersects with https://github.com/Koenkk/zigbee2mqtt/issues/1731

Koenkk, 2019-08-04:
Once I receive this device I will look into this and fix the integration (https://github.com/Koenkk/zigbee2mqtt/issues/1731#issuecomment-517996035). 

TLDR; we should treat it as one instead of two separate lights.

Koenkk, 2019-09-29:
I've improved this integration (https://github.com/Koenkk/zigbee2mqtt/issues/1731#issuecomment-533683900), please checkout the dev branch.

Can somebody verify that it is OK now?

Koenkk, 2019-10-03:
This issue should be fixed now in the latest dev branch. Assuming this can be closed now.

closed 2019-10-03 by Koenkk

igorpuchkovcom, 2019-11-13:
Unable to adjust RGB and WHITE simultaneously.

If white_value is present RGB controller is just switching off.
Same behaviour is for RGB. When there is no white_value value present WHITE controller switching off.

No way to have both controlles switched on and managed at the same time.

There is three independent parameters:
- Color
- Brightness (applied only for RGB controller)
- White value (applied only for WHITE controller)

ON/OFF state should be shared between controllers.

bjrne, 2020-11-27:
> If white_value is present RGB controller just switching off.
@igorpuchkovcom

I know it's a late answer, but I came here through google, so others might as well. The behavior you're describing is the behavior of the controller itself, once you send a colortemp it shuts down rgb and vice-versa. So it's the fault of the controller, not the software. Unfortunately it is very rarely mentioned on product pages of this product that you cannot run both parts simultaneously :/