Koenkk/zigbee2mqtt #1713: zigbee2mqtt does not startup correctly

odx, 2019-07-10:
After rebooting the Raspberry Pi today, zigbee2mqtt seems to hang during startup. From the log I cannot identify any suspicious error. It looks to me like it is just not completing the startup routine. Here is the debug log:

```
> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 7/10/2019, 7:01:44 AM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-10.07-01-43'
  zigbee2mqtt:debug 7/10/2019, 7:01:44 AM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-06-16.19-51-06'
  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:debug 7/10/2019, 7:01:45 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 7/10/2019, 7:01:45 AM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 7/10/2019, 7:01:45 AM Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
  zigbee2mqtt:info 7/10/2019, 7:01:45 AM Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyUSB0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +11ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyUSB0 +17ms
  serialport/stream _write 1 bytes of data +3ms
  serialport/binding-abstract write 1 bytes +31ms
  serialport/stream _read reading +4ms
  serialport/binding-abstract read +3ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned null 1 +12ms
  serialport/bindings/unixWrite wrote 1 bytes +1ms
  serialport/bindings/unixWrite Finished writing 1 bytes +1ms
  serialport/stream binding.write write finished +22ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +18ms
  serialport/bindings/poller Polling for "readable" +39ms
  cc-znp The serialport /dev/ttyUSB0 is opened. +1s
  zigbee-shepherd:request REQ --> SYS:version +0ms
  cc-znp:SREQ --> SYS:version, {} +0ms
  serialport/stream _write 5 bytes of data +1s
  serialport/binding-abstract write 5 bytes +1s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s
  serialport/bindings/unixWrite write returned null 5 +2ms
  serialport/bindings/unixWrite wrote 5 bytes +1ms
  serialport/bindings/unixWrite Finished writing 5 bytes +1ms
  serialport/stream binding.write write finished +10ms
  serialport/bindings/poller received "readable" +1s
  serialport/binding-abstract read +16ms
  serialport/bindings/unixRead Starting read +1s
  serialport/bindings/unixRead Finished read 19 bytes +1ms
  serialport/stream binding.read finished +13ms
  cc-znp { sof: 254,
  cc-znp   len: 14,
  cc-znp   type: 'SRSP',
  cc-znp   subsys: 'SYS',
  cc-znp   cmd: 'version',
  cc-znp   payload:
  cc-znp    { transportrev: 2,
  cc-znp      product: 0,
  cc-znp      majorrel: 2,
  cc-znp      minorrel: 6,
  cc-znp      maintrel: 3,
  cc-znp      revision: 20190425 },
  cc-znp   fcs: 146,
  cc-znp   csum: 146 } +109ms
  serialport/stream _read reading +63ms
  serialport/binding-abstract read +69ms
  serialport/bindings/unixRead Starting read +67ms
  cc-znp:SRSP <-- SYS:version, { transportrev: 2, product: 0, majorrel: 2, minorrel: 6, maintrel: 3, revision: 20190425 } +0ms
  zigbee-shepherd:request RSP <-- SYS:version +138ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +5ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +134ms
  serialport/stream _write 8 bytes of data +26ms
  serialport/binding-abstract write 8 bytes +27ms
[... 239 lines cut ...]
  cc-znp:SRSP <-- ZDO:startupFromApp, { status: 0 } +822ms
  serialport/bindings/unixRead Finished read 6 bytes +4ms
  serialport/stream binding.read finished +6ms
  cc-znp { sof: 254,
  cc-znp   len: 1,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'stateChangeInd',
  cc-znp   payload: { state: 8 },
  cc-znp   fcs: 140,
  cc-znp   csum: 140 } +17ms
  serialport/stream _read reading +12ms
  serialport/binding-abstract read +17ms
  serialport/bindings/unixRead Starting read +14ms
  cc-znp:AREQ <-- ZDO:stateChangeInd, { state: 8 } +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:stateChangeInd +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +7ms
  serialport/bindings/poller Polling for "readable" +43ms
```

is there anything suspicous?

## Debug Info
zigbee2mqtt version: 1.40
CC2530 firmware version: 20190425

Koenkk, 2019-07-10:
`  cc-znp:AREQ <-- ZDO:stateChangeInd, { state: 8 } +0ms` indicates that the coordinator does not start, can you try with the latest firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

odx, 2019-07-10:
Thanks. works again. This is now the second time I need to upgrade the firmware because there is something wrong with the coordinator. I am wondering if actually the new firmware fixes the problems or just the case that the coordinator has been flashed again and there is some integrity problem underlying as this occurs suddenly without any configuration change.

Should we drill deeper the next time?

Koenkk, 2019-07-11:
If it fails with this firmware, we should indeed check, but at least `20190425` is a problematic firmware (e.g. https://github.com/Koenkk/Z-Stack-firmware/issues/86#issuecomment-510338782)

odx, 2019-07-28:
A very similar behavior reappeared:
The network was dysfunctional again. I hat to reset the CC2530 in order to spin up zigbee2mqtt again. I did so by rebooting the complete pi.

This worked two times. The third time the coordinator "crashed" I was no longer able to start zigbee2mqtt any more. I updated to 1.5.1 with no luck.

Here is the debug log when starting up. I can also provide the "standard"-logs of the crashed states (will require some anonymization so I don't post them yet). I believe that it has something to do with my IKEA tradfri bulb which seems to be involved in the last transactions before the coordinator crashes.

```
> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:info 7/28/2019, 8:06:20 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-28.20-06-18'
  zigbee2mqtt:debug 7/28/2019, 8:06:20 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-07-28.19-30-03'
  zigbee2mqtt:debug 7/28/2019, 8:06:20 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 7/28/2019, 8:06:20 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 7/28/2019, 8:06:20 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 7/28/2019, 8:06:20 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyUSB0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +11ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyUSB0 +15ms
  serialport/stream _write 1 bytes of data +3ms
  serialport/binding-abstract write 1 bytes +27ms
  serialport/stream _read reading +3ms
  serialport/binding-abstract read +3ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned null 1 +11ms
  serialport/bindings/unixWrite wrote 1 bytes +1ms
  serialport/bindings/unixWrite Finished writing 1 bytes +1ms
  serialport/stream binding.write write finished +18ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +15ms
  serialport/bindings/poller Polling for "readable" +31ms
  cc-znp The serialport /dev/ttyUSB0 is opened. +1s
  zigbee-shepherd:request REQ --> SYS:version +0ms
  cc-znp:SREQ --> SYS:version, {} +0ms
  unpi:writer --> frame [254,0,33,2,35] +0ms
  serialport/stream _write 5 bytes of data +1s
  serialport/binding-abstract write 5 bytes +1s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s
  serialport/bindings/unixWrite write returned null 5 +2ms
  serialport/bindings/unixWrite wrote 5 bytes +0ms
  serialport/bindings/unixWrite Finished writing 5 bytes +1ms
  serialport/stream binding.write write finished +7ms
  cc-znp:SRSP <-- SYS:version, __timeout__ +0ms
  zigbee-shepherd:request RSP <-- SYS:version +3s
  zigbee-shepherd:request REQ --> SYS:osalNvRead +15ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +3s
  unpi:writer --> frame [254,3,33,8,0,15,0,37] +3s
  serialport/stream _write 8 bytes of data +3s
  serialport/binding-abstract write 8 bytes +3s
  serialport/bindings/unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +3s
  serialport/bindings/unixWrite write returned null 8 +1ms
  serialport/bindings/unixWrite wrote 8 bytes +1ms
  serialport/bindings/unixWrite Finished writing 8 bytes +1ms
  serialport/stream binding.write write finished +7ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)
    at Object.onceWrapper (events.js:286:20)
    at CcZnp.emit (events.js:198:13)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
[... 31 lines cut ...]
  zigbee-shepherd:request REQ --> SYS:version +1m
  cc-znp:SREQ --> SYS:version, {} +1m
  unpi:writer --> frame [254,0,33,2,35] +1m
  cc-znp:SRSP <-- SYS:version, __timeout__ +1m
  zigbee-shepherd:request RSP <-- SYS:version +3s
  zigbee-shepherd:request REQ --> SYS:osalNvRead +4ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +3s
  unpi:writer --> frame [254,3,33,8,0,15,0,37] +3s
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)
    at Object.onceWrapper (events.js:286:20)
    at CcZnp.emit (events.js:198:13)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
[... cut ...]
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 7/28/2019, 8:07:34 PM Exiting...
```

Koenkk, 2019-07-29:
Which firmware are you currently running? If that is the latest one, can you verify that reflashing the CC2530 fixes it? If so, we probably need to increase the memory heap for this firmware. (I would recommend https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

odx, 2019-07-29:
Yes I used Junes Source Routing Firmware.
After reflashing the coordinator initializes as usual.

odx, 2019-07-30:
Was only operational for a few hours...

Koenkk, 2019-07-30:
Do you have the reporting or availability feature enabled? (please post your `configuration.yaml`.)

odx, 2019-07-30:
Yes, reporting is on.

```
homeassistant: false
permit_join: false
advanced:
  network_key:
    - my
    - super
    - secret
    - key
    - ...
  last_seen: ISO_8601
  elapsed: true
  report: true
mqtt:
  base_topic: zigbee
  server: 'mqtt://localhost'
  user: thing
  password: mymqttpassword
serial:
  port: /dev/ttyUSB0
devices:
  '0x00158d000273cfdc':
    friendly_name: temp2
    retain: true
  '0x00124b001b6e0976':
    friendly_name: router1
    retain: false
  '0x00158d00027305c3':
    friendly_name: pir1
    retain: false
    occupancy_timeout: 300
  '0x00158d0002b6ee13':
    friendly_name: pir3
    retain: false
    occupancy_timeout: 300
  '0x00158d00027315e8':
    friendly_name: temp1
    retain: true
  '0x00158d0002c6371f':
    friendly_name: pir2
    retain: false
    occupancy_timeout: 300
  '0x00158d000273e222':
    friendly_name: door1
    retain: false
  '0x00158d0002d481c3':
    friendly_name: terrassentuer_rechts
    retain: false
  '0x00158d0002d48263':
    friendly_name: terrassentuer_links
    retain: false
  '0x00158d000273e216':
    friendly_name: terrassentuer_kueche
    retain: false
  '0x00158d0002d489fb':
    friendly_name: wohnzimmerfenster_links
    retain: false
  '0x00158d0002e20b98':
    friendly_name: wohnzimmerfenster_rechts
    retain: false
  '0x00158d0002d489e2':
    friendly_name: wc-fenster
    retain: false
  '0x00158d0002d48259':
    friendly_name: fenster_abstellraum
    retain: false
  '0x00158d0002e25588':
    friendly_name: badfenster
    retain: false
  '0x00158d0002ca2d76':
    friendly_name: dachfenster_baz
    retain: false
  '0x00158d0002d48174':
    friendly_name: garagentor
    retain: false
  '0x00158d0002d481ed':
    friendly_name: fenster_foobar_links
    retain: false
  '0x00158d0002d48177':
    friendly_name: fenster_foobar_rechts
    retain: false
  '0x00158d0002e20d06':
    friendly_name: dachfenster_galerie
    retain: false
  '0x00158d0002d481e9':
    friendly_name: giebelfenster_links
    retain: false
  '0x00158d0002d48919':
    friendly_name: giebelfenster_rechts
    retain: false
  '0x00158d000273e1a3':
    friendly_name: balkontuer
    retain: false
  '0x00158d0002d481b3':
    friendly_name: garage_gartentuer
    retain: false
  '0x00124b001bb8e9e9':
    friendly_name: router3
    retain: false
  '0x00124b001bb8620b':
    friendly_name: router2
    retain: false
  '0x000b57fffe900c86':
    friendly_name: nachttisch_bar
    retain: false
  '0x000b57fffe976159':
    friendly_name: nachttisch_foo
    retain: false
  '0x000d6ffffe24cb50':
    friendly_name: sofalampe
    retain: false
  '0xd0cf5efffef808ed':
    friendly_name: RemoteSZ
    retain: false
  '0x00158d0003042554':
    friendly_name: pir_wz
    retain: false
    occupancy_timeout: 300
  '0x00158d000305875c':
    friendly_name: pir_bad
    retain: false
    occupancy_timeout: 300
  '0x00158d000349cb43':
    friendly_name: temp_sz
    retain: false
  '0x00158d00031b315d':
    friendly_name: spare_broken
    retain: false
  '0x00158d00031b28b5':
    friendly_name: pir_sz
    retain: false
    occupancy_timeout: 90
  '0x000d6ffffe7aca4b':
    friendly_name: tradfri_repeater
    retain: false
  '0x00158d0002758682':
    friendly_name: smoke_wz
    retain: true
  '0x00158d0002754836':
    friendly_name: smoke_sz
    retain: true
  '0x00158d0002755923':
    friendly_name: smoke_foobar
    retain: true
  '0x00158d0002758628':
    friendly_name: smoke_galerie
    retain: true
  '0x00158d000275472a':
    friendly_name: smoke_baz
    retain: true
  '0x00158d000275878d':
    friendly_name: smoke_flur_og
    retain: true
  '0x00158d0002759edc':
    friendly_name: smoke_flur_eg
    retain: true
  '0x00158d0002758987':
    friendly_name: smoke_broken
    retain: false
  '0x00158d0003021a63':
    friendly_name: pir_abstell
    retain: false
  '0x7cb03eaa0a068a51':
    friendly_name: plug1
    retain: false

```

odx, 2019-07-31:
Reflashed the CC2530 again and turner reporting off. It is running for a day (and night) now.
Hope that was it. Thanks for the hint. I wasn’t aware that the experimental reporting was still on.

closed 2019-07-31 by odx

dmak, 2019-12-10:
Same issue here:
```
12/10/2019, 1:30:35 AM - error: Not connected to MQTT server!
12/10/2019, 1:30:36 AM - info: Logging to directory: '/var/log/homeassistant/zigbee2mqtt'
12/10/2019, 1:30:36 AM - info: Starting zigbee2mqtt version 1.6.0 (commit #unknown)
12/10/2019, 1:30:36 AM - info: Starting zigbee-shepherd
12/10/2019, 1:30:55 AM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: SRSP - SYS - osalNvRead after 6000ms)
```
Reboot does not help... What helped is just unplug and plug back USB stick.

I haven't upgraded or changed anything since last boot.

I have reporting feature enabled.