Koenkk/zigbee2mqtt #1715: Zigbee2mqtt failed to start

BennieGroen, 2019-07-10:
I bought a pre-flashed cc2531, and trying to use this with domoticz on a raspberry pi 3B+ running debian buster. The initial install ran for an evening, I could operate an IKEA wall plug from domoticz, and was getting data from an Xiaomi temperature sensor in domoticz. But at some point the setup stopped working, I do not know exactly what I did to cause this, I was trying to pair an additional Xiaomi temperature sensor, but never succeeded. At some point I noticed the other devices stopped working as well.

I read several "failed to start" issues for inspiration to solve this. Tried a lot of suggested fixes, including resetting the stick with the button closest to the USB (light turned off). Tried a fresh installation twice, formatting the SD and starting over. At this point i am running stretch and still get the same log at start-up of zigbee2mqtt. Find the response from `DEBUG=cc-znp* npm start` below.

[https://wtools.io/code/raw/zjW](url)

It looks like the stick is not responding to cc-znp. Am I right? Any help sorting this out is appreciated.

ondoteam, 2019-07-10:
> But at some point the setup stopped working

Are you talking (or writing) about the temperature sensor? I have read many reports like that (it happens to me too). It is working fine a random number of days, and then stop sendind data. I have tried to change the battery, leave it near a router, etc., but nothing works.

>  I was trying to pair an additional Xiaomi temperature sensor

http://www.zigbee2mqtt.io/devices/WSDCGQ01LM

About your log: are you  using the "dev" branch? Because there are other reports about the same error.

BennieGroen, 2019-07-10:
> About your log: are you using the "dev" branch? Because there are other reports about the same error.

I do not think so. I am using the command `sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt`, as indicated in the [guide](https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html). That takes me to the "master" branch? 
I did try to find the same error on the forum, but couldn't find it. Both `cc-znp:SREQ --> SYS:version` and `cc-znp:SREQ --> SYS:osalNvRead` timeout.

> Are you talking (or writing) about the temperature sensor? I have read many reports like that (it happens to me too). It is working fine a random number of days, and then stop sendind data. I have [...]

Both the IKEA wall plug and the sensor stopped functioning in Domoticz. I started trouble shooting the problem, restarting the pi, and I noticed error's in starting zigbee2mqtt.
But, I am not sure that I am getting the same errors now. I did several fresh installations, I lost track...

Koenkk, 2019-07-10:
The stick is not responding anymore, I'm afraid it was running an unstable firmware and you need to reflash it (last months we spend a lot of time in improving the firmware stability). 

Current stable firmwares can be found here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

BennieGroen, 2019-07-10:
> The stick is not responding anymore, I'm afraid it was running an unstable firmware and you need to reflash it (last months we spend a lot of time in improving the firmware stability).
>
> Current [...]

I was hoping I did not need to go this route.  But, I got my order in for the debugger, downloader cable and an extra stick to be save.
tnx for the quick reply! I'll let you know if this solves the issue.

BennieGroen, 2019-07-19:
Works like a charm after reflashing. Tnx!

closed 2019-07-19 by BennieGroen