Koenkk/zigbee2mqtt #1716: Allow to configure binding via `configuration.yaml`

Koenkk, 2019-07-10:
Similar to allow configuring of groups via `configuration.yaml`, we should also allow it for bindings.

directman66, 2019-07-11:
We are looking forward to the decision, I would also like to receive the result through the MQTT query

directman66, 2019-07-19:
Are there any updates on this topic?

Koenkk, 2019-07-20:
Not yet.

BudBundi, 2019-07-22:
I little feature request from my side, because we have no light link and can't transfer binding groups from remote to remote.
At the moment to control a lamp or group with more than one remote you have to, sniff the group to which the remote sends (IKEA), create a group for every remote and at the lamp to every group.

Maybe it would possible to change

'13':
    friendly_name: group_1
'41':
    friendly_name: group_2
'71':
    friendly_name: group_3

to

'13','41','71':
    friendly_name: group

directman66, 2019-08-20:
the issue is not resolved?

Koenkk, 2019-10-19:
Unstale

zsiti, 2019-12-18:
This would absolutely fantastic! Is it still on the roadmap?

closed 2020-02-23 by stale[bot]